moveit/moveit #2317: install moveit on noetic binaries fails

sonelu, 2020-09-15:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? Binary
* If binary, which release version? Last one?
* If source, which branch?

### Steps to reproduce
Follow instructions from [https://moveit.ros.org/install/](https://moveit.ros.org/install/):

- install ROS Noetic as described in [http://wiki.ros.org/noetic/Installation/Ubuntu](http://wiki.ros.org/noetic/Installation/Ubuntu)
- execute ``sudo apt install ros-noetic-moveit``

### Expected behaviour
Expected to be able to install moveit (and later plugins, planners, etc.)

### Actual behaviour
After successful installation of ROS Noetic when running
``sudo apt install ros-noetic-moveit``
I get:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-noetic-moveit
```

**Context:**
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

```
$ cat /etc/apt/sources.list.d/ros-latest.list
deb http://packages.ros.org/ros/ubuntu focal main
```

```
$ sudo apt-get update
Hit:1 http://gb.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://archive.ubuntu.com/ubuntu focal InRelease                                  
Hit:3 http://gb.archive.ubuntu.com/ubuntu focal-updates InRelease                       
Hit:4 http://gb.archive.ubuntu.com/ubuntu focal-backports InRelease                     
Hit:5 http://archive.canonical.com/ubuntu focal InRelease                      
Hit:6 http://gb.archive.ubuntu.com/ubuntu focal-security InRelease             
Hit:7 http://packages.ros.org/ros/ubuntu focal InRelease 
Reading package lists... Done
```

```
$ apt search ros-noetic-moveit*
Sorting... Done
Full Text Search... Done
ros-noetic-moveit-msgs/focal 0.11.0-1focal.20200820.045932 amd64
  Messages, services and actions used by MoveIt

ros-noetic-moveit-resources/focal 0.7.0-1focal.20200826.194755 amd64
  Resources used for MoveIt! testing

ros-noetic-moveit-resources-fanuc-description/focal 0.7.0-1focal.20200817.131837 amd64
  Fanuc Resources used for MoveIt! testing

ros-noetic-moveit-resources-fanuc-moveit-config/focal 0.7.0-1focal.20200826.191813 amd64
  MoveIt Resources for testing: Fanuc M-10iA.

ros-noetic-moveit-resources-panda-description/focal 0.7.0-1focal.20200817.131930 amd64
  panda Resources used for MoveIt! testing

ros-noetic-moveit-resources-panda-moveit-config/focal 0.7.0-1focal.20200826.194134 amd64
  MoveIt Resources for testing: Franka Emika Panda A project-internal configuration for testing in MoveIt.

ros-noetic-moveit-resources-pr2-description/focal 0.7.0-1focal.20200817.132007 amd64
  PR2 Resources used for MoveIt! testing
```

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2020-09-15:
I believe this is a duplicate of #2308.

tylerjw, 2020-09-16:
I'm sorry if you found this confusing.  We probably announced the release too early.  The binaries are available in the testing repos and will be available for install using the normal repos after the next noetic sync.

sonelu, 2020-09-17:
Ok, thanks for the notice. I managed to install from the testing repository.

closed 2020-09-17 by sonelu