moveit/moveit #2321: Error in moving to the desired position

nihalar, 2020-09-24:
### Description

I want to move the robot only in the X-direction. I gave velocity 10 units in x-direction, but it caused the robot to move in all three direction (slight movement of less than 2cm, but I need very accurate movements). Is there something we can do to solve this.

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source 

### Steps to reproduce
1. Create a geometry_msgs::TwistStamped message
2. Give the following values:
twist.twist.linear.x = 10.0
twist.twist.linear.y = 0.0
twist.twist.linear.y = 0.0
twist.twist.angular.x = 0.0;
twist.twist.angular.y = 0.0;
twist.twist.angular.z = 0.0;
3. Publish to the topic "/servo_server/delta_twist_cmds"

### Expected behaviour
Only x position of the end effector should be changed

### Actual behaviour
All x,y,z,r,p,y of the end effector was changed. (Apart from x, the changes in all other directions is less than 2cm, but our application requires very accurate movements)

AndyZe, 2020-09-24:
My first thought is that twist.linear.x of 10 m/s is probably too fast.

Is there a chance the robot is starting near a singularity?

What kind of robot is it? Can you post the servo_settings.yaml file you're using?

nihalar, 2020-09-24:
Thank you for the reply. The settings file is below

###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: true # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
robot_link_command_frame:  world  # commands must be given in the frame of a robot link. Usually either the base or end effector
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.6  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.3 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.01
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
publish_period: 0.008  # 1/Nominal publish rate [seconds]

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: std_msgs/Float64MultiArray

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## MoveIt properties
move_group_name:  arm  # Often 'manipulator' or 'arm'
planning_frame: world  # The MoveIt planning frame. Often 'base_link' or 'world'

## Stopping behaviourbut 
incoming_command_timeout:  0.1  # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 4

## Configure handling of singularities and joint limits
lower_singularity_threshold:  17  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: servo_server/delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: servo_server/delta_joint_cmds # Topic for incoming joint angle commands
joint_topic:  joint_states
status_topic: servo_server/status # Publish status to this topic
command_out_topic: joint_group_position_controller/command # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 50 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the distance is decreasing. Requires joint acceleration limits
collision_check_type: threshold_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.001 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.005 # Start decelerating when a scene collision is this far [m]
# Parameters for "stop_distance"-type collision checking
collision_distance_safety_factor: 1000 # Must be >= 1. A large safety factor is recommended to account for latency
min_allowable_collision_distance: 0.001 # Stop if a collision is closer than this [m]

The robot is a modified version of ur5.

Just to add a question. If I want to move the robot +0.2m in the +x direction, what value should be given to twist.linear.x? I'm a bit confused on how the twist message is converted to position. 

Thank You :) 

> My first thought is that twist.linear.x of 10 m/s is probably too fast. I guess the algorithm is throttling the joint speeds because it is too fast, and that causes some inaccuracy.
> 
> Is there a chance the robot is starting near a singularity? Accuracy won't be great near singularities.
> 
> What kind of robot is it? How many degrees of freedom? Can you post the servo_settings.yaml file you're using?

AndyZe, 2020-09-24:
`Just to add a question. If I want to move the robot +0.2m in the +x direction, what value should be given to twist.linear.x? I'm a bit confused on how the twist message is converted to position.`

Ah, there's the fundamental problem. Twist is a velocity command. Servo doesn't really handle position commands yet, so there's no easy way to send it to a certain position.

This PR adds position-tracking functionality. I think it might get merged soon. There's an example in `src/cpp_interface_example/pose_tracking_example.cpp`

closed 2020-09-24 by AndyZe

nihalar, 2020-09-24:
> pose_tracking_example.cpp

Couldn'nt find the pose_tracking_example.cpp file, could you please share the link?

AndyZe, 2020-09-24:
https://github.com/ros-planning/moveit/blob/30abf49533df905c8fc99888bec77828f9ffe2f6/moveit_ros/moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp

nihalar, 2020-09-25:
Hi, thanks for the share. 
I tried moving 10cm in the x direction using the pose tracker. The distance moved was sometimes off by around 1cm. Any reason for it? Is there anything I can do to reduce the error?

nihalar, 2020-09-25:
> https://github.com/ros-planning/moveit/blob/30abf49533df905c8fc99888bec77828f9ffe2f6/moveit_ros/moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp

Hi, thanks for the share.
I tried moving 10cm in the x direction using the pose tracker. The distance moved was sometimes off by around 1cm. Any reason for it? Is there anything I can do to reduce the error?