moveit/moveit #2322: IKfast plugin not working

AnlageM, 2020-09-24:
### Description

Hi, I installed the new version of Moveit for Noetic, but can't make IKFast plugin work.
Running the package as it is I get this error:

```
Traceback (most recent call last):
  File "/home/usr/catkin_ws/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/create_ikfast_moveit_plugin.py", line 50, in <module>
    import yaml
ImportError: No module named yaml
```

I imagined it was due to the compatibility with python3 so I tried changing the shebang line of the scritps `create_ikfast_moveit_plugin.py `and `round_collada_numbers.py` to `#!/usr/bin/env python3` and now I get:

```
Traceback (most recent call last):
  File "/home/usr/catkin_ws/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/create_ikfast_moveit_plugin.py", line 399, in <module>
    main()
  File "/home/usr/catkin_ws/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/create_ikfast_moveit_plugin.py", line 390, in main
    create_ikfast_package(args)
  File "/home/usr/catkin_ws/src/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts/create_ikfast_moveit_plugin.py", line 186, in create_ikfast_package
    etree.ElementTree(root).write(f, xml_declaration=True, pretty_print=True, encoding="UTF-8")
  File "src/lxml/etree.pyx", line 2057, in lxml.etree._ElementTree.write
  File "src/lxml/serializer.pxi", line 758, in lxml.etree._tofilelike
  File "src/lxml/etree.pyx", line 318, in lxml.etree._ExceptionContext._raise_if_stored
  File "src/lxml/serializer.pxi", line 682, in lxml.etree._FilelikeWriter.write
TypeError: write() argument must be str, not bytes
```

Which is the correct way to overcome these errors?

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source build, branch: noetic-devel

rhaschke, 2020-09-24:
Testing IKFast plugin generation is part of our Travis CI jobs and [does work on Noetic](https://travis-ci.com/github/ros-planning/moveit/jobs/386874924). So, the issue is probably with your specific environment.
`ImportError: No module named yaml`: This indicates that you are missing the package `python3-yaml` - and we didn't declare this dependency.

What is your default python version? For ROS Noetic it should be python3 now.

AnlageM, 2020-09-24:
Thanks for your help: it seems the problem is not the yaml package missing, because it was already installed and at its newest version. Python version is 3.8.2. If you can provide any other suggestion, they are really wlecome.

Suggestion about alternatives to IKfast working in Noetic, for generating a file that describes the inverse kinematics of a robot? I need it for teleoperation purposes so the output script should be quite fast in processing pose data. And how can I get updates on your work about ikfast on Travis CI?

rhaschke, 2020-09-24:
Do you have both, python2 and python3 installed? What is the default in your ROS environment?
#2323 just declares the missing package dependency, this won't directly affect the execution of the script and you stated that python3-yaml is installed already.
I don't have further ideas then. Sorry.

closed 2020-09-24 by v4hn

AnlageM, 2020-09-25:
Yes I think I have both, may this affect? The default one is 3.8.2.

rhaschke, 2020-09-25:
Having python3-yaml installed, you should not see `ImportError: No module named yaml`. So, something is screwed in your config.
I suggest checking, which python is actually executed by create_ikfast_moveit_plugin.py. Do you have conda or virutalenv versions as well? Next, check your python system path. Are the correct paths in there? I.e. what are the results of the following cmds:
```
python3 -c "import yaml"
python3 -c "import sys; print(sys.executable); print(sys.path)"
```

AnlageM, 2020-09-25:
I decided to remove Python2 from the PC so now I have only Python3. It now behaves exactly as it was when I forced the shebang line, this means removing Python 2 I overcome the yaml error but **still get the `write() argument` error** I reported when I raised the issue. 

What I get from the command you suggested is (please note I get no output from the first command.):
```
$ python3 -c "import yaml"
$ python3 -c "import sys; print(sys.executable); print(sys.path)"
/usr/bin/python3
['', '/opt/ros/noetic/lib/python3/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/<myPCname>/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
```
Don't know if it can be useful but after sourcing my ws_IKtrial I get, with the same commands:
```
$ python3 -c "import yaml"
$ python3 -c "import sys; print(sys.executable); print(sys.path)"
/usr/bin/python3
['', '/home/<myPCname>/ws_IKtrial/devel_isolated/srdfdom/lib/python3/dist-packages', '/home/<myPCname>/ws_IKtrial/devel_isolated/franka_example_controllers/lib/python3/dist-packages', '/home/<myPCname>/ws_IKtrial/devel_isolated/franka_msgs/lib/python3/dist-packages', '/home/<myPCname>/ws_IKtrial/devel_isolated/franka_gripper/lib/python3/dist-packages', '/opt/ros/noetic/lib/python3/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/<myPCname>/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
```

rhaschke, 2020-09-25:
Probably you have a broken (or too new) version of `lxml` installed in `/home/<myPCname>/.local/lib/python3.8/site-packages`.
Could you please remove this folder, at least temporarily, and see what changes? (Note, pip installs packages into this folder.)
Also, do you have the system package `python3-lxml` installed?

AnlageM, 2020-09-28:
Yes, I have `python3-lxml` installed. Unfortunately, the same `write() `error comes out also after removing the `site-packages` folder you mentioned.
How can I downgrade the lxml version?
Does the fact that I am following [this tutorial ](https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html) helps in hypothesizing which is my problem?
Thanks for your help

AnlageM, 2020-09-29:
Please can you tell me exactly which versions of all the involved packages you used to test that IKfast works? Maybe reproducing your environment I can make it work too... 
Moreover, may [this ](https://github.com/PixarAnimationStudios/OpenTimelineIO/pull/646) be related to my issue? It states that the `write() argument `error is due to the encoding, and it seems to be a consequence of the switching between python2 and 3. Known that, and given that the default for Noetic is python3, any suggestion for me on how to solve?

rhaschke, 2020-09-29:
Sorry, I was misled by our Travis CI. But Travis was testing against Melodic, not Noetic. I can reproduce the issue and will have a look.

rhaschke, 2020-09-29:
I filed a PR that hopefully fixes your issue: #2327. Please review and test it.

PR #2323 (2020-09-24): fix some more python3 issues
Fixes #2322:
* `ikfast` was missing some python dependencies
* [python scripts should be installed via `catkin_install_python()`](http://wiki.ros.org/UsingPython3/SourceCodeChanges#Changing_shebangs)
PR #2327 (2020-09-30): Fix various issues with Noetic build
Travis wasn't yet testing `master` branch against Noetic. Switching to Noetic revealed several open issues, which are tackled by this PR:
- Our ikfast unittests were failing and we didn't notice: https://travis-ci.com/github/ros-planning/moveit/jobs/391217949.
- `lxml` API has changed, giving rise to #2322. This PR fixes #2322.
- There were several new warnings, particularly by newer clang compiler.