moveit/moveit #2326: can't find collision_world_fcl.h, thank you.

bbhgyyy87, 2020-09-29:
### Description
can't find collision_world_fcl.h   in the package,    when building, as follows.
(unfortunately , i also find currently there is not this file in this package. whether it had been deleted? thank you.)
Thank you！
yang.

*** Error description:
[ 63%] Building CXX object 
C:\moveit_ws\src\moveit_tutorials\doc\visualizing_collisions\src\visualizing_collisions_tutorial.cpp(47): fatal error C1083: Cannot open include file: 'moveit/collision_detection_fcl/**_collision_world_fcl.h': No such file or directory_**

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. window 10

v4hn, 2020-09-29:
You build incompatible versions of the `moveit` and the `moveit_tutorials` repository.
Either pick `master` or `melodic-devel` for both.

closed 2020-09-29 by v4hn

bbhgyyy87, 2020-09-29:
I have find the reason.  Because my **moveit_tutorials** package is not the latest one. 
So the solution is that i should update **moveit_tutorials** 

Sorry, this is not a bug.   I just install according to the “https://ms-iot.github.io/ROSOnWindows/GettingStarted/UsingROSonWindows.html” 

BR
yang

..\src\moveit_tutorials\doc\visualizing_collisions\src\visualizing_collisions_tutorial.cpp