moveit/moveit #2329: Setup Assistant not visualizing URDF in start screen

JeroenDM, 2020-09-30:
### Description
The MoveIt Setup Assistant does not visualize a loaded URDF by default anymore. I seem to remember it did a couple of months ago. I went through the latest changes in the source code on Github, but cannot find the cause for this.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary install version `1.0.6-1bionic.20200828.222638`

### Steps to reproduce
Launch the setup assistant in a sourced workspace with a robot description package (so it can find the meshes).
```bash
roslaunch moveit_setup_assistant setup_assistant.launch
```
Click "Create new MoveIt configuration package" and select the urdf file of the robot.

### Expected behaviour
I would expect to see a visual of the robot in the right pane.

### Actual behaviour
The pane is empty, like this:
[image]

If you go to the "Robot Poses" tap and click "show default pose", you do see the robot:
[image]

### Backtrace or Console output
[Console output](https://gist.github.com/JeroenDM/39edf650d98788513a5703f471b0c643).

rhaschke, 2020-09-30:
This was already fixed in #2245, but not yet released.

JeroenDM, 2020-09-30:
Ah, ok. Thank you!

closed 2020-09-30 by JeroenDM