moveit/moveit #2330: "This project is licensed under the terms of the LGPL v2.1 or later" ?

gavanderhoorn, 2020-09-30:
Not sure what's happening here (from [this Fossa scan](https://app.fossa.com/projects/git%2Bgithub.com%2Fros-planning%2Fmoveit/refs/branch/master/6eeee4b7a46c80d95323a929d3570e8690d0f47b/preview), part of the checks run for 6eeee4b):

[image]

Note the:

> ### Using moveit as a dependency
> 
> This project is licensed under the terms of the GNU Lesser General Public License v2.1 or later and contains 3 other licenses.

Not sure we want GPL to be identified as the main license -- even if there are "3 other licenses" identified.

gavanderhoorn, 2020-09-30:
Oh and I tagged this `ci` only because the Fossa scan is part of the CI config.

Not because there is something obviously broken in the CI setup.

tylerjw, 2020-10-14:
These are the two offending files:

```
moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/chainiksolver_vel_mimic_svd.hpp
moveit_kinematics/kdl_kinematics_plugin/src/chainiksolver_vel_mimic_svd.cpp
```

What should we do about this?  How broken would it be to remove these files from the repo?  Are they avalible elsewhere or should we factor them out into a separate repo we can link against?

tylerjw, 2020-10-28:
Maybe a good solution for this would be to move the kdl plugin outside of the moveit repo?  Thoughts @gavanderhoorn?

gavanderhoorn, 2020-11-09:
There are two sides to this:

 1. whatever conclusion Fossa reaches and displays
 1. the legal implications of having (L)GPL licensed sources hosted in a repository

The first is essentially a marketing problem: according to the license scan/test MoveIt itself runs *on its own repository* (ie: the Fossa test), it's (L)GPL licensed. Just seeing those words may scare off potential users/contributors.

The second is not necessarily a problem, depending on whether the license is actually the LGPL, and whether the code in question and MoveIt fulfil the requirements of that license.

Afaik, the mentioned files are copies of KDL provided ones. Reason for copying them:

https://github.com/ros-planning/moveit/blob/09afa50b40ac8cb8644ba1a627bb7643137959d3/moveit_kinematics/kdl_kinematics_plugin/src/chainiksolver_vel_mimic_svd.cpp#L22-L24

v4hn, 2021-03-24:
> Maybe a good solution for this would be to move the kdl plugin outside of the moveit repo?

Actually, the repository advertises BSDv3 as its overall license, which is apparently wrong for the KDL plugin.
That's frustrating because it took a lot of time to migrate all the kinematics plugins *into* the same package(`moveit_kinematics`) in this repository.
I believe @davetcoleman and PickNik have an incentive to resolve this - probably by splitting off the code files from this repository and making it clear that the KDL plugin is a small *optional* (though pretty much default!) plugin for the whole MoveIt system.

tylerjw, 2021-03-24:
This is a bit of a gray area because GPL doesn't infect your code if it comes through a dynamic linker.  This is atleast the workaround that's sort of become the industry standard for using GPL code without it infecting the repo.

However, it seems that Stallman and the people behind GPL don't think the dynamic linker is a way of avoiding copy-left: https://softwareengineering.stackexchange.com/questions/167773/how-does-the-gpl-static-vs-dynamic-linking-rule-apply-to-interpreted-languages

Not having this in the repo would probably be enough to confuse some layers and that might be the most we'd ever want (maybe use a git submodule to fool text searchers).  If we wanted to be 100% certain we haven't become GPL we'd have to replace this code.

mikeferguson, 2021-03-24:
> This is a bit of a gray area because GPL doesn't infect your code if it comes through a dynamic linker. This is atleast the workaround that's sort of become the industry standard for using GPL code without it infecting the repo.

As a clarification - the KDL library (which we've copied this code from) is LGPL not GPL (and that license is very clear that linking to the library does not make your code LGPL/GPL).

mikeferguson, 2021-03-24:
A further note - you can probably move just the customized KDL mimic solver into a separate repo/package - and then moveit_kinematics can depend on that new package and still have the actual moveit plugins all exist in the main single moveit repo.

gavanderhoorn, 2021-03-24:
I did not open this issue because I feared copyleft incursions into MoveIt.

The issue is that the Fossa scan lists the LGPL as the first license.

tylerjw, 2021-03-24:
> A further note - you can probably move just the customized KDL mimic solver into a separate repo/package - and then moveit_kinematics can depend on that new package and still have the actual moveit plugins all exist in the main single moveit repo.

I misunderstood because I obviously didn't look at it close enough.  Thank-you for clarifying.  Moving into another repo seems like a reasonable solution.

v4hn, 2021-03-24:
> The issue is that the Fossa scan lists the LGPL as the first license.

Yes, I would expect that they list the most restrictive license they find in the sources first. No matter how relevant that file might be.

> you can probably move just the customized KDL mimic solver into a separate repo/package - and then moveit_kinematics can depend on that new package and still have the actual moveit plugins all exist in the main single moveit repo.

Sounds like a reasonable option.