moveit/moveit #2332: setup assistant collision detection issues

tylerjw, 2020-09-30:
### Description

I'm not sure if this is an issue with the franka description or the setup assistant.  These issues were discovered when trying to use the setup assistant according to the tutorial instructions for creating an update to `panda_moveit_config` based on the latest version of MoveIt master on Noetic in preparation for a Noetic release of the tutorials.  

Here is a simple launch file to load a play with the panda in rviz I used for creating some of these images:

```xml
<launch>
  <arg name="model" default="$(find franka_description)/robots/panda_arm_hand.urdf.xacro"/>
  <arg name="gui" default="true" />
  <arg name="rvizconfig" default="$(find franka_visualization)/launch/franka_visualization.rviz" />
  <arg name="add_all_collision_objects" default="false" />

  <param name="robot_description" command="$(find xacro)/xacro $(arg model) add_all_collision_objects:=$(arg add_all_collision_objects)" />

  <node if="$(arg gui)" name="joint_state_publisher" pkg="joint_state_publisher_gui" type="joint_state_publisher_gui" />

  <node unless="$(arg gui)" name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />

  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(arg rvizconfig)" required="true" />
</launch>
```

Before you ask, here is the collision matrix, generated using defaults:
[image]

## Default pose (these should be collisions)

Here are a couple images showing the default pose in the setup assistant, you can see that it should be in collision in both positions (I included an image from rviz showing the collision geometry):
[image]
[image]
[image]

### Collision geometry not protecting cable on eff in model

This seems like an issue with the model:
[image]
[image]

## Disappearing robot model after adding the virtual joint

This is another issue with the setup assistant.  I documented it here: https://github.com/ros-planning/moveit_tutorials/issues/533

rhaschke, 2020-10-01:
Hi Tyler,
I'm not sure yet what issue you are exactly complaining about. Yes, the collision geometry of franka_description has changed with the new release 0.7 (which is only available in Noetic). We know that since https://github.com/ros-planning/panda_moveit_config/pull/35#pullrequestreview-390715967.

Obviously, we need to adapt several poses in panda_moveit_config, which are used in the tutorials.
Other related issues and comments are:
https://github.com/ros-planning/panda_moveit_config/issues/71
https://github.com/ros-planning/panda_moveit_config/pull/35#issuecomment-671341548
https://github.com/ros-planning/panda_moveit_config/pull/35#issuecomment-645989992

tylerjw, 2020-10-01:
I'm mostly complaining about the "robot is in collision" warning not making much sense to me.  The extended pose for example is hard to achieve a similar pose.  I didn't realize this was a known issue.

tylerjw, 2020-10-01:
For example that "default pose" is not in collision and you can twist joint7 a bit without it becoming in collision.  It does look like from the collision model that no change in joint7 should get it out of collision.

v4hn, 2020-10-01:
I guess #780 is related.

I don't think we always want to show the colliding collision geometries in these cases though.