moveit/moveit #2339: [moveit_setup_assistant] Define Planning Groups vertical space

tylerjw, 2020-10-02:
Here is a minor UI tweak that would be nice.  The `Default Planning Groups` needs to be adjusted to have more vertical space.

[image]

rhaschke, 2020-10-02:
I noticed this issue as well, but I'm not sure why the minimal widget size is not correctly propagated up the window hierarchy.

felixvd, 2020-10-02:
I wouldn't mind taking a look if someone can tell me how to open this in QtCreator. I don't see a .ui file.

rhaschke, 2020-10-02:
Unfortunately, Dave didn't use .ui files when programming this, but manually coded all the gui elements.
It's a long-standing task to migrate to .ui files...

closed 2020-10-06 by rhaschke

lyh458, 2021-07-14:
Hi there, thanks for the good issue and commits.
But I still have the similar problem under high DPI screen, while everything is good under a normal DPI screen.
Here is my device info:
- system: Ubuntu 18.04, ROS Melodic, and Moveit! is up-to-date
- screen info: resolution 3840*2160, scale 200%

---

- Some items are overlapping in the first default windows, and the font size is small, unfriendly.
[image]

- The **Robot Pose** window: totally can't see and change the value of each joints
[image]

PR #2349 (2020-10-06): MSA: Fix layout
Hunting down #2339, I came across several improvements of the (manual :cry:) GUI creation.
However, I didn't yet find the underlying reason why Qt cannot correctly layout the widget - the minimal size is incorrectly computed. Finally, I just increase the minimum height of the main widget...
Fixes #2339 for now.