moveit/moveit #2341: MoveIt Noetic Release (1.1.1)

tylerjw, 2020-10-02:
# MoveIt Noetic Release (1.1.1)

[_moveit.ros.org release documentation_](https://moveit.ros.org/documentation/contributing/releases.html)

### Checklist
- [ ] Sync noetic-devel with master
- [ ] Run ROS buildfarm prerelease test
- [ ] Update changelogs
- [ ] Bloom
- [ ] Write discourse post

rhaschke, 2020-10-03:
@tylerjw, for when do you plan the next release? What are open PR, we should tackle beforehand?

tylerjw, 2020-10-03:
This is the first release on the new schedule (linked above in the moveit.ros.org page).  The plan for now is to release every 6 weeks.  Specifically the date I'm hoping to run bloom is next Friday (the 9th).

As to new features I want to work on, getting the panda config updated and the tutorials released for noetic is my goal for this release.  Along with that I'd like to release each of the packages we have that have changes since the last release.

tylerjw, 2020-10-03:
> @tylerjw, for when do you plan the next release? What are open PR, we should tackle beforehand?

I think we should if possible just release the state of master yesterday on the `noetic-devel` branch (and whatever the noetic release branch is on the other repos).  Then we can continue to review and merge changes into master like we normally do and release those changes in the next release in 6 weeks.

The work I'm trying to do on `panda_moveit_config` and the tutorials should not block this release although it would be nice to do it at roughly the same time because then people who want to use Noetic from the debians would have updated tutorials that all work.

tylerjw, 2020-10-05:
I want to point out that I don't plan on blocking this (or any of the other releases I'm hoping to do on Friday) with the work on the tutorials.  I am doing that hoping to finish it in time but it is a lot more work than I thought it would be (so it probably won't be included in this release).  Thank you @v4hn for educating me about some of the config files, @rhaschke for fixing an issue in MSA, @AndyZe for reviewing changelogs for moveit, and others for helping me find issues, and get small fixes merged this last weekend.  I won't have a ton of time for iterating on this during the week so I'll focus on the chagelogs and anything that will actually block the release on Friday.   Any help reviewing/merging those boring PRs would be greatly appreciated.

rhaschke, 2020-10-09:
@tylerjw, IMO this release is currently blocked by https://github.com/ros/rosdistro/pull/26706.
We agreed to upgrade to FCL 0.6 [here](https://github.com/ros/rosdistro/issues/26527#issuecomment-702013343) to handle some open issues in the current release.
If you insist on releasing today, please sync master and noetic-devel as discussed in https://github.com/ros-planning/moveit/pull/2342#issuecomment-703893063.
By the way, why are there two open PRs w.r.t. changelogs: #2342 and #2343?

tylerjw, 2020-10-09:
> By the way, why are there two open PRs w.r.t. changelogs: #2342 and #2343?

Because that was my attempt to simplify adding changelogs to both master and noetic (they come from the same origin branch).  I realize now that submitting changelogs into master and syncing noetic-devel would do the same thing and be more straight forward as there isn't a chance of the history being a different order.  I'll close the one to noetic-devel.

tylerjw, 2020-10-09:
@rhaschke how quickly do you think the FCL rosdistro PR will be resolved?  Any chance we can convince them to merge that today?  I planned on pushing out a release today because I was given the time to do this every 6 weeks on this schedule.  My hope was that the regular releases would be an improvement over the current process as they'd reduce the amount of work for each release and get the work people are putting into PRs into debians faster.

gavanderhoorn, 2020-10-09:
This is exactly the reason why "releasing" is not a simple process of just pushing some buttons.

The goal to release more often is laudable, but especially on a new platform like Focal it's just non-trivial.

I hope we all appreciate the *enormous* amount of work @rhaschke is taking on every release ("let's create a `ros-noetic-fcl`"), and the maintainers before him who have done this.

rhaschke, 2020-10-09:
As you say, @gavanderhoorn, it's a lot of effort to ping people before a release and to check that everything worked well after a release. The colleagues at OSRF are busy with way more projects and I don't want to put pressure on them because I don't like to have pressure on me as well. However, I will do a friendly ping.

tylerjw, 2020-10-09:
Thank you for the ping.  It looks like it will be done.  Do you think we should wait on releasing this repo until then?  I've done most of the trivial smaller repo releases today.  I'll wait to hear from you @rhaschke.  Would blooming moveit before the FCL package change is accepted into rosdistro put us in a worse state than we are now on Noetic?

rhaschke, 2020-10-10:
fcl is released: https://github.com/ros/rosdistro/pull/26706. I created the PR to switch to this version: https://github.com/ros-planning/moveit/pull/2359. With this, I think, you (@tylerjw) can proceed with prerelease tests. Ensure that you don't see linker warnings about conflicting fcl / octomap libs anymore (particularly in Debian Buster).

rhaschke, 2020-10-10:
There is also an issue with `ros-noetic-spacenav-node` on Buster, which should be fixed before a new release: http://build.ros.org/job/Ndev_db__moveit__debian_buster_amd64/14/display/redirect

tylerjw, 2020-10-11:
This is the spacenav issue (from the link you posted):

```
09:39:12 Starting pkgProblemResolver with broken count: 1
09:39:12 Starting 2 pkgProblemResolver with broken count: 1
09:39:12 Investigating (0) ros-noetic-spacenav-node:amd64 < none -> 1.14.0-1buster.20200929.232436 @un puN Ib >
09:39:12 Broken ros-noetic-spacenav-node:amd64 Depends on libspnav0:amd64 < none | 0.2.3-1 @un uH > (>= 0.2.2)
09:39:12   Considering libspnav0:amd64 0 as a solution to ros-noetic-spacenav-node:amd64 9999
09:39:12   Re-Instated libspnav0:amd64
09:39:12 Broken ros-noetic-spacenav-node:amd64 Depends on libspnav-dev:amd64 < none | 0.2.3-1 @un uH >
09:39:12   Considering libspnav-dev:amd64 1 as a solution to ros-noetic-spacenav-node:amd64 9999
09:39:12   Re-Instated libspnav-dev:amd64
09:39:12 Broken ros-noetic-spacenav-node:amd64 Depends on spacenavd:amd64 < none @un H >
09:39:12 Done
09:39:12 python3 is already the newest version (3.7.3-1).
09:39:12 python3-yaml is already the newest version (3.13-2).
09:39:12 Some packages could not be installed. This may mean that you have
09:39:12 requested an impossible situation or if you are using the unstable
09:39:12 distribution that some required packages have not yet been created
09:39:12 or been moved out of Incoming.
09:39:12 The following information may help to resolve the situation:
09:39:12 
09:39:12 The following packages have unmet dependencies:
09:39:12  ros-noetic-spacenav-node : Depends: spacenavd but it is not installable
09:39:12 E: Unable to correct problems, you have held broken packages.
09:39:12 Invocation failed without any known error condition, printing all lines to debug known error detection:
09:39:12   1 'Reading package lists...'
09:39:12   2 'Building dependency tree...'
09:39:12   3 'Reading state information...'
09:39:12   4 'Starting pkgProblemResolver with broken count: 1'
09:39:12   5 'Starting 2 pkgProblemResolver with broken count: 1'
09:39:12   6 'Investigating (0) ros-noetic-spacenav-node:amd64 < none -> 1.14.0-1buster.20200929.232436 @un puN Ib >'
09:39:12   7 'Broken ros-noetic-spacenav-node:amd64 Depends on libspnav0:amd64 < none | 0.2.3-1 @un uH > (>= 0.2.2)'
09:39:12   8 '  Considering libspnav0:amd64 0 as a solution to ros-noetic-spacenav-node:amd64 9999'
09:39:12   9 '  Re-Instated libspnav0:amd64'
09:39:12   10 'Broken ros-noetic-spacenav-node:amd64 Depends on libspnav-dev:amd64 < none | 0.2.3-1 @un uH >'
09:39:12   11 '  Considering libspnav-dev:amd64 1 as a solution to ros-noetic-spacenav-node:amd64 9999'
09:39:12   12 '  Re-Instated libspnav-dev:amd64'
09:39:12   13 'Broken ros-noetic-spacenav-node:amd64 Depends on spacenavd:amd64 < none @un H >'
09:39:12   14 'Done'
09:39:12   15 'python3 is already the newest version (3.7.3-1).'
09:39:12   16 'python3-yaml is already the newest version (3.13-2).'
09:39:12   17 'Some packages could not be installed. This may mean that you have'
09:39:12   18 'requested an impossible situation or if you are using the unstable'
09:39:12   19 'distribution that some required packages have not yet been created'
09:39:12   20 'or been moved out of Incoming.'
09:39:12   21 'The following information may help to resolve the situation:'
09:39:12   22 ''
09:39:12   23 'The following packages have unmet dependencies:'
09:39:12   24 ' ros-noetic-spacenav-node : Depends: spacenavd but it is not installable'
09:39:12   25 'E: Unable to correct problems, you have held broken packages.'
```

tylerjw, 2020-10-11:
The problem is the daemon for spacenav: https://packages.debian.org/sid/spacenavd is not in buster.

gavanderhoorn, 2020-10-11:
On recent kernels the daemon should not be necessary. The spacenav will register as a regular HID joystick.

tylerjw, 2020-10-11:
Then maybe the ros project should remove the daemon from the requirements for Noetic?

rhaschke, 2020-10-11:
Thanks for this hint @gavanderhoorn. @tylerjw, yes I think you should try to run the demos without this daemon installed.
If that works, we can go with https://github.com/ros-drivers/joystick_drivers/pull/186.

tylerjw, 2020-10-11:
Unrelated to the spacenav, I ran the prerelease testing (Noetic on master branch) locally and it passed (ran on 18.04 as the `python3-ros-buildfarm` isn't available on Noetic, I believe it does the testing in a docker image though).

gavanderhoorn, 2020-10-12:
@tylerjw wrote:
> Then maybe the ros project should remove the daemon from the requirements for Noetic?

"the ros project" -> that's an odd thing to say?

tylerjw, 2020-10-12:
@gavanderhoorn :disappointed: I edited your comment, I meant to click "quote".  I didn't know it was possible to edit someone else's comment.  I'm sorry.

tylerjw, 2020-10-13:
I've tested this Buster and 20.04 and believe the current state of the master is good.  I merged my changelogs into master.  Here is the PR for sync into noetic-devel: https://github.com/ros-planning/moveit/pull/2365

tylerjw, 2020-10-13:
:tada: https://github.com/ros/rosdistro/pull/26938

tylerjw, 2020-10-13:
Closing this for now, it has been bloomed.

closed 2020-10-13 by tylerjw