moveit/moveit #2344: [moveit_setup_assistant] missing finter joint

tylerjw, 2020-10-03:
### Description

When configuring default poses, if I try to add one for the end effector the last joint doesn't have a slider.  See the images.  Something else I noticed is that if I try to edit the joints for the arm the last joint is a fixed joint and doesn't show up in the list (that's probably fine) but might be why we didn't notice this before.

[image]
[image]

tylerjw, 2020-10-03:
Looking at the code that creates these sliders I noticed that it skips mimic joints.  I went and looked at the xacro and it doesn't look like they are mimic joints (from franka_description):

```xml
    <joint name="${ns}_finger_joint1" type="prismatic">
      <parent link="${ns}_hand"/>
      <child link="${ns}_leftfinger"/>
      <origin xyz="0 0 0.0584" rpy="0 0 0"/>
      <axis xyz="0 1 0"/>
      <limit effort="20" lower="0.0" upper="0.04" velocity="0.2"/>
    </joint>
    <joint name="${ns}_finger_joint2" type="prismatic">
      <parent link="${ns}_hand"/>
      <child link="${ns}_rightfinger"/>
      <origin xyz="0 0 0.0584" rpy="0 0 0"/>
      <axis xyz="0 -1 0"/>
      <limit effort="20" lower="0.0" upper="0.04" velocity="0.2"/>
      <mimic joint="${ns}_finger_joint1" />
    </joint>
```

tylerjw, 2020-10-03:
At this point I'm just confused, I used my old friend `gdb` to look at what is happening and it looks like the second finger is a minic joint.  That being said, if it is a minic joint the position of the second finger doesn't seem to be updated as you move the first finger.  That might just be a small cosmetic detail but it might confuse others.

```gdb
(gdb) print *joint_models_[2]
$10 = {
  _vptr.JointModel = 0x7ffff6b563f0 <vtable for moveit::core::PrismaticJointModel+16>,
  name_ = "panda_finger_joint2",
  type_ = moveit::core::JointModel::PRISMATIC,
  local_variable_names_ = std::vector of length 0, capacity 0,
  variable_names_ = std::vector of length 1, capacity 1 = {"panda_finger_joint2"},
  variable_bounds_ = std::vector of length 1, capacity 1 = {{
      min_position_ = 0,
      max_position_ = 0.040000000000000001,
      position_bounded_ = true,
      min_velocity_ = -0.20000000000000001,
      max_velocity_ = 0.20000000000000001,
      velocity_bounded_ = true,
      min_acceleration_ = 0,
      max_acceleration_ = 0,
      acceleration_bounded_ = false
    }},
  variable_bounds_msg_ = std::vector of length 1, capacity 1 = {{
      joint_name = "panda_finger_joint2",
      has_position_limits = 1 '\001',
      min_position = 0,
      max_position = 0.040000000000000001,
      has_velocity_limits = 1 '\001',
      max_velocity = 0.20000000000000001,
      has_acceleration_limits = 0 '\000',
      max_acceleration = 0
    }},
  variable_index_map_ = std::map with 1 element = {
    ["panda_finger_joint2"] = 0
  },
  parent_link_model_ = 0x5555565d9bc0,
  child_link_model_ = 0x5555565db370,
  mimic_ = 0x555555e023f0,
  mimic_factor_ = 1,
  mimic_offset_ = 0,
  mimic_requests_ = std::vector of length 0, capacity 0,
  descendant_link_models_ = std::vector of length 1, capacity 1 = {0x5555565db370},
  descendant_joint_models_ = std::vector of length 0, capacity 0,
  non_fixed_descendant_joint_models_ = std::vector of length 0, capacity 0,
  passive_ = false,
  distance_factor_ = 1,
  first_variable_index_ = 8,
  joint_index_ = 11
}
(gdb) print *joint_models_[1]
$11 = {
  _vptr.JointModel = 0x7ffff6b563f0 <vtable for moveit::core::PrismaticJointModel+16>,
  name_ = "panda_finger_joint1",
  type_ = moveit::core::JointModel::PRISMATIC,
  local_variable_names_ = std::vector of length 0, capacity 0,
  variable_names_ = std::vector of length 1, capacity 1 = {"panda_finger_joint1"},
  variable_bounds_ = std::vector of length 1, capacity 1 = {{
      min_position_ = 0,
      max_position_ = 0.040000000000000001,
      position_bounded_ = true,
      min_velocity_ = -0.20000000000000001,
      max_velocity_ = 0.20000000000000001,
      velocity_bounded_ = true,
      min_acceleration_ = 0,
      max_acceleration_ = 0,
      acceleration_bounded_ = false
    }},
  variable_bounds_msg_ = std::vector of length 1, capacity 1 = {{
      joint_name = "panda_finger_joint1",
      has_position_limits = 1 '\001',
      min_position = 0,
      max_position = 0.040000000000000001,
      has_velocity_limits = 1 '\001',
      max_velocity = 0.20000000000000001,
      has_acceleration_limits = 0 '\000',
      max_acceleration = 0
    }},
  variable_index_map_ = std::map with 1 element = {
    ["panda_finger_joint1"] = 0
  },
  parent_link_model_ = 0x5555565d9bc0,
  child_link_model_ = 0x5555565da420,
  mimic_ = 0x0,
  mimic_factor_ = 1,
  mimic_offset_ = 0,
  mimic_requests_ = std::vector of length 1, capacity 1 = {0x555555e01800},
  descendant_link_models_ = std::vector of length 2, capacity 2 = {0x5555565da420, 0x5555565db370},
  descendant_joint_models_ = std::vector of length 1, capacity 1 = {0x555555e01800},
  non_fixed_descendant_joint_models_ = std::vector of length 1, capacity 1 = {0x555555e01800},
  passive_ = false,
  distance_factor_ = 1,
  first_variable_index_ = 7,
  joint_index_ = 10
}
```

felixvd, 2020-10-04:
I remember that mimic joints are not being updated in that view. There’s an
issue about it I believe. It would certainly be smoother if they were.

rhaschke, 2020-10-05:
@tylerjw: The urdf snippet you provided in https://github.com/ros-planning/moveit/issues/2344#issuecomment-703159399 clearly states that `joint2` mimics `joint1`. So, joint2 there shouldn't be a slider for joint2.
I will handle the problem, the joints not updating correctly...

PR #2348 (2020-10-05): MSA: fix pose definitions
Fix #2344 and other issues:
- The joint values of the initial robot pose (shown in rviz window after loading) were not taking into account for saving poses.
- mimic joints were written to SRDF (with wrong values), overriding auto-updates of mimic joints (causing #2344)
- Cleaning up code to use a RobotState for robot poses (instead of a separate joint_states_map_)
closed 2020-10-05 by v4hn