moveit/moveit #2345: Rviz panel of movit_setup_assistant not loading.

OmkarKabadagi5823, 2020-10-04:
### Description

I am new to ros and trying to learn moveit. In the moveit setup assistant, I tried to load the panda_arm_hand.urdf.xacro part of franka_description package. It shows that the file has been loaded successfully but the rviz panel on the right in the assistant is blank for me. This is for any model I am trying to load. How do I fix this? 

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? Source
* If source, which branch? master

### Expected behaviour
Model should load with rviz panel displaying the model.

### Actual behaviour
Shows model has loaded but rviz panel is blank.

### Backtrace or Console output
omkar@omkar-G3-3579:~$ roslaunch moveit_setup_assistant setup_assistant.launch
... logging to /home/omkar/.ros/log/3be3846c-0639-11eb-b00e-7c2a31effce5/roslaunch-omkar-G3-3579-24182.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://omkar-G3-3579:43983/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [24192]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3be3846c-0639-11eb-b00e-7c2a31effce5
process[rosout-1]: started with pid [24203]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [24207]
Gtk-Message: 17:30:50.118: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 17:31:03.244: GtkDialog mapped without a transient parent. This is discouraged.
[ INFO] [1601812900.576027969]: Loaded panda robot model.
[ INFO] [1601812900.577254490]: Setting Param Server with Robot Description
[ INFO] [1601812900.581133909]: Robot semantic model successfully loaded.
[ INFO] [1601812900.581206762]: Setting Param Server with Robot Semantic Description
[ INFO] [1601812900.597057688]: Loading robot model 'panda'...
[ INFO] [1601812900.597140583]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1601812900.789327520]: Stereo is NOT SUPPORTED
[ INFO] [1601812900.789456194]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1601812900.882563402]: Loading robot model 'panda'...
[ INFO] [1601812900.882623932]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1601812901.351435124]: Loading Setup Assistant Complete
^C[moveit_setup_assistant-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[image]

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2020-10-04:
This is a duplicate of #2286. See fix there.

closed 2020-10-04 by rhaschke

OmkarKabadagi5823, 2020-10-04:
So do I need to uninstall the existing package or can I build the source over the existing?

rhaschke, 2020-10-05:
Just build the source package. It will shadow the existing binary package.

OmkarKabadagi5823, 2020-10-05:
I don't know if I should post this here but, my laptop is crashing every time while building moveit.

Steps followed:
cd ~/catkin_ws/src
git clone -b melodic-devel https://github.com/ros-planning/moveit.git
cd ~/catkin_ws
catkin_make -DCMAKE_BUILD_TYPE=Release

Issue:
My laptop hangs when compilation reaches about 90% and recompiling after a manual shutdown still doesn't give any progress.

JafarAbdi, 2020-10-05:
> Issue:
> My laptop hangs when compilation reaches about 90% and recompiling after a manual shutdown still doesn't give any progress.

This probably means that your laptop out of RAM, try increasing your swap space

rhaschke, 2020-10-05:
First try to limit the number of parallel build jobs:
`catkin build --jobs 1`

Heavy swapping isn't a good idea...

OmkarKabadagi5823, 2020-10-05:
Thank you @rhaschke, sir, you made my day. I have been trying to run moveit since the last 4-5 days and finally, I built the moveit from the source by limiting the ram usage and the models are also now loading properly in rviz. You have been quite a help.

rhaschke, 2020-10-05:
You are welcome.