moveit/moveit #2346: python planning scene interface incorrectly handles namespacing

PeterMitrano, 2020-10-05:
### Description

Naive concatenating of strings breaks ros namespacing in `planning_scene_interface.py`. Just doing `ns + '/myname'` doesn't work because if `ns=''` then you get a leading slash which prevents ROS_NAMESPACE & `rospy.resolve_name` from working.

I can submit a PR if you want, although this should be an extremely simple fix:

something like this would work
```  
def namespace(ns, name):
    if ns != '':
        return ns + '/' + name
    return name
```

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Binary
* 1.1.0-Alpha

### Steps to reproduce

run the following and observe that it attempts to publish to `/collision_object` and not `/my_robot/collision_object`

```
import os
os.environ['ROS_NAMEPSACE'] = 'my_robot'

from geometry_msgs.msg import PoseStamped
import rospy
import moveit_commander

moveit_commander.roscpp_initialize([])
rospy.init_node("testing_scene")
scene = moveit_commander.PlanningSceneInterface()

for i in range(10):
    scene.add_box("box", PoseStamped())
    rospy.sleep(1.0)

```

### Expected behaviour

It should publish to `/my_robot/collision_object`

### Actual behaviour

it published to `/collision_object`

PeterMitrano, 2020-10-05:
I'll also add that I've found everyone re-implementing this "append namespace" function... maybe we need this in rospy

ex: https://github.com/ros-controls/ros_control/blob/12ef7a4fe11636ccbfcbdd97edcd37ba41dc9c87/rqt_controller_manager/src/rqt_controller_manager/controller_manager.py#L449

v4hn, 2020-10-05:
> maybe we need this in rospy

Turns out this already exists, even though not in rospy:
```
rosgraph.names.ns_join
```
I love how well-structured and accessible the ROS python API is... *sigh*

Please provide a pull-request with a fix against `master`.
We usually honor the contributor who found&fixed the issue in our commit history.

PR #2347 (2020-10-05): fix planning python scene namespacing
### Description

fix planning python scene namespacing and add a test for it. Fixes #2346 

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit/blob/master/MIGRATION.md) notes
- [x] Create tests, which fail without this PR [reference](https://ros-planning.gith [... cut]
closed 2020-10-05 by rhaschke