moveit/moveit #2364: Can't connect to a MoveGroup in the global namespace from a node in a deeper namespace through MoveGroupCommander

marcoesposito1988, 2020-10-11:
### Description

`moveit_commander.MoveGroupCommander` takes a `move_group_namespace` argument, which allows to connect from any namespace to a `MoveGroup` located within a namespace, e.g. `/iiwa/move_group`. However, from a node located in `/my_namespace` I did not manage to connect to a `MoveGroup` in the global namespace, i.e. `/move_group`. 

I tried using `""` and `"/"` as namespaces. I also tried changing `MoveGroupInterfaceWrapper` [here](https://github.com/ros-planning/moveit/blob/3cc363372a6e52d378e6907096979ccff49cdccf/moveit_ros/planning_interface/move_group_interface/src/wrap_python_move_group.cpp#L75) to create a `NodeHandle` without arguments if the namespace is empty or contains only a slash (`(ns.empty() || ns == "/") ? ros::NodeHandle() : ros::NodeHandle(ns)`).

Even with the workarounds above, `waitForAction` throws an error because it can't connect to the action, although the used `NodeHandle` is either global or has a `"/"` namespace.

I don't understand why this happens. Am I missing something?

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source and Binary build
* 1.1.0

### Steps to reproduce
Easier on 18.04 / Melodic, since ViSP is not released for Noetic yet; as a workaround, the dependency of `easy_handeye` on the `visp_*` packages can be disabled.
* check out the `development` branch of `easy_handeye` (i.e. [this commit](https://github.com/IFL-CAMP/easy_handeye/commit/2289f30d05b298f9801536a2709cc5ba9561f7bb))
* check out the `development` branch of `easy_handeye_demo` (i.e. [this commit](https://github.com/marcoesposito1988/easy_handeye_demo/commit/967e2555fb581e5fd9cb817dd1abbe3e66bc8f65))
* `roslaunch easy_handeye_demo calibrate.launch`  to observe expected behavior, which happens when the `MoveGroup` is in a namespace (`/iiwa/move_group` in this case)
* `roslaunch easy_handeye_demo panda_calibrate.launch` to see the bug. In this case, the `MoveGroup` is in the global namespace as usual (`/move_group`), but I couldn't find a way to successfully connect to it

felixvd, 2020-10-13:
Hi Marco,

I am not sure how the `NodeHandle` and its argument works, so I'd take a closer look at that first. It might be easier to debug if you take the `move_group_interface` out of the equation. Can you connect a `SimpleActionClient` in your node to a `move_group` action in the higher namespace? 

You could use [this line](https://github.com/ros-planning/moveit/blob/3cc363372a6e52d378e6907096979ccff49cdccf/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L151) for the action client and then send a [plan request](https://github.com/ros-planning/moveit/blob/3cc363372a6e52d378e6907096979ccff49cdccf/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L746)? Or, if you have another action server ready in the global namespace, just try to connect to that action from `/my_namespace`.

Since the MoveGroupCommander instantiates a `move_group_interface`, C++ nodes will have this issue as well.

marcoesposito1988, 2020-10-13:
Hi Felix,

thanks for chiming in!

glad that I didn't miss something trivial. I will then try to build a minimal example to pinpoint the problem.

v4hn, 2020-10-13:
Doing that right now.

v4hn, 2020-10-13:
Turns out this is an inherent shortcoming in roscpp and I am utterly confused about how I've never heard of this one before or why I can't find a workaround either.

`ros::NodeHandle{}` and `ros::NodeHandle{""}` are supposed to create a handle in the namespace of the node (not its private namespace) and that's what they do. However, `ros::NodeHandle{"/"}` behaves the same way and I do not see how that would make sense from the user perspective.

The following illustrates the behavior (and shows that even NodeHandle-based remapping does not work here):

```
#include <ros/ros.h>
#include <std_msgs/Header.h>

int main(int argc, char** argv){
   ros::init(argc, argv, "ns_test");

   ROS_INFO_STREAM("node ns: '" << ros::this_node::getNamespace() << "'");

   ros::NodeHandle nh{};
   ROS_INFO_STREAM("'' namespace '" << nh.getNamespace() << "'");

   ros::NodeHandle gnh{ "/" };
   ROS_INFO_STREAM("'/' namespace '" << gnh.getNamespace() << "'");

   ros::NodeHandle rgnh{ "/", ros::M_string { { ros::this_node::getNamespace(), "/" } } };

   ros::Publisher pub{ rgnh.advertise<std_msgs::Header>("bar", 1) };
   ROS_INFO_STREAM("publisher topic from remapped node: '" << pub.getTopic() << "'");

   ros::Publisher apub{ rgnh.advertise<std_msgs::Header>("/bar", 1) };
   ROS_INFO_STREAM("publisher topic from remapped node: '" << apub.getTopic() << "'");

   ros::spin();

   return 0;
}
```

Started in a namespace `"/foo"`, this outputs 
```
1602577464.346011054: /foo/ns_test.ros.psi_test: node ns: '/foo'
1602577464.352168463: /foo/ns_test.ros.psi_test: '' namespace '/foo'
1602577464.353890719: /foo/ns_test.ros.psi_test: '/' namespace '/foo'
1602577464.354729984: /foo/ns_test.ros.psi_test: publisher topic from remapped node: '/foo/bar'
1602577464.355323922: /foo/ns_test.ros.psi_test: publisher topic from remapped node: '/bar'
```

So it seems the only way to access the global namespace is via a "/" in each topic/service interface.
This is very bad news for the MoveGroupInterface because we rely on the NodeHandle to hold that information...

One thing you could do is remap all the relevant names in the launch file for the node, but this is indeed inconvenient.

gavanderhoorn, 2020-10-13:
Not getting `/` remapped makes sense to me: it's an absolute name, and those don't get remapped ([source](https://github.com/ros/ros_comm/blob/089dd64a6c2d7769d2d8d3b10808eec621a3acf3/clients/roscpp/src/libros/node_handle.cpp#L266-L269)).

v4hn, 2020-10-13:
You are right, I was just trying to find a workaround and this approach does not work.
(nor does `{ ros::this_node::getNamespace().substr(1), "" }`)

It's noteworthy to point out that
```
ROS_NAMESPACE=foo rosrun package node /foo/bar:=/bar
```
successfully remaps the absolute topic though.

v4hn, 2020-10-13:
> Edit: not completely sure, but perhaps ros/ros_comm#692 is related.

Thanks for the reference @gavanderhoorn It looks like `NodeHandle { "/" }` is handled as a special case for some reason.
Using `NodeHandle { "/new_ns" }` actually works as expected (the resolved namespace is `/new_ns` instead of `/foo/new_ns`), but `/` still ends up with `/foo` although it follows the same logic (the first character is `/`).

felixvd, 2020-10-14:
> It's noteworthy to point out that
> ```
> ROS_NAMESPACE=foo rosrun package node /foo/bar:=/bar
> ```
> successfully remaps the absolute topic though.

So to use a `move_group_interface` in a node that's in a lower namespace while the `move_group` is in the global namespace, you would have to remap [these](https://github.com/ros-planning/moveit/blob/master/moveit_ros/move_group/include/moveit/move_group/capability_names.h) topics/services? E.g.:

```
<node name="your_nodename" pkg="your_package" type="your_node" ns="/foo" output="screen"/>
<remap from="/foo/move_group" to="/move_group"/>
<remap from="/foo/execute_trajectory" to="/execute_trajectory"/>
<remap from="/foo/apply_planning_scene" to="/apply_planning_scene"/>
...
```

Ramdambo, 2021-03-09:
Any luck on that issue ? I am facing the same problem right now, so I would be very grateful for any advice.

v4hn, 2021-03-09:
As discussed above, either remap every single topic/service in your launch file *or* start MoveIt in a namespace like "/moveit" as well.

In the latter case you can instantiate MoveGroupCommander as you would expect:
```
c= moveit_commander.MoveGroupCommander("panda_arm", ns="/moveit" )
```
Note that if your robot_description is also located in the moveit namespace (as in MoveIt's demo mode), you also have to specify the correct robot description parameter:
```
c= moveit_commander.MoveGroupCommander("panda_arm", robot_description="/moveit/robot_description", ns="/moveit" )
```

I will close this. I specified two workarounds for this issue and the actual problem is an edge case in roscpp that should be discussed/addressed there.

oskarsVismanis, 2022-01-04:
Not sure if this would be the correct place to ask and I wasn't sure if I should open a new issue, but would it be possible to do this the other way around, where a C++ node, that creates a `move_group_interface` instance, is located in the global namespace, but a `move_group` node/action server is located in a robot specific namespace? So in the case of multiple robots, the one C++ node in the global namespace could access two separate, namespaced move_group nodes?

v4hn, 2022-01-04:
Yes, just specify the namespace `ns` parameter accordingly as I wrote in my last reply above.

oskarsVismanis, 2022-01-04:
> Yes, just specify the namespace `ns` parameter accordingly as I wrote in my last reply above.

Sorry if I'm misunderstanding, but does this work only with `MoveGroupCommander `in python or is there a way to initialize `move_group = moveit::planning_interface::MoveGroupInterface(group_name, **ns**)` somehow?  I got lost in the documentation for `moveit_ros_planning_interface ` and can't seem to find how the python wrapper sets the namespace, so if you could maybe point me in the right direction for where to look for that!