moveit/moveit #2367: robot padding parameter default_object_padding is never used

mamoll, 2020-10-13:
### Description

The planning scene monitor can be configured to use a default object padding parameter through the parameter `robot_description + "_planning/default_object_padding"`. The padding should increase the clearance but is currently ignored.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? master

### Steps to reproduce

```
grep -r getDefaultRobotPadding .
./moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h:  double getDefaultRobotPadding() const
grep -r default_object_padd .
./moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h:    return default_object_padd_;
./moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h:  double default_object_padd_;
./moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:    default_object_padd_ = 0.0;
./moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:  nh_.param(robot_description + "_planning/default_object_padding", default_object_padd_, 0.0);
```

### Expected behaviour

Increasing the padding should increase the clearance between the robot and the environment.

### Actual behaviour

The parameter is ignored.

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

nyxrobotics, 2023-06-01:
Is there any solution to this problem?

Alexandriun, 2024-12-04:
I did also some testing to set the padding for the robot and also the attached objects. For the robot the padding is working as expected. But the attached collision objects do not respect the padding value. Is there any solution? I am using moveti2 and humble. 

best regards,
alex