moveit/moveit #2369: Melodic release

rhaschke, 2020-10-14:
Dear @tylerjw, will you also tackle a new Melodic release? In my eyes, this is even more important than a Noetic release, because we have fixed several bugs since 1.0.6 and [people are waiting for a release](https://github.com/ros-planning/moveit/issues/2286#issuecomment-701855289). Of course, we should consider back-porting new commits from master since `reviewed-for-backport`.

tylerjw, 2020-10-14:
I am happy to help with a Melodic release.  Generally backporting bug fixes and quality of life improvements to Melodic seem like good low hanging fruit and we should do that atleast once more though as there have been a decent number of them that should be trivial to do.

While all of our ROS1 customers are still on Melodic, none of them are using the version in the apt repositiories because there are many api braking changes in the master branch we want to use.  This makes me wonder how many people are actually using whatever is in melodic-devel (by way of the debain release) who shouldn't just transition to using Noetic at this point.  To that end I'm not sure how much effort we should spend on a melodic release when we could be working on getting open PRs merged or making Noetic easier to use.

The first step to making this happen is to make a list of the PRs merged since reviewed-for-backport and figure out which ones we want to backport.  I will compile that list and we can start a getting that done.  As to a schedule, how about we get the backports done in the next 6 weeks and at the next time when I will do a Noetic release, I'll also cut a Melodic release?

tylerjw, 2020-10-14:
> Anecdotal, but just wanted to add a data point.

My points are just anecdotal too.  I take your word for it and I'm happy to help make this happen.  I will at my next opportunity this week to work on maintenance tasks prepare the list of possible backports.

I'm curious, what are the reasons they can't migrage to Noetic easily?  Are there still large amounts of python2 code that they would have to port or is it some other ROS dependency that isn't in Noetic yet?

The reasons our customers are still on Melodic is just that I think they would rather transition to ROS2 than Noetic, and haven't dedicated the work to making that transition either.  I'd argue that nothing you do to transition to Noetic would be wasted if the end goal is Foxy (python3, 20.04, etc).

felixvd, 2020-10-15:
I know a few groups using the Melodic binaries as well, and I suspect that Melodic will remain the most used distribution for a while. It's barely halfway to its EOL, and judging by how many systems are still on Kinetic, many of those will probably do a lazy move to Melodic rather than the jump to Noetic.

gavanderhoorn, 2020-10-18:
> I'm curious, what are the reasons they can't migrate to Noetic easily?

*Other* code which depends on components which have introduced breaking changes between Bionic and Focal.

This is larger than just MoveIt.

tylerjw, 2020-11-20:
This is happening today.  I'm sorry this took so long to make happen.  I'll try to make this a regular part of our releasing.

tylerjw, 2020-11-20:
I updated changelogs and package versions and pushed a new tag (1.0.7), I'm now running the pre-release test locally.

tylerjw, 2020-11-25:
I created a new issue to track the work on the melodic release.  I'm closing this issue as it was asking for a release to be done for melodic and that work is underway.  We can continue discussion in the new issue.

closed 2020-11-25 by tylerjw