moveit/moveit #2374: Unable to extract cylinder from point cloud

abhyudit309, 2020-10-15:
I'm following the move-it tutorials for melodic. In the Perception pipeline tutorial, on running:
roslaunch moveit_tutorials detect_and_add_cylinder_collision_object_demo.launch
I get the Octomap in Rviz, but its not able to extract the cylinder from the point cloud.
Expected output:
[image]
My Output:
[image]
I also get an  error message in my terminal saying this:
[image]
Can someone help me fix this issue ?

abhyudit309, 2020-11-12:
Can anyone please help me resolve this issue ? I'm really stuck because of this and it's been nearly a month I posted this issue.

felixvd, 2020-11-12:
Sorry for the long silence. I can't confirm this error. Does it happen if you run everything inside a docker container [as described here](https://moveit.ros.org/install/docker/)? What if you start with a clean install? If the error persists, have a good look at why the point cloud preprocessor is dying. I don't work with point clouds, so I am not very familiar.

abhyudit309, 2020-11-13:
Thanks for your reply. I went through this: https://moveit.ros.org/install/docker/ , but I could not follow it. Could you please guide on how to do it ?

felixvd, 2020-11-13:
This is not a forum for user support, so please ask a concrete question on
ROS answers instead if you’re stuck.

abhyudit309, 2020-11-13:
I was able to run it inside a docker container. But the error still persists. I get the same error message saying: [point_cloud_preprocessor-11] process has died.

JafarAbdi, 2021-02-03:
@abhyudit309 Look like the `cylinder_segment` executable is segfaulting, could you please provide us with the stack trace using gdb .?