moveit/moveit #2376: Problems with RVIZ on Windows using Melodic

drewhamiltonasdf, 2020-10-20:
### Description
OPENGL errors running RViz on Windows with Melodic. I've found solutions to similar issues on Ubuntu, but not for Windows. I'm prepared to hear the inevitable, "install it on Linux", but I'm posting here hoping there is an easy solution.

### Environment
* ROS Distro: [Melodic]
* OS Version: Windows 10
* Installed yesterday with: choco upgrade ros-melodic-moveit -y --execution-timeout=0 
[ INFO] [1603154308.264114000]: rviz version 1.13.13
[ INFO] [1603154308.264318900]: compiled against Qt version 5.10.1
[ INFO] [1603154308.265067100]: compiled against OGRE version 1.10.11 (Xalafu)

### Steps to reproduce
Running: _roslaunch panda_moveit_config demo.launch rviz_tutorial:=true_ in the moveit1 tutorials, I end up with a whole host of error messages. Curiously, I was able to successfully run RViz yesterday evening with the same commands, and I had a blast playing around with the Panda arm. Today, running the same command, I run into one of two behaviours:

1. The errors listed below, or
2. The RViz window opens (I believe) off-screen somewhere past the monitor edge. I believe this is what is happening because I can see a ghost of a blank RViz GUI window when I mouseover the RViz icon on the taskbar.

### Backtrace or Console output
[ INFO] [1603154670.207433400]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1603154670.207769700]: MoveGroup context initialization complete
[92m
You can start planning now!

[0m[ INFO] [1603154671.526501900]: Stereo is NOT SUPPORTED
[ INFO] [1603154671.526681200]: OpenGl version: 4.6 (GLSL 4.6).
[ERROR] [1603154671.575143700]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.597487300]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.619834800]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.641855500]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.663998400]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.686321300]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.708262900]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.730628000]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.752855000]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.775168500]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.797667400]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.820230200]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.842693900]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.865005300]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.887211100]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
[ERROR] [1603154671.910003300]: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp (line 493)
rviz::RenderSystem: error creating render window: RenderingAPIException: wglShareLists() failed in  Win32Window::create at C:\temp\rosdeps\desktop\ogre-1.10.11\RenderSystems\GLSupport\src\win32\OgreWin32Window.cpp ( [... cut ...]
[31m[rviz_Dhami123579_27332_2505790771525501991-7] process has died [pid 17960, exit code -1073741819, cmd C:\opt\ros\melodic\x64\lib\rviz\rviz.exe -d C:\moveit_ws\src\panda_moveit_config\launch\moveit_empty.rviz __name:=rviz_Dhami123579_27332_2505790771525501991 __log:=C:\Users\Drew Hamilton\.ros\log\64f2c8a1-126d-11eb-b67c-d050998091fd\rviz_Dhami123579_27332_2505790771525501991-7.log].
log file: C:\Users\Drew Hamilton\.ros\log\64f2c8a1-126d-11eb-b67c-d050998091fd\rviz_Dhami123579_27332_2505790771525501991-7*.log[0m

drewhamiltonasdf, 2020-10-20:
As a followup. I was able to reproduce the 2nd behavior, in which the window doesn't show up, or shows up off-screen. And this is the last of the command-line output:

[ INFO] [1603155652.552929800]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1603155652.553232000]: MoveGroup context initialization complete
[92m
You can start planning now!

[0m[ INFO] [1603155653.779585600]: Stereo is NOT SUPPORTED
[ INFO] [1603155653.779732300]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1603155654.000950900]: Loading robot model 'panda'...
[ INFO] [1603155654.505753200]: Loading robot model 'panda'...
[ INFO] [1603155654.644715700]: Starting planning scene monitor
[ INFO] [1603155654.647053300]: Listening to '/move_group/monitored_planning_scene'
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)

rhaschke, 2020-10-20:
The console messages for the 2nd behavior are normal. The error messages above are new to me. In any case, this issue isn't related to MoveIt, but to Windows rviz. Please direct your questions to https://answers.ros.org as stated on the [ROS for Windows installation page](http://wiki.ros.org/Installation/Windows).

closed 2020-10-20 by rhaschke