moveit/moveit #2377: robot model not visible in moveit after loading urdf file, using melodic and Ubuntu 18.04

shirazbaig, 2020-10-20:
When I launch moveit to create a new config file, it asks me to load the files. I use /opt/ros/melodic/share/franka_description/robots/panda_arm_hand.urdf.xacro  file
It loads successfully but not show robot. 
This issue existed in older versions of ros and it was fixed as #786.  And in the solution it was also mentioned that this issue does not appear in melodic. But it is still appearing in melodic.
I am using Ubuntu 18.04 and  
/rosdistro: melodic
/rosversion: 1.14.9
ros-melodic-moveit-runtime/bionic 1.0.6-1bionic.20200930.144856 amd64
please suggest a solution so that robot model is visible in moveit. 
Note: the model is perfectly working and visible in  
roslaunch      panda_moveit_config     demo.launch

rhaschke, 2020-10-20:
I guess, when referring to MoveIt, you actually mean MoveIt Setup Assistant. The problem was fixed in source (2ba284275cb1dd99d53bb9cd01e143a1fd6211da), but not yet released. @tylerjw, we should really aim for a new Melodic release asap.

closed 2020-10-20 by rhaschke

Sission, 2020-11-09:
> I guess, when referring to MoveIt, you actually mean MoveIt Setup Assistant. The problem was fixed in source ([2ba2842](https://github.com/ros-planning/moveit/commit/2ba284275cb1dd99d53bb9cd01e143a1fd6211da)), but not yet released. @tylerjw, we should really aim for a new Melodic release asap.

Hi! Actually, I modified those 3 files in source (2ba2842), but it still does not work.

rhaschke, 2020-11-09:
@Sission, are you sure you are actually using your modified MoveIt?

Sission, 2020-11-09:
> @Sission, are you sure you are actually using your modified MoveIt?

I did modify all 3 files in the MoveIt install path... Any other thing I need to do to "activate" it?

rhaschke, 2020-11-09:
@Sission: First of all which files you are talking about exactly?
Just changing source files in the _install_ path doesn't help at all. You need to setup a catkin workspace and compile from source. Please follow the [MoveIt build instructions](https://moveit.ros.org/install/source).

Sission, 2020-11-09:
Thanks, fixed after installing from source.