moveit/moveit #2379: Attached Collision Objects can collide with regular collision objects

DLu, 2020-10-20:
### Description

In updating [the moveit tutorials](https://github.com/ros-planning/moveit_tutorials/pull/542#issuecomment-712941017), I discovered that attached objects can collide with regular collision objects. 

[image]

(the attached cylinder would collide with the rectangle if the plan was executed).

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Source
* If source, which branch? noetic-devel

### Steps to reproduce
This behavior is illustrated with the [Move Group C++ Interface Tutorial](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html) with [this PR applied](https://github.com/ros-planning/moveit_tutorials/pull/542)

### Expected behaviour
Planned path should result in a path with no collisions, either higher or turning the wrist. 

### Actual behaviour
As in the above gif, the cylinder would go through the box.

AndyZe, 2020-10-20:
This PR will at least point us toward the right area of code:  https://github.com/ros-planning/moveit/pull/2355/files

AndyZe, 2020-10-20:
Also, @JafarAbdi noticed that there is no collision if planned via the GUI plugin. That could be another hint.

DLu, 2020-10-20:
@JafarAbdi figured it out. 
A) The object needed to be attached to the hand.
B) The start state needed to be updated. 

See https://github.com/ros-planning/moveit_tutorials/pull/542/commits/3dda4a010f200dd60fda0eb0d66c747cae907177#diff-a5c0141aeeaa8ad7836f3b230e2e8cfa49fd26ea3dcfe3496b83478f58d7a265L401

closed 2020-10-20 by DLu

felixvd, 2020-10-21:
I didn't know that `setStartStateToCurrentState` is required after attaching/detaching too, although it makes sense. It might be worth outputting "Starting planning from a state that is not the current state (attached objects differ)" or similar when necessary. Most users probably assume that the "real" robot is used by default.