moveit/moveit #2383: Convert use of new to std::make_unique in background_processing

tylerjw, 2020-10-22:
## Background

As pointed out in [this issue](https://github.com/ros-planning/moveit/issues/1411) using new in the constructor for smart pointers is less performant than using the methods `std::make_shared` or `std::make_unique`.  This issue is for updating one of those uses in moveit_core.

## Instructions
Hi, this is a `first-timer-only` issue. This means we've worked to make it more legible to folks who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**. 

If that's you, we're interested in helping you take the first step and can answer questions and help you out as you do. Note that we're especially interested in contributions from people from groups underrepresented in free and open-source software!

We know that the process of creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this one for someone new**, and looking through our general [bug](https://github.com/ros-planning/moveit/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️Create a local workspace for making your changes and testing [follwing these instructions](https://moveit.ros.org/install/source/)

- [ ] 📝 **Update** the file `moveit_core/background_processing/src/background_processing.cpp`by applying the change shown in the diff below.

```diff
diff --git a/moveit_core/background_processing/src/background_processing.cpp b/moveit_core/background_processing/src/background_processing.cpp
index 37991357b..5614aa514 100644
--- a/moveit_core/background_processing/src/background_processing.cpp
+++ b/moveit_core/background_processing/src/background_processing.cpp
@@ -46,7 +46,7 @@ BackgroundProcessing::BackgroundProcessing()
   // spin a thread that will process user events
   run_processing_thread_ = true;
   processing_ = false;
-  processing_thread_.reset(new boost::thread(boost::bind(&BackgroundProcessing::processingThread, this)));
+  processing_thread_ = std::make_unique<boost::thread>(boost::bind(&BackgroundProcessing::processingThread, this));
 }
 
 BackgroundProcessing::~BackgroundProcessing()
```

- [ ] 🤖 Apply `clang-format-10` auto formatting, [following these instructions](https://moveit.ros.org//documentation/contributing/code/?#clang-format-auto-code-formatting)

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:
1. If you are not familiar with GitHub or the pull request model, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.
2. If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it. 

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Leave a comment below!

## Meta

This was adapted from [publiclab](https://github.com/publiclab/plots2/blob/master/.github/first-timers-issue-template.md) as a trial of how we would do first-timer-only issues.  I didn't

LegoYoda112, 2020-11-12:
Hey! Super interested in getting involved in open source robotics projects. Always been a bit worried about just jumping into an open source project, but this seems like a wonderful way to start, Ill claim this issue! 😄

LegoYoda112, 2020-11-13:
Some questions / notes after working through this:

#### rosdep
More of a note for the guide at https://moveit.ros.org/install/source/, rosdep does not appear to be included in a ROS desktop install, so `rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}` failed initially. (this might have just been user error on my part through)

#### Forking
I was unsure of how to properly clone my own fork into the workspace, since the code in `src` is being managed by wstool. Ended up just deleting the moveit folder and then cloning my fork, but it felt a bit hacky (I think wstool would attempt to overwrite if I ever updated). Would the correct way be changing my fork's `moveit.rosinstall` file and using that in the wstool merge command?

#### Pull request
Once I had my fork ready, I was unsure which branch to pull request to. I chose master after looking through some other pull requests. Was this correct?

Thanks for taking the time to write out this awesome guide!

gavanderhoorn, 2020-11-13:
> More of a note for the guide at https://moveit.ros.org/install/source/, rosdep does not appear to be included in a ROS desktop install, so `rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}` failed initially. (this might have just been user error on my part through)

No, this is not user error. `rosdep` was removed from the default set of installed packages some time ago. So if you're running Noetic, you'll have to install it manually.

tylerjw, 2020-11-13:
> I was unsure of how to properly clone my own fork into the workspace, since the code in `src` is being managed by wstool. Ended up just deleting the moveit folder and then cloning my fork, but it [...]

One approach I've used when I don't want wstool the complain at me is to edit the .rosinstall file and remove whatever repo I'm working on.  Then you can use wstool for the rest of the packages and use git directly to interact with the repo you have forked and are working on.

tylerjw, 2020-11-13:
> Once I had my fork ready, I was unsure which branch to pull request to. I chose master after looking through some other pull requests. Was this correct?

`master` is the correct branch to target for PRs.  Would you mind creating a `first-timer-only` issue using the issue template (click new issue) and include a diff to the file `.github/ISSUE_TEMPLATE/first_timers_only.md` to clarify this in the instructions.  That would help as you can provide a similarly curated issue for someone else and help make it clearer for them.

Thank you for joining us and working to make open source robotics better.  We look forward to you being a part of the community.

davetcoleman, 2020-11-18:
@LegoYoda112 I've converted your rosdep catch into a first timer issue :-)
https://github.com/ros-planning/moveit/issues/2416

davetcoleman, 2020-11-18:
Also, this issue should be closed as https://github.com/ros-planning/moveit/pull/2412 is merged

tylerjw, 2020-11-20:
closing as this is done, congrats 🥇

closed 2020-11-20 by tylerjw