moveit/moveit #2386: Jump in moveit_servo velocity command

AndyZe, 2020-10-22:
### Description

When moveit_servo is configured to publish velocity commands, we occasionally see a jump in the outgoing command (from moveit_servo to the ros_control velocity command topic). It's not 100% clear if this is a moveit_servo problem but I figured I will create the issue and see if anybody else has a similar problem.

### Your environment
* ROS Distro: Melodic
* Ubuntu 18.04
* Building master branch from source

### Steps to reproduce
Here are the parameters from the Servo config file:

### Plots
This first plot (from top to bottom) shows:  

- Outgoing command, from Servo to the ros_control controller. The jump is obvious
- The hardware's velocity state. It reports the jump correctly, seems OK
- The hardware's position state. It seems OK
- The hardware's position command (not really relevant, since the arm wasn't in velocity control mode at the time)

[image]

I didn't capture the /joint_states topic but it seemed steady, too.

Second plot, zoomed in to show one joint clearly:

[image]

Possibly it's a discrete differentiation problem. Internally, moveit_servo filters joint positions then does discrete differentiation to calculate velocity.

v4hn, 2021-06-15:
Is this still relevant? I believe you solved a number of issues w.r.t. velocity commands in the recent past.

AndyZe, 2021-06-15:
I haven't seen it recently or heard any complaints   :+1:

closed 2021-06-15 by AndyZe