moveit/moveit #2387: Melodic backports

tylerjw, 2020-10-23:
Here is the list of changes since reviewed-for-backport on master:

 - [ ] Start new joint_state_publisher_gui on param use_gui (#2257)
 - [ ] add soname version to moveit_servo (#2266)
 - [ ] Move constraint representation dox to moveit_tutorials (#2147)
 - [ ] Update last_sent_command_ at ServoCalcs start (#2249)
 - [ ] Add a utility to print collision pairs (#2275)
 - [ ] fix some spelling errors in the ompl interface (#2279)
 - [ ] Cleanup: Add a utility to print collision pairs (#2278)
 - [ ] Replace $(find xacro)/xacro -> xacro (#2282)
 - [ ] adjust code coverage yaml to make it more forgiving (#2283)
 - [ ] [maint] Remove kinetic from Travis CI testing (#2295)
 - [ ] Temporarily add srdfdom to moveit.rosinstall
 - [ ] Revert "Temporarily add srdfdom to moveit.rosinstall"
 - [ ] Noetic docker files
 - [ ] 1.1.0
 - [ ] MSA: only write default_planner_config field if any is selected (#2293)
 - [ ] Replace panda_moveit_config -> moveit_resources_panda_moveit_config (#2300)
 - [ ] [maint] Remove patch status from codecov (#2306)
 - [ ] Update codeowners file (#2309)
 - [ ] [fix] Servo heap-buffer-overflow bug (#2307)
 - [ ] Add comment to MOVEIT_CLASS_FORWARD (#2315)
 - [ ] robot_description is already loaded in move_group.launch (#2313)
 - [ ] Refactor velocity limit enforcement and add a unit test (#2260)
 - [ ] Noetic was released (#2304)
 - [ ] Fix Servo thread interruption (#2314)
 - [ ] MGC: Improve exception messages (#2318)
 - [ ] Clean up Rviz Motion Planning plugin, add tooltips (#2310)
 - [ ] ikfast: add missing python dependencies
 - [ ] Install python scripts via catkin_install_python()
 - [ ] Remove shebang line from moveitjoy_module.py
 - [ ] Adapt to API changes in geometric_shapes (#2324)
 - [ ] Travis: default to Noetic
 - [ ] Remove custom copy-assignment operator for DistanceResultsData
 - [ ] perception: add missing dependency to OpenMP
 - [ ] Silent deprecation warnings of gtest's typed test suites
 - [ ] Handle multiple link libraries for FCL (#2325)
 - [ ] Fix clang-tidy warning (#2334)
 - [ ] Fix some clang-tidy issues on Travis (#2337)
 - [ ] [fix] Segfault when editing pose in moveit_setup_assistant (#2340)
 - [ ] Fix "Clear Octomap" button, disable when no octomap is published (#2320)
 - [ ] moveit_cpp: more informative error message, cover another potential failure condition. (#2336)
 - [ ] MSA: Update RobotState *before* visualization
 - [ ] remove commented, copy-pasted code
 - [ ] Use PlanningScene's RobotState for postures
 - [ ] Fix ROS namespacing in moveit_commander's PSI (#2347)
 - [ ] Reorder buttons in GroupEditWidgets
 - [ ] Replace dummy QWidgets with QSpacerItems
 - [ ] Use QStackedWidget instead of QWidget+QStackedLayout
 - [ ] Remove released upstream packages from docker files
 - [ ] Increase minimal window size
 - [ ] Cleanup Qt includes
 - [ ] Cleanup ROS includes
 - [ ] Remove diagnostics macros
 - [ ] Fix unused-parameter warnings
 - [ ] MSA: Allow showing both, visual and collision geometry (#2352)
 - [ ] MSA: Fix group editing (#2350)
 - [ ] move_group: simplify test launch files
 - [ ] A library for servoing toward a moving pose (#2203)
 - [ ] Depend on ros-noetic-fcl (0.6) in Noetic (#2359)
 - [ ] Make GILReleaser exception-safe (#2363)
 - [ ] Changelogs for Noetic 1.1.1
 - [ ] 1.1.1
 - [ ] Python interface improvements. Fix #1966, add enforceBounds (#2356)
 - [ ] Fix ordering of windup args to control_toolbox::Pid (#2370)
 - [ ] MSA launch files: fix indentation (#2371)
 - [ ] Add test for PlanningContextManager in ompl interface (#2248)
 - [ ] Update .github files (#2380)
 - [ ] Fix servo trajectory point timestamping (#2375)
 - [ ] Change servo namespacing logic (#2354)
 - [ ] Cleanup current state handling in servo (#2372)

felixvd, 2020-10-23:
What's the best way to do this? A separate branch from `melodic-devel` on the main repo to test commits on, and then fast-forwarding `melodic-devel` to it after the necessary commits were cherry-picked?

tylerjw, 2020-10-23:
I got started on it and will submit a pull request with a handful of these that are easy to backport with some notes about the ones that I couldn't figure out.  Basically what you do is create a new branch (like you would for any pull request) that is based on melodic-devel and then start cherry-picking these commits onto that branch.  You can then submit a normal PR except you target melodic-devel.  When it is accepted we do a rebase merge and it will just place those commits on the melodic-devel branch.

tylerjw, 2020-10-23:
I find it helpful to have a second checkout of moveit on master so I can look through the git log of it easily to find the commit hashes and then cherry-pick those commits into the melodic-devel setup I'm testing with.

tylerjw, 2020-10-23:
Here are a set of backports I found straightforward to apply: https://github.com/ros-planning/moveit/pull/2391

tylerjw, 2020-11-20:
This is now done, I will release these changes today onto melodic and move the reviewed-for-backport tag after the release.

closed 2020-11-20 by tylerjw