moveit/moveit #2396: Kinematics plugins cannot be loaded with multiple parameters

felixvd, 2020-10-28:
I don't know if I should tag this as bug or enhancement, but it looks like the UR kinematics package cannot be used for multiple robots in the scene, because it is only loaded once, so only one of the robots' prefixes can be set.

See https://github.com/ros-industrial/universal_robot/issues/247

simonschmeisser, 2020-11-02:
Could you add/link an example that doesn't work?

felixvd, 2020-11-02:
Can't guarantee that the example fails out of the box (I am not on an Ubuntu machine), but if you load [this URDF](https://gist.github.com/felixvd/74b9deb220227b055fb3b9601680c599) with a setup assistant and try setting the URKinematics package for both of them, the issue should occur.

I adapted it from [an old setup](https://github.com/o2as/ur-o2as/blob/master/catkin_ws/src/o2as_scene_description/urdf/o2as_base_scene.urdf.xacro) where we had that problem.

gavanderhoorn, 2020-11-03:
Could well be the problem / a contributing cause is in `ur_kinematics`. It hasn't been updated in some time. There are a few outstanding PRs, but I'd like someone with some insight into kinematics to go over them before merging.

simonschmeisser, 2020-11-03:
@gavanderhoorn please tag me in the one you'd like to merge first (or link it here or whatever). We had some not quite convincing results from it recently so I should have a look anyway

gavanderhoorn, 2020-11-03:
Thanks for the offer @simonschmeisser.

There is actually an entire project board for IK issues and PRs: [ros-industrial/universal_robot/projects#1](https://github.com/ros-industrial/universal_robot/projects/1).