moveit/moveit #2399: MoveIt setup assistant

michelepenzo, 2020-10-29:
Hi, I'm trying to create a new definition of my robot with MoveIt wizard.
The robot definitionis ok bu the EE has an hand and two prismatic fingers. If I use the "prismatic" tag in urdf.xacro definition this error appears:

**The complete state of the robot is not yet known.  Missing finger_joint1**

The same error appears if I set it as revolute joint, but If it's a fixed joint this error doesn't appear. 

Thanks a lot!
Michele

felixvd, 2020-10-29:
That is not an error that occurs when running `moveit_setup_assistant`, so I suspect this is a usage problem and not a bug. If it is not an actual bug in the software, would you mind posting this (with more detail, e.g. by including your URDF and error messages) over at [ROS Answers](https://answers.ros.org/questions/ask/) instead? Thanks.

michelepenzo, 2020-10-29:
Thanks @felixvd !

closed 2020-10-29 by michelepenzo