moveit/moveit #2400: Retiming trajectory after compute cartesian path

wlawler45, 2020-11-03:
### Description

I have been having trouble getting the move group interface to replan trajectory with time optimal parameterization. Im using compute cartesian path in the python interface and the plan that results has inconsistent velocity at the end effector and I'm trying to then replan the trajectory using the generated waypoints so that it will then have the time parameterization I want from that but I don't know how I should do that in python, are there any suggestions on how I can do that? Please let me know.

### Your environment
Kinetic, source build

felixvd, 2020-11-04:
Why not use [retime_trajectory](http://docs.ros.org/en/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a5fb1d772b6d248a6c4113412eacb4330)?

wlawler45, 2020-11-04:
I tried, it didn't change the behavior of the robot or the timing of the trajectory at all.

felixvd, 2020-11-05:
Can you paste how you are using it in your code?

wlawler45, 2020-11-05:
for i in range(len(dxf_points)):
            pose_goal = Pose()
            rpy=[math.pi,0.0,sines[i]-math.pi/2]
            print(rpy)
            R_result=rox.rpy2R(rpy)
            quat=rox.R2q(R_result)
            print(quat)
            pose_goal.orientation.w=0.0
            pose_goal.orientation.x=quat[1]
            pose_goal.orientation.y=quat[2]
            pose_goal.orientation.z=0.0
            
            x_val=(20-dxf_points[i][0]) *0.0254
            y_val=dxf_points[i][1]*0.0254
            pose_goal.position.x=0.8+y_val
            pose_goal.position.y=x_val
            pose_goal.position.z=0.3
            print(pose_goal)
            
            waypoints.append(pose_goal)

        
        (plan, fraction) = self.move_group.compute_cartesian_path(
                                       waypoints,   # waypoints to follow
                                       0.01,        # eef_step
                                       0.0)
plan=self.move_group.retime_trajectory(moveit_robot_state,plan,1.0)

Is this what you needed?

gavanderhoorn, 2020-11-05:
@wlawler45: where do you request MoveIt to use a different time-parameterisation from its default?

wlawler45, 2020-11-05:
I tried changing the retime line to this but it gave me an error for too many arguments
plan=self.move_group.retime_trajectory(moveit_robot_state,plan,1.0,1.0,"iterative_spline_parameterization") did I do that wrong?

wlawler45, 2020-11-05:
Is that version of retime_trajectory not in the branch I'm using? If so is there a moveit branch for kinetic that would use that version of retime_trajectory?

gavanderhoorn, 2020-11-05:
If you are on Kinetic it's likely that functionality is not available to you.

I haven't checked in which version that first appeared. I'd recommend using a more up-to-date version of MoveIt in any case. There have been *many* improvements added to MoveIt since the last Kinetic release.

AndyZe, 2020-11-05:
The best time parameterization algorithm right now is `time_optimal_trajectory_generation`. It looks like it is not available on Kinetic. Sorry.

If you do upgrade to, say, Melodic, here are some slides about using TOTG:  https://picknik.ai/docs/moveit_workshop_macau_2019/TOTG.pdf

AndyZe, 2020-11-05:
Once you specify the use of TOTG in the yaml file, you shouldn't need to manually re-time anything.

gavanderhoorn, 2020-11-05:
@AndyZe: does `computeCartesianPath(..)` use the planning request adapter? Afaik it's hard-coded to use the IPTP?

AndyZe, 2020-11-05:
I guess I don't know. OK, disregard -- it may be necessary to retime the path.

I stand by TOTG being the best available, though.

wlawler45, 2020-11-06:
I did update my workspace to melodic and now I can't use the retime trajectory command because it gives the error Validation failed: Missing valid timestamp data for trajectory pt 1. What should I do about that?

felixvd, 2020-11-09:
Did your plan succeed? It sounds like the plan failed and the trajectory you're trying to retime is empty.

felixvd, 2021-01-28:
Friendly ping @wlawler45

felixvd, 2021-02-19:
I'll close this for lack of activity. Feel free to reopen if this is a bug.

closed 2021-02-19 by felixvd

luiz0897, 2022-02-14:
Hey guys,

I'm working with ROS Noetic and MoveIt.

I'm interested in parameterizing my trajectories with **TOTG** after using the cartesian path. 
Using **retime_trajectory** I can parameterize the trajectory with a delta time (dt) of 0.1 seconds. Is there any way to change this time parameter?

I tried modifying **time_optimal_trajectory_generation.h** but it didn't work.

AndyZe, 2022-02-14:
Unfortunately it looks like the time step parameter is hard-coded in time_optimal_trajectory_generation.cpp:

`constexpr double DEFAULT_TIMESTEP = 1e-3;`

A PR to turn that into an optional parameter would probably be welcome!

luiz0897, 2022-02-15:
Thanks for the answer!

After checking my MoveIt installation files, I found that I didn't have the **time_optimal_trajectory_generation.cpp** file (I used MoveIt binary installation). In my trajectory_processing folder I only had the .h files.

I tried adding the files manually, but it still didn't work.
The strange thing is that even deleting the folder, I can still use the **retime_trajectory** with dt = 0.1s (TOTG).

Any idea how I can resolve this?

Thanks.

hassan-shehawy, 2023-02-28:
Hi, can you please share how you use it in c++? I can't seem to find a `retime_trajectory` function in the `MoveGroupInterface` class [here](http://docs.ros.org/lunar/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroupInterface.html)

v4hn, 2023-02-28:
There is no need for it in c++ because you can directly instantiate and use the respective classes in moveit_core/trajectory_processing (notably TOTG).

AndyZe, 2023-02-28:
Here's a link:  https://github.com/ros-planning/moveit/blob/326e9b5fe76bdf856f9c4a2be56ce7e136651ff2/moveit_core/trajectory_processing/include/moveit/trajectory_processing/time_optimal_trajectory_generation.h#L170