moveit/moveit #2405: Questions about PlanningSceneMonitor: can this improve the code?

rafaelrojasmiliani, 2020-11-05:
## Background

I'm currently studing the class `PlanningSceneMonitor` and I think I have found two elements that may be a bug.

1. In [lines 193-195](https://github.com/ros-planning/moveit/blob/47884198c2585215de8f365a7ff20479f8bb4b51/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L193) the variable `PlanningSceneMonitor::scene_` is used passed to `CollisionPluginLoader::setupScene`. However, after this the lines 196-231 handle what happen if `scene_` is null and what to do is it is instantiated correctly. I think that calling `CollisionPluginLoader::setupScene` before this logic is confusing, because if `scene_` is null this function is called twise. I wrote a sketch of my idea of how the code should be in order to call `CollisionPluginLoader::setupScene` just once
```C++
    robot_model_ = rm_loader_->getModel();
    scene_ = scene;
    scene_const_ = scene_;
    if (!scene_)
    {(/*same code with line 201 with collision_loader_.setupScene(nh_, scene_) removed*/ }
    if(scene_){
      collision_loader_.setupScene(nh_, scene_);
      // the same code here
    }
```
2. The null pointer handling of `CollisionPluginLoader::setupScene` [here](https://github.com/ros-planning/moveit/blob/0ba091ce1faf6e11896fd0d06bea8234cf642ca3/moveit_ros/planning/collision_plugin_loader/src/collision_plugin_loader.cpp#L110) can hide errors and make them difficult to debug.

3. The member variable `PlanningSceneMonitor::robot_model_` is private and only appears in the left hand side of assignations and a reset. What is the purpose of this variable?

I can create a PR if the moveit community thinks that these points are valid.

rhaschke, 2020-11-05:
Thanks for this feedback. Indeed, looking at the code, it doesn't seem to be very clear. For 1. I would like to see a PR that cleans up the code, first initializing the scene and only in the end assigning it once to `scene_const_` and configuring the collision loader.

Regarding 2. feel free to add a warning message.

3. The `robot_model_` copy is required to keep the shared pointer alive as long as the PlanningSceneMonitor needs it.

rafaelrojasmiliani, 2020-11-05:
Hello @rhaschke , I have just seen another use of `PlanningSceneMonitor::robot_model_`, it is returned by `PlanningSceneMonitor::getRobotModel` and is used in `MoveGroupContext` to initialize the `TrajectoryExecutionManager` and `PlanningPipeline`. So this variable is really required.
I will do the PR as soon I have time.
Regards

felixvd, 2021-02-19:
@rafaelrojasmiliani Friendly ping :)

rafaelrojasmiliani, 2021-03-04:
Hello. Is there an automated way to test MoveIt? after I implement my modifications, is there a standard way to test if it works?

felixvd, 2021-03-04:
Yes, it's described [here](https://moveit.ros.org/documentation/contributing/) under "Running Tests Locally". The page also lists a few more procedures (`clang-format`, `clang-tidy`) you will need to know about.

There is also [this stub tutorial](https://ros-planning.github.io/moveit_tutorials/doc/tests/tests_tutorial.html) for writing your own tests.

rafaelrojasmiliani, 2021-03-19:
Hello. Work in progress!. I implement my modifications to the code to [this branch](https://github.com/rafaelrojasmiliani/moveit/tree/Issue%232405). However, I have to test using the Docker CI.

felixvd, 2021-03-24:
Feel free to submit a [draft PR](https://github.blog/2019-02-14-introducing-draft-pull-requests/) to make your modifications easier to review.

JafarAbdi, 2021-04-30:
Addressed in https://github.com/ros-planning/moveit/commit/f62c30f61e539541c7a7d0387db75c559b0974e2

closed 2021-04-30 by JafarAbdi