moveit/moveit #2407: 'tuple' object has no attribute 'joint_trajectory in  _Display_trajectory

hariharan382, 2020-11-06:
## Background

i am trying to launch the code to control universal robot via python node
[https://answers.ros.org/question/318963/beginner-question-on-ur5-robot-simulation-on-gazebo-with-rviz-visualization/]

## Instructions
i executed and ran the the my python file which is stored ur_driver folder
rosrun ur_driver manipulator.py

>melodic
>ubuntu 18.04

### 📋 Step by Step
but getting error

`/home/hari/ur10_ws/src/universal_robot/ur_driver/manipulator.py:20: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  moveit_msgs.msg.DisplayTrajectory)
Traceback (most recent call last):
  File "/home/hari/ur10_ws/src/universal_robot/ur_driver/manipulator.py", line 24, in <module>
    ose_target.position.x = 0.1   #0.1
NameError: name 'ose_target' is not defined
hari@hari18:~$ rosrun ur_driver manipulator.py
[ INFO] [1604681574.021283407, 11577.515000000]: Loading robot model 'ur5'...
[ WARN] [1604681574.029625604, 11577.525000000]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1604681574.029948935, 11577.526000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1604681574.309633966, 11577.788000000]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1604681575.382638170, 11578.791000000]: Ready to take commands for planning group manipulator.
/home/hari/ur10_ws/src/universal_robot/ur_driver/manipulator.py:20: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  moveit_msgs.msg.DisplayTrajectory)
Traceback (most recent call last):
  
File "/home/hari/ur10_ws/src/universal_robot/ur_driver/manipulator.py", line 33, in <module>
    display_trajectory_publisher.publish(display_trajectory);
  
File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  
File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
  
File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msg.py", line 152, in serialize_message
    msg.serialize(b)
  F
File "/home/hari/ws_moveit/devel/lib/python2.7/dist-packages/moveit_msgs/msg/_DisplayTrajectory.py", line 454, in serialize
    _v1 = val1.joint_trajectory
AttributeError: 'tuple' object has no attribute 'joint_trajectory'
`
please help me with i have no idea what this error is

felixvd, 2020-11-06:
1. Usage questions should be posted at [ROS Answers](https://answers.ros.org). Please put some more effort into formatting your question and show what you have tried.

2. `plan` [now returns a tuple](http://docs.ros.org/en/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a79a475263bffd96978c87488a2bf7c98). If your original line was this:

```
plan1 = group.plan()
```

then you can change it to this:

```
plan_success, plan1, planning_time, error_code = group.plan()
```

closed 2020-11-06 by felixvd

hariharan382, 2020-11-06:
thank you sir worked well,but i want to understand all these can you
suggest any article or book

felixvd, 2020-11-08:
I have nothing specific to recommend but the [ROS](http://wiki.ros.org/ROS/Tutorials) and [MoveIt tutorials](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/index.html), but since this was not complicated to debug with some Python knowledge, maybe you would do well to look into Python some more as well.