moveit/moveit #2416: Improve Install Instructions for rosdep

davetcoleman, 2020-11-18:
## Background

@LegoYoda112 reported in https://github.com/ros-planning/moveit/issues/2383#issuecomment-726777938 👍 

>More of a note for the guide at https://moveit.ros.org/install/source/, rosdep does not appear to be included in a ROS desktop install, so rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} failed initially.

## Instructions
Hi, this is a `first-timer-only` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-planning/moveit/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️**Clone the moveit website** for making your changes to the install instructions (git clone https://github.com/ros-planning/moveit.ros.org)

- [ ] 📝 **Update** the file(s) `/install/source/index.markdown` by applying the diff below:

```diff
- sudo apt install python3-wstool python3-catkin-tools clang-format-10
+ sudo apt install python3-wstool python3-catkin-tools clang-format-10 python3-rosdep
```

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.
2. If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [MoveIt FAQ](https://moveit.ros.org/documentation/faqs/)
* [MoveIt Tutorials](https://ros-planning.github.io/moveit_tutorials/)
* [MoveIt contribution guide](https://moveit.ros.org/documentation/contributing/)
* [ROS Tutorials](https://wiki.ros.org/ROS/Tutorials)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Jidnyasa-git, 2020-11-18:
Hi, I am new to open source community and want to work on this issue. Can you assign it to me ?

rhaschke, 2020-11-26:
Closed via https://github.com/ros-planning/moveit.ros.org/pull/525.

closed 2020-11-26 by rhaschke