moveit/moveit #2424: Update the First Timers Only Issue Template

harshgupta246, 2020-11-20:
## Background

The `first-timers-only` issue template needs to be updated to add simple and clear instructions on how to ping someone. Additional suggestions are welcomed to improve the template. To get a better understanding of what it means to ping someone, you can refer to - https://github.com/ros-planning/moveit/pull/2419#issuecomment-731276590

## Instructions
Hi, this is a `first-timer-only` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-planning/moveit/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🍴 **Fork the ros-planning/moveit repository** using the handy button at the top of the repository page and clone it onto your local machine. [Here is a guide that you can follow](https://guides.github.com/activities/forking/)

- [ ] 🗄️ **Clone the newly forked repository**

- [ ] 📝 **Update** the file(s) `.github/ISSUE_TEMPLATE/first_timers_only.md`by applying the change shown in the diff below.

```diff
diff --git a/.github/ISSUE_TEMPLATE/first_timers_only.md b/.github/ISSUE_TEMPLATE/first_timers_only.md
index c85e0638a..26cb87849 100644
--- a/.github/ISSUE_TEMPLATE/first_timers_only.md
+++ b/.github/ISSUE_TEMPLATE/first_timers_only.md
@@ -57,6 +57,9 @@ $DIFF
 ### 🤔❓ Questions?
 
 Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
+
+💬️ - You can even ping us by using an `@ mention` in the comments.
+
 Furthermore, you find helpful resources here:
 * [MoveIt FAQ](https://moveit.ros.org/documentation/faqs/)
 * [MoveIt Tutorials](https://ros-planning.github.io/moveit_tutorials/)
```

- [ ] 💾 **Commit** your changes

- [ ] 🔀 **Start a Pull Request**. There are two ways how you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.
2. If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [MoveIt FAQ](https://moveit.ros.org/documentation/faqs/)
* [MoveIt Tutorials](https://ros-planning.github.io/moveit_tutorials/)
* [MoveIt contribution guide](https://moveit.ros.org/documentation/contributing/)
* [ROS Tutorials](https://wiki.ros.org/ROS/Tutorials)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Prakhar030, 2020-11-20:
Hi, I am new to open source community and I would like to work upon this issue.

tylerjw, 2020-11-22:
I assigned it to you, enjoy!

Prakhar030, 2020-11-22:
Thank You @tylerjw for assigning me this issue. I have made the required change and made a pull request. It is showing that reviews are needed for the pull request to be merged. Please do give your review upon it and also please can you clarify that review is always required before a pull request is accepted or are my changes not matching any required criteria that is why such a thing is happening?

tylerjw, 2020-11-23:
All PRs have to be reviewed before they can be merged, that is a policy across the whole repo.  I'll go look at it now.

Prakhar030, 2020-11-23:
Thank you @tylerjw for reviewing my pull request and clearing my doubt :-)

d-dixon, 2021-01-24:
Trying to get a pull request

lov-1234, 2021-01-25:
Hi, Is this issue still open? I just dived into Open-Source today and would like to contribute to a project in any way I can. I figured this might be a good starting place, and therefore, I would like to request to work on this issue. May I be assigned this issue if it's still available?

AdilGM, 2021-01-30:
Hey guys, thank you for creating a space for beginners! I would like to claim this issue but as lov-1234 has asked prior to my comment maybe I can help them if needed?

rjtsingh, 2021-02-05:
I want to work on this project for practice

rjtsingh, 2021-02-08:
Review it.

dudugvili, 2021-04-27:
Hi, would like to work on this project if it's still open.

rjtsingh, 2021-04-27:
> Hi, would like to work on this project if it's still open.

Sure, sir.

rjtsingh, 2021-04-27:
> Hi, would like to work on this project if it's still open.

Sure sir, but i have only basic knowledge of  C and C++ language.

mansoorahmedmemon, 2021-05-31:
Hello, Can I claim this issue?

paritoshprogram, 2021-08-28:
Hi I am new to open source , I would love to work on this issue if it's open . so please give me an opportunity

void-hr, 2021-10-05:
Hello, I am also new to open source and want to contribute to this issue. If it's open kindly allow me to do my part.

tylerjw, 2021-10-05:
Hello everyone, I'm going to close this issue as the new-timer issue was an effort that died.  The primary reason it died is creating the issues in a way that was productive for the project proved too difficult.  Secondly, because of the nature of moveit, working on the project as a really new person proved really difficult.  Thank you everyone who commented here.  I would encourage every one of you, if you are interested in MoveIt, to go do the tutorials.  After you do that, try to build something with moveit.  That experience is what you need in order to contribute.  You will find things that are broken or edges of the project that are rough and then you'll be much better equipped to contribute.

closed 2021-10-05 by tylerjw