moveit/moveit #2425: Fix typo in first-timer-only issue template

tylerjw, 2020-11-20:
## Background

The first-timer-only issue template has a typo.  'timers' vs 'timer'.  This has had the sideafect of new 'first-timer-only' issues not receiving the 'first-timer-only' label when they are created.

## Instructions
Hi, this is a `first-timer-only` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-planning/moveit/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://moveit.ros.org/install/source/)

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_moveit/src/moveit`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `moveit` folder before cloning your own fork)

- [ ] 📝 **Update** the file(s) `.github/ISSUE_TEMPLATE/first_timers_only.md`by applying the change shown in the diff below.  Note that this also includes renaming the file to `first_timer_only.md`

```diff
diff --git a/.github/ISSUE_TEMPLATE/first_timers_only.md b/.github/ISSUE_TEMPLATE/first_timer_only.md
similarity index 98%
rename from .github/ISSUE_TEMPLATE/first_timers_only.md
rename to .github/ISSUE_TEMPLATE/first_timer_only.md
index c85e0638a..57cdf54df 100644
--- a/.github/ISSUE_TEMPLATE/first_timers_only.md
+++ b/.github/ISSUE_TEMPLATE/first_timer_only.md
@@ -1,8 +1,8 @@
 ---
-name: First Timers Only Issue
+name: First Timer Only Issue
 about: Create an issue to welcome a new contributor into the community.
 title: ''
-labels: first-timers-only
+labels: first-timer-only
 assignees: ''
 
 ---
```

- [ ] 🤖 **Apply `clang-format-10`** auto formatting, [following these instructions](https://moveit.ros.org//documentation/contributing/code/?#clang-format-auto-code-formatting)

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.
2. If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [MoveIt FAQ](https://moveit.ros.org/documentation/faqs/)
* [MoveIt Tutorials](https://ros-planning.github.io/moveit_tutorials/)
* [MoveIt contribution guide](https://moveit.ros.org/documentation/contributing/)
* [ROS Tutorials](https://wiki.ros.org/ROS/Tutorials)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Aman-Makkar, 2020-11-20:
Hello, I am relatively new to open source and haven't worked on any large projects in the past. I would like to make a start now and want to claim this task :D

tylerjw, 2020-11-20:
> Hello, I am relatively new to open source and haven't worked on any large projects in the past. I would like to make a start now and want to claim this task :D

It is yours, enjoy!

Aman-Makkar, 2020-11-21:
Thanks Tyler, I am having an issue with downloading and installing ROS. I am not sure why, I have followed all the steps.

Prakhar030, 2020-11-23:
Hi @Aman-Makkar ,  
What issue are you facing in downloading and installing ? I maybe able to help you as I recently completed an issue which resembled this issue. Please can explain the issue.

Aman-Makkar, 2020-11-23:
Hello @Prakhar030 ,
Thank you for your reply, please find the error the issue that I am having below:
C:\Windows\System32>mkdir c:\opt\chocolatey

C:\Windows\System32>set ChocolateyInstall=c:\opt\chocolatey

C:\Windows\System32>choco source add -n=ros-win -s="https://aka.ms/ros/public" --priority=1
Chocolatey v0.10.15
Directory 'c:\opt\chocolatey\lib' does not exist.
Added ros-win - https://aka.ms/ros/public (Priority 1)

C:\Windows\System32>choco upgrade ros-noetic-desktop_full -y --execution-timeout=0
Chocolatey v0.10.15
Directory 'c:\opt\chocolatey\lib' does not exist.
Upgrading the following packages:
ros-noetic-desktop_full
By upgrading you accept licenses for the packages.
ros-noetic-desktop_full is not installed. Installing...
Error retrieving packages from source 'https://aka.ms/ros/public':
 The underlying connection was closed: An unexpected error occurred on a send.
ros-noetic-desktop_full not installed. The package was not found with the source(s) listed.
 Source(s): 'https://aka.ms/ros/public;https://chocolatey.org/api/v2/'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.

Chocolatey upgraded 0/1 packages. 1 packages failed.
 See the log for details (c:\opt\chocolatey\logs\chocolatey.log).

Failures
 - ros-noetic-desktop_full - ros-noetic-desktop_full not installed. The package was not found with the source(s) listed.
 Source(s): 'https://aka.ms/ros/public;https://chocolatey.org/api/v2/'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more
 assistance.

tylerjw, 2020-11-25:
Hello @Aman-Makkar,

Workspace setup issues can be frustrating, I'm sorry you are having issues.  However, none of the maintainers here use Windows as their primary development enviroment so we'll have difficulty helping you with the issues you are experiencing.  Windows support (and those isntructions you found) is something that was contributed by Microsoft some time ago, however I don't believe anyone is actively using them or maintaining it.  The easiest way for you to use moveit is going to be on ubuntu as that is what the vast majority of ROS robotics development is using.  Ubuntu is the only platform that is tested and used daily by others here.  I'm sorry I can't be of more help with Windows.

felixvd, 2020-11-26:
[ROS Answers](https://answers.ros.org/questions/) may help if you can't switch to Ubuntu. That would be the easier option, however.

pawanw17, 2020-12-03:
Hello, I'd like to work on this issue, I haven't contributed to moveit until now and I think this would be a good starting point.

pawanw17, 2020-12-03:
Hey @tylerjw I have submitted a PR.

Dharmesh78, 2020-12-28:
Hey @tylerjw  is this issue still open. I would like to contribute as this would be my first attempt in contribution.

Devansh0106, 2021-01-02:
Hello
Is this issue still open? I would like to claim this task.

Samiatrix, 2021-01-11:
I'm new to Open Source Contribution, I would like to claim this task

tylerjw, 2021-01-11:
This issue is already claimed and there is a PR that fixes this issue.  The reason this and the PR is still open is issues with our CI.  Once those are resolved that PR will be accepted and this will be closed.  I encourage @Dharmesh78 @Devansh0106 @Samiatrix to reach out on Discord and I can try to help you find something else to contribute.

Dharmesh78, 2021-01-11:
@tylerjw how to find you on discord??... i would love to put my hands on in contributions :)

Anurag331, 2021-01-20:
I am Completely new to open source and i would love to claim this task??
can i start working on it without claiming it?

gavanderhoorn, 2021-01-20:
@tylerjw explained in [his comment](https://github.com/ros-planning/moveit/issues/2425#issuecomment-757987162) (two above yours):

> This issue is already claimed and there is a PR that fixes this issue.

so it would seem this particular issue has already been dealt with.

I'd encourage you to find something else. There should be more issues available on the tracker.

tylerjw, 2021-01-20:
> @tylerjw how to find you on discord??... i would love to put my hands on in contributions :)

The "Get Involved" button on our website has links to all the various ways to contact us, including our Discord server.  https://moveit.ros.org/about/get_involved/

The invite link to the discord server is: https://discord.gg/RrySut8

ADHIKSHA, 2021-02-04:
Hey even I would love to contribute.

closed 2021-02-04 by tylerjw