moveit/moveit #2431: MoveIt Melodic Release (1.0.7)

tylerjw, 2020-11-25:
# MoveIt Melodic Release (1.0.7)

[_moveit.ros.org release documentation_](https://moveit.ros.org/documentation/contributing/releases/)

### Checklist
- [x] Merge backports
- [ ] Run ROS buildfarm prerelease test
- [x] Update changelogs
- [ ] Bloom
- [ ] Write discourse post

## Meta

This issue is to capture the discussion that is happening over various issues relating to upcomming melodic release.  On the [backports PR](https://github.com/ros-planning/moveit/pull/2391) we are discussing the failures in the prerelease tests.

[Here is the issue asking for the release](https://github.com/ros-planning/moveit/issues/2369).  I am creating this issue to track the work of doing the release.

## Pre-release test failures

`moveit_opw_kinematics_plugin` has three tests that fail to run in the pre-release.  I believe this has to do with how `moveit_resources` changed.  I posted [here in an issue on the downstream repo](https://github.com/JeroenDM/moveit_opw_kinematics_plugin/issues/66) about why.  @JeroenDM  I don't think this should block the release as this was broken before the release.  I think the issue there is just that the dependencies need to be updated to point to the parts of the resources package needed for testing and not the meta package.

A bunch of `jsk_pcl_ros_utils` tests are failing in the pre-release test.  @knorth55 commented in the backports issue "does jsk_pcl_ros matter pre-release test of moveit?"  That is my question.  I don't know anything about this package, but I will try testing it locally with or without the latest changes in melodic-devel to see if these tests failing is caused by the backports.

tylerjw, 2020-11-25:
I built a workspace for `jsk_pcl_ros` and tried running the failing tests and I'm having a bunch of "Error opening file" issues with them when I use rostest to run them.  I wasn't able to run them with either the released version of moveit on melodic or the latest in melodic-devel.  @knorth55 do you have instructions for how I could validate that the latest changes in moveit are not breaking things for `jsk_pcl_ros` or do you have another way you'd like me to test these changes?

knorth55, 2020-11-25:
@tylerjw How can I reproduce your build error? Do you just build `jsk_pcl_ros` with latest `moveit@melodic-devel`? Can you share `.rosinstall` of your checking workspace?

tylerjw, 2020-11-25:
Here is the .rosinstall:

```
- git:
    local-name: geometric_shapes
    uri: https://github.com/ros-planning/geometric_shapes.git
    version: melodic-devel
- git:
    local-name: moveit
    uri: https://github.com/ros-planning/moveit.git
    version: melodic-devel
- git:
    local-name: moveit_msgs
    uri: https://github.com/ros-planning/moveit_msgs.git
    version: melodic-devel
- git:
    local-name: moveit_resources
    uri: https://github.com/ros-planning/moveit_resources.git
    version: master
- git:
    local-name: moveit_tutorials
    uri: https://github.com/ros-planning/moveit_tutorials.git
    version: melodic-devel
- git:
    local-name: moveit_visual_tools
    uri: https://github.com/ros-planning/moveit_visual_tools.git
    version: melodic-devel
- git:
    local-name: rviz_visual_tools
    uri: https://github.com/PickNikRobotics/rviz_visual_tools
    version: melodic-devel
- git:
    local-name: srdfdom
    uri: https://github.com/ros-planning/srdfdom.git
    version: melodic-devel
- git:
    local-name: jsk_recognition
    uri: https://github.com/jsk-ros-pkg/jsk_recognition.git
    version: 1.2.15
```

I'm probably not setting up my enviroment correctly to test your repo.  I didn't dig into your testing infrastructure configs, I just tried to build and test that one package with all the tests that are angry in the pre-release using catkin.

knorth55, 2020-11-26:
@tylerjw 
we need to carefully install several packages `gdown`, `wget`, `chainer<7.0.0` and build the package to download sample data for the test.
I tried with your `.rosinstall` and check all the test passed.

```bash
root@1fad3ec6f745:~/catkin_ws# catkin_test_results build/jsk_pcl_ros
Summary: 184 tests, 0 errors, 0 failures, 0 skipped
```

tylerjw, 2020-11-27:
Thank you for your response and for testing that, that was really helpful.  Now I just need to figure out the other tests and I'll be ready to release for melodic.

tylerjw, 2020-12-02:
Bloom is done for this: :tada:

tylerjw, 2020-12-02:
I moved `reviewed-for-backport` from `ed3e196d2` to `e337c0808`.

tylerjw, 2020-12-02:
Posted to discourse: https://discourse.ros.org/t/moveit-melodic-1-0-7-release/17712

closed 2020-12-02 by tylerjw