moveit/moveit #2432: Moveit using wrong namespace when looking for rosservice

Chick92, 2020-11-26:
### Description

Issue with namespace.

I'm using a simulation package, which is simulating the real world counterpart - a manipulator in a cell. The simulation is all working fine, but because it replicates actual hardware, that i am not able to tinker with settings on, i'm having a problem. The robot has the namespace "iiwa". 

I've gone through the tutorials and got all that working - 
http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#start-rviz-and-movegroup-node

`roslaunch panda_moveit_config demo.launch`
`rosrun moveit_tutorials move_group_python_interface_tutorial.py`

But when i make alterations to the move_group_python_interface_tutorial.py file, to work with the iiwa robot, i get the following - 

`waitForService: Service [/get_planning_scene] has not been advertised, waiting`

This service is up and running, however it is under the iiwa namespace, i.e /iiwa/get_planning_scene

I've configured the group name etc in the node - 

`super(MoveGroupPythonIntefaceTutorial, self).__init__()

    moveit_commander.roscpp_initialize(sys.argv)
    rospy.init_node('move_group_python_interface_tutorial', anonymous=True)

    robot = moveit_commander.RobotCommander()

    scene = moveit_commander.PlanningSceneInterface()

    group_name = "/iiwa"
    move_group = moveit_commander.MoveGroupCommander(group_name)

    display_trajectory_publisher = rospy.Publisher('/iiwa/move_group/display_planned_path',
                                                   moveit_msgs.msg.DisplayTrajectory,
                                                   queue_size=20)`

But can't seem to get the namespace to change. I've also looked through the moveit code base. but can't find where it is passing the service name, if that makes sense. 

I'm sure i'm missing something really obvious, as this seems like a pretty daft problem, i just can't work out what i'm missing. Unfortunately i can't just get rid of the namespace, as the physical robot is not mine, and can't be modified. 

Thanks for your help! 

Ben

### Your environment
* ROS Distro: [Kinetic|Melodic|Noetic] - Melodic
* OS Version: e.g. Ubuntu 18.04 - UB18.04
* Source or Binary build? - Source
* If binary, which release version? 
* If source, which branch? Melodic latest

### Steps to reproduce
http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#start-rviz-and-movegroup-node

with different robot, real world or simulated in gazebo

### Expected behaviour
should look for rosservice "iiwa/get_planning_scene"

### Actual behaviour
looks for rosservice "/get_planning_scene"

### Backtrace or Console output
`waitForService: Service [/get_planning_scene] has not been advertised, waiting`

closed 2020-11-27 by Chick92

felixvd, 2020-11-27:
Can you post how you fixed your problem, for the benefit of whoever searches this in the future?

Chick92, 2020-11-27:
Hi, 

Yeah sorry, i'd kinda fixed that problem, but it caused more so was waiting till i'd sorted that first. Which i've now done. 

The `move_group = moveit_commander.MoveGroupCommander(group_name, ns='/iiwa')` has a name space argument, which i've used here. However, that then caused problems with it not being able to find the group name, which was again caused by lack of name space. 

A more elegant solution was to use a launch file and add a namespace to do that, as per this - https://github.com/IFL-CAMP/iiwa_stack/issues/180

That particular package does have a launch file included with it, however it's not used in the tutorial, the node is run with 'rosrun'.

If someone could update the tutorial to include a passage on the likely need for a namespace, and to use the launch file as a means to implement that, that would be great. I feel like this is one of those "ROSisms" where the tutorial is only applicable to doing exactly what the tutorial says and nothing more, without considerable effort and googling for errors, that could easily be mitigted with a scentence in the documentation. Otherwise though, the documentation is far beyond the standard of pretty much any ros package i've ever seen. It's coe on quite a lot since i last looked at it!

felixvd, 2020-11-27:
> that, as per this - IFL-CAMP/iiwa_stack#180

Is this an issue with the iiwa_stack documentation or the moveit tutorials?
Either way if you submit a PR now while your suffering is still fresh on
your mind I’ll review it right away. I had issues with namespaces when I
started using both packages but didn’t write anything before i forgot what
gave me trouble.

gavanderhoorn, 2020-11-27:
> That particular package does have a launch file included with it, however it's not used in the tutorial, the node is run with 'rosrun'.

`rosrun` can also be used to start nodes in a namespace:

```
ROS_NAMESPACE=iiwa rosrun ..
```

you should not need to use a `.launch` file for that alone.

See [ROS/EnvironmentVariables: ROS_NAMESPACE](http://wiki.ros.org/ROS/EnvironmentVariables#ROS_NAMESPACE).

krysc0, 2021-12-07:
Using ROS_NAMESPACE= <namespace> rosrun ...

fixed my problem. Thanks!