moveit/moveit #2436: Linear Y twist command with MoveIt Servo and 4DoF arm

dudasdavid, 2020-11-28:
### Description

I'm trying to use the MoveIt Servo with an OpenMANIPULATOR-X which is a 4DoF manipulator from ROBOTIS.
I modified the open_manipulator_controller to listen to the JointTrajectory messages from moveit_servo so communication works properly. When sending linear x or z commands the manipulator works as expected. When sending linear y twist command moveit_servo cannot move the arm. With a 4 DoF manipulator, obviously, I don't expect a clean y translation, but moveit_commander is able to handle such a request in compute_cartesian_path, so I expect a similar behavior here.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Tried both master and melodic-devel branch from source

### Steps to reproduce
You can reproduce the issue using Gazebo.

Use the moveit-servo-experiment branch from my fork for the modified open_manipulator_controller:
https://github.com/dudasdavid/open_manipulator/tree/moveit-servo-expriment

My config file for moveit_servo is:
https://gist.github.com/dudasdavid/61d815a318760cf71f7eb7168a263ed5

- roslaunch open_manipulator_gazebo open_manipulator_gazebo.launch
- roslaunch open_manipulator_controller open_manipulator_controller.launch use_platform:=false use_moveit:=true
- roslaunch moveit_servo spacenav_cpp.launch (without input device)
- send twis command:
```
rostopic pub -r 100 -s /servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto
twist:
  linear:
    x: 0.0
    y: 0.1
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```

### Expected behaviour
Similar behavior as compute_cartesian_path of moveit_commander for a 4DoF manipulator.

### Actual behaviour
X and Z linear twist commands are fine, but Y doesn't work as expected.

AndyZe, 2020-11-29:
I took a shot at reproducing the issue but got this error upon `roslaunch moveit_servo spacenav_cpp.launch`:

```
ros.rosconsole_bridge.console_bridge: No name given for the robot.
ros.moveit_ros_planning.rdf_loader: Unable to parse URDF from parameter '/robot_description'
```

Do you see any warnings in terminal about singularities? If so, decrease `lower_singularity_threshold` in the yaml.

You can also allow one or more dimensions to "drift" with a service call like this. I would recommend allowing two drift dimensions because the robot is underactuated by 2 DOF. This example will allow small pitch and yaw but should allow very accurate y-translation.

```
rosservice call /servo_server/change_drift_dimensions "drift_x_translation: false
drift_y_translation: false
drift_z_translation: false
drift_x_rotation: false
drift_y_rotation: true
drift_z_rotation: true
transform_jog_frame_to_drift_frame:
  translation: {x: 0.0, y: 0.0, z: 0.0}
  rotation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}"
```

AndyZe, 2020-11-29:
I can see from the Gazebo sim that it's just not possible for the robot to translate along Y without some rotation. I'm not sure how Cartesian planning gets around that. Maybe loose tolerances? Or allowing a joint jump from one waypoint to the next?

[image]

dudasdavid, 2020-11-29:
Thank you for your answer! I've no idea why do you have the above error, it was tested on more systems, I installed open_manipulator_gazebo with apt-get, only the controller and a few other nodes are running from my fork. The colors are also missing on your model:
[image]

I didn't need any trick to make Cartesian planning work, it was working out of the box, and it can execute 100% of trajectories, see [my video](https://www.youtube.com/watch?v=_nlq3crDGWU), I'm using only Cartesian planning here.

I tried setting drift for y and z rotation with the service call, and it works for the y translation until it reaches y = 0 (middle), I get a `Very close to a singularity, emergency stop` warning there and servoing stops, of course, I can move it manually using the interactive marker of MoveIt so it's not a singularity of this arm. I tried to increase the singularity threshold without any luck. Do you have an idea of how could I adjust singularity detection for this arm?

AndyZe, 2020-11-29:
Well, unfortunately there's nothing to be done other than what I already wrote. Increase the singularity thresholds or make the service call to enable drift. I think you will be pleasantly surprised by how "drift" helps.

AndyZe, 2020-11-29:
Try enabling drift for all three degrees of rotation +till, pitch, and yaw).

dudasdavid, 2020-11-29:
Interestingly, setting the drift on x rotation solved the singularity issue. It's great that moveit_servo can work with such a low DoF arm after setting the drift, thanks for the help, I close this issue!

closed 2020-11-29 by dudasdavid

AndyZe, 2020-11-29:
Cool video!

JeremieBourque1, 2021-03-20:
Hi @dudasdavid, I am also trying to get moveit_servo working for the open_manipulator 4-dof arm. I tried to use your setup but I couldn't get it working for odd reasons.
I have made my own setup by using only the open_manipulator_description package and configured it with moveit along with your ikfast plugin. My setup can be found in this repo: git@github.com:JeremieBourque1/open_manipulator.git
Currently the moveit configuration of the arm has two controllers: 

1. A JointTrajectoryController for the motion planner, which works without any problems
2. A JointGroupPositionController which is used for moveit_servo, just like in [this tutorial](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html) 

The problem I get is that when I use moveit_servo, the arm won't move, it only shakes in place, no matter what twist command I send it. Do you have any idea what could cause this problem?

Also, second unrelated wueation, what iktype did you use to generate you ikfast plugin?

Thank you for your help

### How to reproduce the bug ###

 1. I have put all of the necessary files in this repo which you can clone: https://github.com/JeremieBourque1/open_manipulator
 2. Start MoveIt along with the gazebo simulation `roslaunch open_manipulator_moveit_config demo_gazebo.launch`
 3. Switch to the joint_group_position_controller, which is used for moveit_servo instead of the arm_controller wich is used by the motion planner. `rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_position_controller']
stop_controllers: ['arm_controller']
strictness: 0
start_asap: false
timeout: 0.0"`
 4. Launch the servo server. If you look closely in gazebo you should se the robot shaking `roslaunch open_manipulator_moveit_config spacenav_cpp.launch`
5. Publish a twist command. You should see the arm shaking in the direction specified in the twist command. 
    `rostopic pub -r 100 -s /servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto
    twist:
      linear:
        x: 0.0
        y: 0.01
        z: -0.01
      angular:
        x: 0.0
        y: 0.0
        z: 0.0"`

### Additional information ###

 * The two controller are specified in https://github.com/JeremieBourque1/open_manipulator/blob/master/open_manipulator_moveit_config/config/ros_controllers.yaml
 * The spacenav configuration file is located here: https://github.com/JeremieBourque1/open_manipulator/blob/master/open_manipulator_moveit_config/config/open_manipulator_simulated_config.yaml