moveit/moveit #2442: RobotState::setJointGroupPositions: Check for matching size of JointModelGroup and passed vector

simonschmeisser, 2020-12-02:
https://github.com/ros-planning/moveit/blob/bdd4bde4938c7bdac3a57a791cf738146c8834e0/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L625

would be nice to have an assert here and might have hinted to https://github.com/ros-planning/moveit_visual_tools/issues/82  and/or similar bugs in our products code

Will open PR

pawanw17, 2020-12-04:
Hey I'd like to work on this issue, can you guide me for the same?

simonschmeisser, 2020-12-04:
A small improvement would be to have an `assert` just like you can find it in a few places in the file I linked already.

The real solution would be to create a new function `setActiveJointGroupPositions` that takes the values for only the active joints as this would make the API a lot less error-prone.

pawanw17, 2020-12-04:
Hey @simonschmeisser should I add both the assertions?  
i.e from the link you provided
``
if (ee_jmg == nullptr)
`` if (!ee_joint_pos.empty())
and
``
 if (!ee_joint_pos.empty())
``
and should I make the pull request to melodic-devel?

PR #2456 (2020-12-31): Provide a function to set the position of active joints in a JointModelGroup
### Description

this fixes #2442 and provides an less error prone alternative for https://github.com/ros-planning/moveit_visual_tools/pull/83

The general problem is that while usually you only concern yourself with the values of active joints when storing positions in a vector, the function `setJointGroupPositions` actually expects a vector/array containing the variables for *all* joints (including mimic joints). Ignoring this will lead to reading out-of-bounds memory at least and to a mismatch at worst (yes, there  [... cut]
commit 59a37248 (2020-12-31): Provide a function to set the position of active joints in a JointModelGroup (#2456)

* RobotState::setJointGroupPositions: assert correct size of  vector

fixes #2442

* setJointGroupActivePositions sets only the positions of active joints

* Add overloads

* fix asserts

* implement JointModelGroup::getActiveVariableCount
closed 2020-12-31 by AndyZe