moveit/moveit #2444: Fix typos in plan_execution.cpp and plan_with_sensing.cpp

sjahr, 2020-12-04:
## Background
Two source files of the moveit_plan_execution package contain small typos which need to be fixed.

## Instructions
Hi, this is a `first-timer-only` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-planning/moveit/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://moveit.ros.org/install/source/)

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_moveit/src/moveit`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `moveit` folder before cloning your own fork)

- [ ] 📝 **Update** the files `plan_execution.cpp`  and ` plan_with_sensing.cpp` by applying the change shown in the diff below.

```diff
diff --git a/moveit_ros/planning/plan_execution/src/plan_execution.cpp b/moveit_ros/planning/plan_execution/src/plan_execution.cpp
index e7657122d..f4ef67a93 100644
--- a/moveit_ros/planning/plan_execution/src/plan_execution.cpp
+++ b/moveit_ros/planning/plan_execution/src/plan_execution.cpp
@@ -220,7 +220,7 @@ void plan_execution::PlanExecution::planAndExecuteHelper(ExecutableMotionPlan& p
       if (preempt_requested_)
         break;
 
-      // execute the trajectory, and monitor its executionm
+      // execute the trajectory, and monitor its execution
       plan.error_code_ = executeAndMonitor(plan);
     }
 
diff --git a/moveit_ros/planning/plan_execution/src/plan_with_sensing.cpp b/moveit_ros/planning/plan_execution/src/plan_with_sensing.cpp
index 0078ad557..1ef314727 100644
--- a/moveit_ros/planning/plan_execution/src/plan_with_sensing.cpp
+++ b/moveit_ros/planning/plan_execution/src/plan_with_sensing.cpp
@@ -150,7 +150,7 @@ bool plan_execution::PlanWithSensing::computePlan(ExecutableMotionPlan& plan,
   unsigned int look_attempts = 0;
 
   // this flag is set to true when all conditions for looking around are met, and the command is sent.
-  // the intention is for the planning looop not to terminate when having just looked around
+  // the intention is for the planning loop not to terminate when having just looked around
   bool just_looked_around = false;
 
   // this flag indicates whether the last lookAt() operation failed. If this operation fails once, we assume that
@@ -212,7 +212,7 @@ bool plan_execution::PlanWithSensing::computePlan(ExecutableMotionPlan& plan,
 
       bool looked_at_result = lookAt(cost_sources, plan.planning_scene_->getPlanningFrame());
       if (looked_at_result)
-        ROS_INFO("Sensor was succesfully actuated. Attempting to recompute a motion plan.");
+        ROS_INFO("Sensor was successfully actuated. Attempting to recompute a motion plan.");
       else
       {
         if (look_around_failed)

```

- [ ] 🤖 **Apply `clang-format-10`** auto formatting, [following these instructions](https://moveit.ros.org//documentation/contributing/code/?#clang-format-auto-code-formatting)

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.
2. If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github) on how to send a pull request using the terminal.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [MoveIt FAQ](https://moveit.ros.org/documentation/faqs/)
* [MoveIt Tutorials](https://ros-planning.github.io/moveit_tutorials/)
* [MoveIt contribution guide](https://moveit.ros.org/documentation/contributing/)
* [ROS Tutorials](https://wiki.ros.org/ROS/Tutorials)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

vlasove, 2020-12-07:
Hi, there. Can I start working on this fix?

Lokesh-Jawale, 2020-12-08:
I am new to open source and will like to contribute this issue.
I will like to work on this.

Lokesh-Jawale, 2020-12-08:
Can anyone please guide me on how to build from source in fedora.
There's no documentation about it.

sjahr, 2020-12-08:
Hi @Lokesh-Jawale,
great to hear that you are interested in contributing!
You should be able to build MoveIt 1 from source, when you follow the [usual instructions from the MoveIt page](https://moveit.ros.org/install/source/), although (depending on your Fedora version) it is likely that you also need to build ROS from source. Unfortunately, I have no experience with Fedora so in case you get stuck at some point with some OS related problems, I recommend you to seek help on [ROS Answers](https://answers.ros.org/questions/),
Let me know if you have problems with the build-from-source instructions :)

StutiV, 2020-12-11:
Hello @sjahr , can you assign me this issue? I'd like to contribute.

Udbhavbisarya23, 2020-12-13:
Hey @sjahr I would like to take this issue if its still open. Could i be assigned to this issue?

sjahr, 2020-12-15:
> Hello @sjahr , can you assign me this issue? I'd like to contribute.

Hi @StutiV, sure feel free to open a PR :) If you need any help just mention me in your PR or you could ask @Udbhavbisarya23 to support you.

Udbhavbisarya23, 2020-12-16:
That sounds great @sjahr! @StutiV Please let me know if you need any help

mail-man26, 2020-12-18:
Hi! @sjahr  I am new to open source and would like to get started with this issue, would you like to assign me this issue?

sjahr, 2020-12-21:
@Udbhavbisarya23 you can also start a PR if you want :) and request @StutiV's help in case you need any.

Udbhavbisarya23, 2020-12-21:
Sounds great @sjahr! I'll get started on it!

PR #2466 (2020-12-22): Fix typos in plan_execution.cpp and plan_with_sensing.cpp
### Description

Fixed the typos in plan_execution.cpp and plan_with_sensing.cpp. This pull request resolves #2444 .

[//]: # "You can expect a response from a maintainer within 7 days. If you haven't heard anything by then, feel free to ping the thread. Thank you!"
closed 2020-12-22 by rhaschke