moveit/moveit #2445: Moveing fast and breaking moveit, meeting summary

tylerjw, 2020-12-04:
This issue is for discussing what the summary of our meeting about the future of MoveIt is and what we should post to discourse.  Here is my draft of the summary.  Please correct what I've missed, misworded, etc:

```
MoveIt has grown in many ways, some good and some bad.  We are worried that this growth has resulted in some stagnation and difficulty in making large changes that would add value.  To that end the maintainers of MoveIt met and made this plan to spur future development:
* Do exercise where we design MoveIt with our experience as if it doesn’t exist.  This should educate a roadmap of features, refactors.
* A drive to simplify and clarify the interfaces and levels of abstraction.  We recognize the library has grown in complexity to the point where none of us understand it fully.
* Defining what is MoveIt to educate what belongs and what doesn't.
* Discuss in GitHub issues parts of MoveIt we are confused about why they are part of the library and consider removing them.
```

nbbrooks, 2020-12-04:
I think this is a great summary, although I would add a more explicit bullet for discussing removal of features that are sufficiently superseded by newer features.

tylerjw, 2020-12-07:
Without any objections I'm going to post this to discourse later today.  Then we can move onto the fun stage of actually doing these things.

rhaschke, 2020-12-07:
For me, the central point is not removing features to reduce complexity. But, rather to define (and document!) cleaner interfaces that allow to decouple different API layers of MoveIt thus facilitating understanding of each of them.

tylerjw, 2020-12-07:
I adjusted the points to reflect both points.  I'll post it now.

closed 2020-12-07 by tylerjw