moveit/moveit #2447: setGoalTolerance() does not work in ros-noetic the way it worked in melodic.

deep0105, 2020-12-07:
### Description

I recently upgraded my robotic application based on moveit to ros-noetic. The same planner that worked earlier and planned seamlessly in melodic is now failing in ros-noetic. On debugging further, I found that  **setGoalTolerance()** of move_group node is making the moveit planner failed. It was set to _1e-6_ in melodic for better accuracy, but the same value does not work in noetic. I tried setting the setGoalJointTolerance(), setGoalOrientationTolerance(), setGoalPositionTolerance() separately to 1e-6 and after trying some combinations, I found that setGoalOrientationTolerance() was working for a minimum value of _0.5_ tolerance and setGoalJointTolerance(), setGoalPositionTolerance() worked perfectly with _1e-6_ . But I don't want to set _0.5_ in setGoalOrientationTolerance(), it's a very high value. 

My default planner is RRTConnect and I tried different planners(RRTstar, RRT, EST etc) but that didn't solved the problem. 

My question is has there been any change with respect to setGoalTolerance() in noetic moveit?

Thanks a lot. 

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? binary
* If binary, which release version? 
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

felixvd, 2020-12-09:
Could be related to https://github.com/ros-planning/moveit/pull/2434

JafarAbdi, 2021-02-03:
@deep0105 Could you please check if this still an issue .?

deep0105, 2021-03-17:
> @deep0105 Could you please check if this still an issue .?

@JafarAbdi, at some point I shifted from pose target to joint target as a workaround. But I will check once and let you know. Thanks

cschindlbeck, 2023-01-25:
I have the same or a similar problem, set_goal_orientation_tolerance does not seem to work. We are using a 5 DOF robot, is this parameter only supported for 6 DOF robots?

v4hn, 2023-01-25:
@cschindlbeck I'm not convinced this is the same issue as the OP confirmed it fixed their problem.
Please create a new issue with more details (best case a minimal example) to track this.

cschindlbeck, 2023-01-31:
Ok, i posted this issue here: https://github.com/ros-planning/moveit/issues/3325 with a containerized minimal ws example