moveit/moveit #2449: pick place flaky test

tylerjw, 2020-12-08:
### Description

Sometimes the test fails to plan to the initial grasp:
[image]

Here is the output when this happens:
```bash
You can start planning now!

ros.moveit_core.planning_scene: Adding planning scene diff
ros.moveit_ros_planning.plan_execution: Planning attempt 1 of at most 1
ros.moveit_ros_manipulation.manipulation: Start thread 0 for 'pick'
ros.moveit_ros_manipulation.manipulation: Start thread 3 for 'pick'
ros.moveit_ros_manipulation.manipulation: Start thread 2 for 'pick'
ros.moveit_ros_manipulation.manipulation: Added plan for pipeline 'pick'. Queue is now of size 1
ros.moveit_ros_manipulation.manipulation: Start thread 1 for 'pick'
ros.moveit_core.kinematic_constraints: The desired rotation matrix for link 'panda_link8' in frame world is:
          0 1.66533e-16           1
  -0.707107   -0.707107 1.66533e-16
   0.707107   -0.707107           0
ros.moveit_ros_manipulation.manipulation: Manipulation plan 0 failed at stage 'reachable & valid pose filter' on thread 0
ros.moveit_ros_manipulation.manipulation: All supplied grasps failed. Retrying last grasp in verbose mode.
ros.moveit_ros_manipulation.manipulation: Start thread 1 for 'pick'
ros.moveit_ros_manipulation.manipulation: Start thread 0 for 'pick'
ros.moveit_ros_manipulation.manipulation: Start thread 2 for 'pick'
ros.moveit_ros_manipulation.manipulation: Start thread 3 for 'pick'
ros.moveit_ros_manipulation.manipulation: Re-added last failed plan for pipeline 'pick'. Queue is now of size 1
ros.moveit_core.kinematic_constraints: The desired rotation matrix for link 'panda_link8' in frame world is:
          0 1.66533e-16           1
  -0.707107   -0.707107 1.66533e-16
   0.707107   -0.707107           0
ros.moveit_core.kinematic_constraints: Orientation constraint violated for link 'panda_link8'. Quaternion desired: 0.653281 -0.270598 0.653281 -0.270598, quaternion actual: 0.653494 -0.272013 0.654656 -0.265294, error: x=3.133457, y=3.136414, z=3.135674, tolerance: x=0.010000, y=0.010000, z=0.010000
ros.moveit_core.kinematic_constraints: Orientation constraint violated for link 'panda_link8'. Quaternion desired: 0.653281 -0.270598 0.653281 -0.270598, quaternion actual: 0.649724 -0.273986 0.655043 -0.271493, error: x=3.141187, y=3.133295, z=3.135022, tolerance: x=0.010000, y=0.010000, z=0.010000
ros.moveit_ros_manipulation.manipulation: Sampler failed to produce a state
ros.moveit_ros_manipulation.manipulation: Manipulation plan 0 failed at stage 'reachable & valid pose filter' on thread 1
ros.moveit_ros_manipulation.manipulation: Pickup planning completed after 0.085019 seconds
ros.moveit_core.robot_state: Trying to get marker for link 'panda_hand'
ros.moveit_core.robot_state: Trying to get marker for link 'panda_leftfinger'
ros.moveit_core.robot_state: Trying to get marker for link 'panda_rightfinger'
ros.moveit_ros_planning_interface.move_group_interface: Fail: ABORTED: No motion plan found. No execution attempted.
[Testcase: testmove_group_pick_place_test] ... ok

[ROSTEST]-----------------------------------------------------------------------

[moveit_ros_planning_interface.rosunit-move_group_pick_place_test/PickPlaceTest][FAILURE]
/home/tyler/code/moveit/ws_moveit/src/moveit/moveit_ros/planning_interface/test/move_group_pick_place_test.cpp:222
      Expected: move_group_->pick("object", grasps)
      Which is: 4-byte object <FF-FF FF-FF>
To be equal to: moveit::planning_interface::MoveItErrorCode::SUCCESS
      Which is: 1
--------------------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1

rostest log file is in /home/tyler/.ros/log/rostest-eboladrome-7130.log
```

tylerjw, 2020-12-08:
Is there a way to ask it to sample more states, increase the timeout for sampling states, or increase the thresholds?

rhaschke, 2020-12-08:
Hi Tyler, looking into the build history on Travis, I think that this test isn't flaky but really broken since #2414.
Before: https://travis-ci.com/github/ros-planning/moveit/builds/201856055
After: https://travis-ci.com/github/ros-planning/moveit/builds/202392619
Note the failing test on Noetic. The clang-tidy-fix just times out, which is unrelated.

rhaschke, 2020-12-08:
It is strange that the test succeeds on the same source in Melodic, but (consistently) fails on Noetic. Can you reproduce this locally?

tylerjw, 2020-12-08:
I can reproduce this failure about 10% of the time on Melodic locally.

tylerjw, 2020-12-08:
Good point about that PR.  I'll submitted a PR that seems to work around the problem locally for me.  However, if it is the pose that should be adjusted, it should be adjusted in the tutorials too as this example was pulled from the tutorials.

rhaschke, 2020-12-08:
As this regression was introduced by #2414, I would like @stuart-fb to have a look at this problem. 
Your PR #2450 is just a workaround. A unit test should succeed deterministically! For the time being, I would prefer disabling the test over such a patch.

AndyZe, 2020-12-08:
#2434 should fix this

rhaschke, 2020-12-08:
Thanks @AndyZe, will have a look...

stuart-fb, 2020-12-08:
Took a look - #2434 looks like the right way to fix this to me as well.  Thanks for the detective work @JafarAbdi .

jschleicher, 2021-01-13:
#2434 was merged - so can this issue be closed?

closed 2021-01-13 by tylerjw