moveit/moveit #2451: Default velocity and acceleration scaling factor do not affect RViz interface

ShotaAk, 2020-12-08:
### Description

The values of `default_velocity_scaling_factor` and `default_acceleration_scaling_factor` in the `joint_limits.yaml` 
do not affect RViz interface.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Binary build
* If binary, which release version? `ros-noetic-moveit-core : 1.1.0-1focal.20200930.003439   amd64 `

### Steps to reproduce

1. Edit `joint_limits.yaml` like this.

```yaml
joint_limits:
  default_velocity_scaling_factor: 1.0
  default_acceleration_scaling_factor: 1.0
```
2. Launch move_group, RViz, etc..

### Expected behaviour

Scaling factors in RViz will be `1.0`.

[image]

### Actual behaviour

Scaling factors in RViz are `0.1`.

rhaschke, 2020-12-08:
The problem is your joint_limits.yaml. You need to define the scaling factors at the top level:
```yaml
joint_limits:
  panda_joint1:
    has_velocity_limits: true
    max_velocity: 2.1750
    has_acceleration_limits: true
    max_acceleration: 3.75

default_velocity_scaling_factor: 1.0
default_acceleration_scaling_factor: 1.0
```

ShotaAk, 2020-12-08:
@rhaschke Thank you for your comments.
This solution is good for RViz but not for `moveit_commander`.
RViz's motion plan be fast, `moveit_commander`'s motion plan be slow.

I checked this PR ( https://github.com/ros-planning/moveit/pull/1890/files ) , 
It seems the `move_group_interface` gets `robot_description_planning/joint_limits/default_velocity_scaling_factor` ,
but the `motion_planning_frame` gets `robot_description_planning/default_velocity_scaling_factor` parameter.

The parameter names not match.

Should I fix the `joint_limits.yaml` like this?

```yaml
default_velocity_scaling_factor: 1.0
default_acceleration_scaling_factor: 1.0

joint_limits:
  default_velocity_scaling_factor: 1.0
  default_acceleration_scaling_factor: 1.0
```

rhaschke, 2020-12-08:
Oops. Thanks for this hint. This is a real bug! Could you prepare a PR removing `joint_limits` in the `move_group_interface`, please?

ShotaAk, 2020-12-08:
Sure. :+1:

PR #2452 (2021-01-13): Fix scaling factor parameter names
### Description

Fix #2451.

Remove the `joint_limits` prefix from parameters `default_velocity_scaling_factor` and `default_acceleration_scaling_factor` that `move_group_interface`  node gets.

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [x] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [x] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit/blob/master/MIGR [... cut]
closed 2021-01-13 by rhaschke