moveit/moveit #2458: Build problem due to linker problems with `shapes::createMeshFromVertices` and `ros::NodeHandle::NodeHandle`.

FabianSchuetze, 2020-12-13:
### Description

I am trying to build moveit from source on noetic but the packages `moveit_setup_assistant` and `moveit_tutorials` fail both for the same reason. Both packages complain about undefined references, like so: 

```Shell 
/usr/bin/ld: /home/fabian/Documents/work/github/ws_moveit/devel/lib/libmoveit_semantic_world.so.1.1.1: undefined reference to `shapes::createMeshFromVertices(std::__debug::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, std::__debug::vector<unsigned int, std::allocator<unsigned int> > const&)'
/usr/bin/ld: /home/fabian/Documents/work/github/ws_moveit/devel/lib/libmoveit_semantic_world.so.1.1.1: undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/moveit_setup_assistant.dir/build.make:332: /home/fabian/Documents/work/github/ws_moveit/devel/.private/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant] Error 1
make[1]: *** [CMakeFiles/Makefile2:562: CMakeFiles/moveit_setup_assistant.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
I found this error curious because the `geometric_shapes` package build successfully and I have build with reference to `ros::NodeHandle` previously. I was following the [Getting Started](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) guide dutifully when attempting to build the package. I am trying to build moveit on ros-noetic, Ubuntu 20.04 from source. 

Does somebody have any idea what I might be doing wrong? Alternatively, does somebody have an idea which further information could prove valuable in understanding the problem? I am grateful for any hints or suggestions!

simonschmeisser, 2020-12-28:
Did you try a clean rebuild of your whole workspace?

FabianSchuetze, 2020-12-29:
Thanks for your reply, @simonschmeisser! I did try this but to no avail so far. To circumvent the problem I do not build the `setup_assistant` and the `tutorials` on `noetic`.

simonschmeisser, 2021-01-01:
Just to be certain, you deleted both `build` and `devel`? Doing this typically get's you rid of inexplicable linker errors ...

felixvd, 2021-02-19:
Is this solved? Have you tried running `rospack list` and confirming that `geometric_shapes` points to your workspace and not the `/opt/ros` one?

FabianSchuetze, 2021-02-19:
Thank you, @simonschmeisser and @felixvd  for being persistent and for providing good advice! The issue is resolved but I have not confess that I do not understand why the problem vanished.

closed 2021-02-19 by FabianSchuetze