moveit/moveit #2462: Changes to improve CI

tylerjw, 2020-12-17:
## Proposed docker image changes for faster CI
CI timeouts and build/runtimes are an ongoing problem with moveit.  After reading the documentation for the nav2 ci system I have come to appreciate a handful of things that they are doing with their docker images we could also do to improve our ci system.

In order to avoid rebuilding in ci things that are not changing often they build their upstream packages in dockerhub and only rebuild those packages in CI if they have changed since the docker image was built.  This would save rebuilding packages like moveit_msgs.  This would involve us splitting up our rosinstall file.  Separating upstream packages (those that moveit depends on) from downstream packages (those that depend on moveit like moveit_tutorials).  nav2 doesn't test any downstream packages as we do and this split would also have to be adapted into moveit_ci so that we would test both moveit and downstream packages.  To fully take advantage of this we would need to hash the workspaces as nav2 is doing and actually just skip building them when none of the dependencies or source changed from when they were built in docker.

Secondly, nav2 is building nav2 in the docker image.  If we adapted our docker images to have a similar structure we could have the images built for noetic, melodic, noetic/ros-testing and noetic/clang-debug as separate docker images that would all be based on the same "cacher" image.  Note that a big part of the time cost of our ci is doing the debug build in clang for code coverage and clang-tidy.  By doing a build in dockerhub and then doing a checkout of the pr branch in moveit_ci and build we would take advantage of cmake not rebuilding parts of moveit that didn't change.

I've started this work in a branch of mine here: https://github.com/tylerjw/moveit/tree/build_in_docker  and am testing generating the docker images in docker hub based on that branch in an account associated with myself. I will use that to also test changes in moveit_ci to make these optimizations.  I'll link it here when it is more ready for review.

### Clang-tidy timeouts
`clang-tidy` itself is timing out on some of the files in the pilz planner.  I've yet to find any good solution for this.  This isn't a timeout of doing all the clang-tidy runs but individual files are occasionally timing out.  This is why the master branch run in Travis is currently broken.  This is a separate problem (from other kinds of timeouts) that needs a technical solution.

### Transitioning to another CI provider
We could transition over to using circleci similar to nav2.  The upsides to this are that they give a very large amount of credit to open source projects and we would have 5 runners and we could use many of the fancy multistage caching features of circle ci to further speed up our CI runs.  The downside to this is that we would loose or have to adapt many of the very nice features that moveit_ci has such as only running clang-tidy on cpp files that were changed in the given or.

Another option would be to transition to github actions and industrial_ci.  The largest downsides to this are that code coverage isn't merged into industrial_ci yet and the same issue with clang-tidy.  In this case one solution could be to adapt the bash scripts in moveit_ci (like the clang-tidy test and code coverage) to be standalone bash scripts and use the hooks system in industrial_ci and put those in some directory inside moveit named something like .ci.  I've started working on this for the clang-tidy test as I'd like to be able to even use it locally.  It would be really nice to be able to run:  `.ci/clang-tidy-test.sh master` and have it run clang-tidy on files that are changed from the master branch.  This sort of script could make it easier for people to test their PRs locally for clang-tidy also.

In either of these cases as paths away from travis the changes I proposed above to how we are using docker images would apply and should help cut down on CI runtime.

### Background
TravisCI changed their pricing model and it has made CI for moveit much worse.  Before they would allow us quite a bit of time and a few runners all the time.  After the change, they cut off our CI jobs after we've used some amount of time each month.  The only way to get around this temporarily was for picknik to start paying for a dedicated runner plan.  We started out with a single runner and that proved to not be enough.  Picknik is currently paying for a two runner plan.  

## Conclusion
To fix this I think there are technical changes we could make that would enable moveit CI to run in much less time and therefore be both better for those submitting PRs and nicer to whatever CI system we use.  Secondly, if these technical changes are not enough to get good enough performance and stay on travis (transitioning back to their open source plan) we should consider other ci providers and at the same time how we would like to adapt our system (moveit_ci).  If we chose to transition to github actions we should consider using and contributing to industrial_ci.  If we choose to transition to circleci we can adapt what nav2 did and hopefully be able to keep much of the nice features in the moveit_ci bash scripts.

rhaschke, 2020-12-17:
Hi Tyler, 
thank you very much for this detailed analysis. If I understand your initial paragraph correctly, you want to shift building time from Travis to dockerhub, such that PR builds in Travis can reuse build artifacts generated in a docker image previously. Indeed, currently, we are only utilizing ccache but essentially rebuilding from scratch every time.
I think, before proceeding we should first decide on a specific CI provider. Travis, providing 2 runners only (and PickNik paying for this), isn't a good option anymore. Did you contact Travis support to ask for a free open-source plan? On their [website](https://travis-ci.com/plans), they still advertise free open-source support:
> Free for open source. We love the Open Source Community, and to show how much we love it, upon validated request placed with our Support Team you may receive free OSS credits for your public builds.

I think we should get back to 5 concurrent runners. Maybe, this is an interesting read for you as well:
https://www.jeffgeerling.com/blog/2020/travis-cis-new-pricing-plan-threw-wrench-my-open-source-works

gavanderhoorn, 2020-12-17:
> Did you contact Travis support to ask for a free open-source plan? On their [website](https://travis-ci.com/plans), they still advertise free open-source support:

I have contacted them about this (for other projects).

The response was that the OSS allotment was put on hold until further notice.

tylerjw, 2020-12-17:
I will reach out to Travis today to see if I can get an answer for us.  After reading that blog post you linked I am less hopeful we can avoid having to migrate away.  In the meantime, I'll try to put together a list of pro/con for where we could go and various approaches so we can start to discuss and make decisions.  I want to have a plan we agree on as a next step.  If the author of that blog is correct about the state of Travis we might not have much choice.

> you want to shift building time from Travis to dockerhub, such that PR builds in Travis can reuse build artifacts generated in a docker image previously

yes, essentially this is what I want to try in order to reduce the runtime of ci.  any technical changes that reduce the runtime of ci (short of reducing testing) will improve the usefulness of it on any ci provider.

rhaschke, 2020-12-17:
> Yes, essentially this is what I want to try in order to reduce the runtime of ci. any technical changes that reduce the runtime of ci (short of reducing testing) will improve the usefulness of it on any ci provider.

Fully agree. I'm inclined towards github actions. As far as I understand they allow for caching of intermediate stages. But I didn't look deeper yet. I'm afraid I need to migrate all our (university) repos as well...

nbbrooks, 2020-12-17:
Based on the information presented in the maintainer meeting today, I lean towards migrating to circleci, collaborating with nav2. 

I've used industrial_ci on a couple projects and had a good experience. However, as Tyler mentioned, if industrial_ci's goal is to be simple to use and spin up (which I found to be true), it will inherently lose some power and be oriented towards projects smaller than MoveIt. Nav2's goals and pain points are very much in line with ours and they are also open to collaborating with us, so I think circleci has the best alignment of our options.

tylerjw, 2020-12-17:
# CircleCI
These are my thoughts regarding the option of switching to CircleCI and collaborating with and learning from nav2's solution.

## Performance
Here is a link to their latest runs on circle_ci: https://app.circleci.com/pipelines/github/ros-planning/navigation2 
You can see that their CI runs are taking about an hour to run for their nightly ci run.  One thing to note is that as a ros2 project they are testing across multiple middleware implementations in their nightly test.  In the configuration that runs when someone submits a PR it is closer to 30min normally.

For our use case, we would want to test multiple ros versions (melodic and noetic) and using the two main compilers (gcc and clang).  A large upside to their approach is that the work is pipelined and cached.  That means that if after building a debug build with clang we want to both run unit tests to generate code coverage and run clang-tidy over the newly changed files those two things could be done in parallel.  Also, as stated above we could move much of the time we spend building into nightly dockerhub builds.

## Pricing (how much CI do we get for moveit)
According to their [pricing faq](https://circleci.com/pricing/):
> CircleCI will be offering organizations on our free plan 400,000 credits per month to use on medium Docker compute for open source repositories, but they can only be spent on Linux compute. Orgs building OSS Windows projects can still use the 2,500 free credits per week that all projects have access to on those projects, and orgs building OSS macOS projects can request free macOS access by contacting billing@circleci.com.

> If you are building a bigger open source project and would like the flexibility of our new plans, let us know how we can help you!

For context on how much CI that is, the nav2 project used 82k credits last month (8084 minutes of a docker-medium runner).  From my testing, there seems to be a limit of 5 concurrent jobs per project.  I've also reached out to a salesperson at circleci and was told that for large OS projects they can increase the number of credits and/or turn on some of their more advanced features.

## Tasks to migrating
If we were to choose to migrate to circle ci this is the way I propose we do it: 
1. Build a ros1 version of the config used by nav2 for moveit
2. Adapt our docker files to build the project nightly in dockerhub (I've partially got this working)
3. Adapt the lining steps (clang-tidy and clang-format) into standalone scripts that can be used locally and on ci

## Downsides
**The yaml file is huge and complex.**  If you have ventured into the nav2 ci configuration you've discovered a huge yaml file is where all of the logic is.  This is primarily because to use the advanced pipelining and caching functionality in circleci there is no other way to specify this.  This means that in the short term we'd be moving much of the logic of ci out of bash files and into yaml.  This makes the ci system less modular and harder to easily apply to other repos.  The circleci solution to this is a thing they call an orb.  [Here is the documentation](https://circleci.com/docs/2.0/orb-intro/).  Orbs are a way to standardize a workflow for a type of project.  This is something we should consider doing in the long term after we have a good understanding of what would make a nice default ros1 circleci pipeline configuration.

One other thing that mitigates the complexity of this large ci configuration file is the documentation created by @ruffsl here: https://github.com/ruffsl/navigation2/tree/ci-docs/doc/continuous_integration

**Lockin.**  By doing this we will be tightly integrating with features of CircleCI and in the future, if they go the way of Travis (or should I say, when) we'll be doing this all over again.  My only counter to this is that moveit_ci is tightly dependent on Travis features as it is and environment variables and that if we want to use pipelining and caching from any ci vender we are going to be digging the same hole.  This is best mitigated by abstracting away the ci provider into just some bash script we run.  The downside to that is the bash script won't be able to tie into any advanced features in the ci system for performance improvements.

## Future opportunities
If we go down this path we have a few future opportunities for further standardizing and simplifying CI.

**local multi-stage caching ci through docker builds**  Since docker with buildkit does nice caching and a docker build file can call bash scripts we could have one that could be launched with a single command and get a similar full ci test locally with fast caching similar to what ci does.

**standardize orbs for ros projects**  Similar to what was done for travis with moveit_ci we could create and maintain orbs that allow us to reuse the ci behavior for other ROS projects.

**build using colcon with ros1**  This one has some upsides with some major downsides.  The upsides are that we could share a ci configuration between ros1 and ros2 and we'd get the declarative ament lint system that makes adding linters to ci easier as it is defined in the package.xml and cmake files instead of by ci.  This second reason is really nice for when you want to test locally that your change will pass CI.  The obvious downsides are that we'd have to refactor our launch files and that it would be relatively non-standard among ros1 projects.  Our users might not love us if we did this.

## Conclusions
[... cut ...]  by the ci runner and the only caching you can do is either putting stuff in the docker image or sharing between entire runs.  This prevents you from doing things like saving off the full build of the upstream workspace from one run to the next and just starting ci from the first part that has a change.  With circleci if you push a new commit to your PR and no upstream packages changed it can just start with building the moveit repo after confirming nothing changed.  This can save a ton of time.

rhaschke, 2020-12-17:
If I understand you correctly, `industrial_ci` running on github actions is still a rather monolithic script. I wouldn't go for this.
As far as I have seen, github action allows modularization based on actions hosted at github as well. Hence, we could benefit from other people providing actions and we could provide our own set of actions. This seems quite similar to the concept of orbs. 
There are some interesting comparisons available online, e.g. [this one](https://medium.com/instrument-stories/using-github-actions-and-circleci-b0a30c0a2608#:~:text=Although%20CircleCI%20currently%20has%20more,config%20files%20are%20noteworthy%20conveniences.).
In my opinion we should decide based on:
- open-source support,
- (maybe) support for non-linux platforms, e.g. windows
- ease of use (from what I learned so far, Circle CI and github actions are similar to this regard)

tylerjw, 2020-12-17:
I'll respond to your third points for judging regarding CircleCI and then I'm going to research using github actions more and see if I can give you a good evaluation of that approach.  I am really liking the idea of moving away from the monolithic script for performance reasons.

## open-source support
CircleCI is currently offering a lot of ci runtime for open source projects.  I'll look into comparing this once I find the numbers for github actions.  I'm not sure what you mean other than that.  The source code for cricleci is not opensource itself, but neither is github.  If that is something we care about we'll need to look at an option like jenkins or buildbot.  The problem there is we'd need to pay for and maintain our own infrastructure.  I think that's a giant waste of time for this project so long as we can make due with one of the cloud solutions.

##  (maybe) support for non-linux platforms, e.g. windows
CircleCI does support Windows and MacOS runners.  However, they do not do this for free.  I wouldn't be surprised if some amount of windows runners were maybe free on github actions since Microsoft owns them.  I don't know and I will get back to you on this.  There is a chance CircleCI might give us windows runners if we asked.  This mostly matters for moveit2 as it is possible to build on Windows now.  This does get into a further discussion of is it worth our limited time supporting windows though.  There is an argument that me might get more users by support some flavor of RT linux instead.

## ease of use (from what I learned so far, Circle CI and github actions are similar to this regard)
For me I think this is the one that is clearly in the camp of CircleCI.  Because the nav2 team has already done much of the work of figuring out how to use pipelining and we could collaborate with them over this it would hopefully reduce the cost initially and in maintencne.

The counter to that point is github actions are native in github and might provide a nicer "ease of use" from the perspective of people submitting PRs.

rhaschke, 2020-12-17:
> CircleCI is currently offering a lot of ci runtime for open source projects. I'm not sure what you mean other than that.

That's exactly what I meant: how many parallel jobs we can run, how many free minutes we have per week/month.
I didn't find clear numbers for github actions yet. I'm curious to see what they offer as well...

> For me ease of use is clearly in the camp of CircleCI, because the nav2 team has already done much of the work...

You are probably right on this. On the other hand there is a growing set of freely available github actions as well.

> github actions might provide a nicer "ease of use" from the perspective of people submitting PRs.

I was primarily concerned about us configuring and maintaining the CI scripts.

tylerjw, 2020-12-17:
How many parallel jobs can we run?
* CircleCI: 5 per repo
* Github actions: 20 per repo

How much time per month?
* CircleCI: ~40k minutes / month (~666 hours).  This can be increased if we ask.
* Github actions: supposedly unlimited for now
Both have documentation language that says it can change at any time.
 
How does caching compare?
* CircleCI: caching can be between jobs (whole workspaces), between all similar jobs (ccache), docker layer caching if we end up building docker images on circleci because dockerhub won't put up with how much we are going to start using them (another cool circleci feature that we might want).
* Github actions: looks like it supports all the same caching features except docker layer caching (nav2 isn't using this but I think we might care about it)
Reference: https://docs.github.com/en/free-pro-team@latest/actions/learn-github-actions/migrating-from-circleci-to-github-actions

Limits and cost of storage of caches and artifacts for open source projects?
* CircleCI: no limit called out in the documentation as far as I can find, free... nav2 used 1.2tb of network transfer and 1.1tb of storage in November without being limited.
* Github actions: 5gb limit on caches.  Reference: https://docs.github.com/en/free-pro-team@latest/actions/guides/caching-dependencies-to-speed-up-workflows#usage-limits-and-eviction-policy
On caching, I'm sure there are ways we could find to use less cache than what nav2 is doing but by being able to cache entire workspaces they are able to really speed up re-runs of ci after small changes in a PR.

ssh access to runner for debugging?
* CircleCI: this feature is awesome.  I just discovered it and am not sure why every CI provider doesn't have this.  At any point, you can start a ci job with ssh turned on and copy and paste a one-liner to ssh into any job you started.  It uses the ssh key in your github account.  This could be a real lifesaver for debugging performance-related flaky tests or other things that are hard to reproduce locally.  Reference: https://circleci.com/docs/2.0/ssh-access-jobs/
* Github Actions: no :(

After looking over the documentation for both I'm favoring circleci because of the limits on caching, the opportunity to learn from and share with the nav2 team, and the debugging with ssh feature.  I was really hoping to use github actions mostly because I really like the native github experience (not using yet another tool).  Both tools face the risk of the people paying for them (investors in CircleCI, and Microsoft for Github) deciding it isn't worth supporting OSS anymore.  I'm not sure I have any good reasons to believe either would be more likely to cancel OSS support.

rhaschke, 2020-12-18:
Thanks for this great summary. Looks like CircleCI is the way to go. Indeed the ssh feature would be great to have.
Regarding caching: 5GB per repo might be fine. I wanted to check our Travis cache usage, but it's [not reported anymore](https://travis-ci.com/github/ros-planning/moveit/caches)!
I'm not sure Docker Layer Caching is a relevant feature for us. That's only important if we are building docker images on CircleCI. But we would continue to build them on dockerhub, wouldn't we? As you pointed out before, on CI we would cache build artifacts like the catkin build/devel folders, wouldn't we?
Having 20 parallel jobs on github vs. 5 on CircleCI might be a pro for github. How do they count different jobs? On Travis, each individual job is counted (and we start 4-5 per PR).

tylerjw, 2020-12-18:
I agree that docker layer caching might not matter so long as we can avoid building moveit timing out on dockerhub.  I think it's ok now with our source image so maybe it's fine.

We would be cashing build/devel folders.  That's what nav2 does.  

20 jobs seems better however one thing I've noticed on GitHub on the one client project we are using it on is jobs seem to sit in queue waiting for a while for it to run.  This might just be because everyone is migrating

tylerjw, 2020-12-18:
I think the way they are counted are the same as they are on Travis.  We would indeed be starting many per job but because we will be doing multi stage runs they should each run for much less time and some maybe not at all if they can recall a cache that would create the same output.

tylerjw, 2020-12-18:
I think the ssh feature sounds awesome.  The number of concurrent jobs sounds like a good argument for GitHub unless there caching limits mean we'd have concurrent jobs invalidating eachothers caches.

felixvd, 2020-12-18:
> Both tools face the risk of the people paying for them (investors in CircleCI, and Microsoft for Github) deciding it isn't worth supporting OSS anymore. I'm not sure I have any good reasons to believe either would be more likely to cancel OSS support.

True, but at least it looks like [both CircleCI (since recently)](https://circleci.com/blog/our-cloud-platform-your-compute-introducing-the-circleci-runner/), [and Github Actions](https://docs.github.com/en/free-pro-team@latest/actions/hosting-your-own-runners/about-self-hosted-runners) support self-hosted runners (which don't require hosting the whole CI; they just execute jobs), so there will be less pressure even if OSS support is dropped. Instinctively I would suspect that Github Actions is the safer bet in this regard, but it is a secondary concern imo.

> I was primarily concerned about us configuring and maintaining the CI scripts.

This is also my main concern. Simple seems best to me, unless the performance gains are enormous, and easy to maintain. I can't really judge though.

gavanderhoorn, 2020-12-18:
@tylerjw wrote:
> ## ease of use (from what I learned so far, Circle CI and github actions are similar to this regard)
>
> For me I think this is the one that is clearly in the camp of CircleCI. Because the nav2 team [...]

just pointing out that someone else having done a lot of work for you to reuse != greater ease of use.

How easy a service is to use would be an inherent characteristic of the service.

It's nice @ruffsl has done a lot of work already, but that doesn't mean circleci is easy to use. It just means we don't have to figure those things out ourselves. Adapting/changing things could still be "hell".

tylerjw, 2020-12-18:
> just pointing out that someone else having done a lot of work for you to reuse != greater ease of use.
> 
> How easy a service is to use would be an inherent characteristic of the service.

That is a fair point.  I don't have enough experience with using either enough to say one is better than the other in this regard.  They both seem to have complete enough documentation, a similar specification language (yaml files), and similar features (as you might expect seeing as they are solving the same problem).

Maybe @ruffsl can speak to the "ease of use" of circleci as a service.  Ideally, we'd have someone who's used both for a while that could speak to the pain points in each.  

In my limited use of GitHub actions, my only real complaint is your job seems to randomly queue for a long time. Yesterday I waited 30 minutes for a job to start on a repo when there were no other jobs running on that repo.  This might just be a temporary resource problem Microsoft is having as many people are migrating off Travis.

With circleci my only real complaint I've had in the time I've played with it is the xterm settings in the ssh session didn't play nice with kitty (the terminal emulator I use).  Although that doesn't really feel like a downfall of circleci... that's a known issue with kitty and it worked just fine with Terminator.  I really liked the ssh debug feature as it let me fix the issue I was having inside the ci runner instead of the much longer cycle time of trying to fix locally, pushing, waiting for it to get to where it was failing, and seeing if my fix worked.

For me being able to learn from and contribute to a common solution seems like an obvious upside.  I'm not sure if there are many other large ros libraries (other than moveit and nav2) that would benefit as much from the performance gains of a multi-stage pipelined ci solution to justify developing one.  In the case of many smaller ros projects, the simpler monolithic script is better because the potential savings in time running ci doesn't overcome the cost of the added complexity.  That is unless someone (like us and the nav2) maintained a robust generic ros project solution that handled the complexity of multi-stage ci in a generic but useful enough way.

Another argument for why the ssh debugging is so useful is that when developing this ci solution we are going to run into situations where we are developing the ci configuration and to test it we have to do full ci runs with changes in the configuration.  Being able to ssh into the machine and start a pipeline at a late stage instead of from the beginning every time seems to me like it would be very useful for those of us developing the ci configuration.  Developing a complex ci solution where your only option for testing is running it from the beginning would be like the "fun" experience of writing a long dockerfile without buildkit caching.

ruffsl, 2020-12-19:
Hey folks, great discussion! It’s nice to see Nav2 isn’t the only large ROS project exploring advanced container based CI services. I’ll try and respond inline to clarify points made throughout the discussion so far, but I’ll assume you’ve skimmed through the documentation I’ve written for Nav2 or at least watched the lightning talk I created for ROSWorld 2020:

- https://github.com/ros-planning/navigation2/tree/643bc089bbd7d8f64a68330543f6871361f50def/doc/continuous_integration
- https://discourse.ros.org/t/ros-docker-pro-tips-and-20-20-hindsights/17274

---

> nav2 doesn't test any downstream packages as we do and this split would also have to be adapted into moveit_ci so that we would test both moveit and downstream packages.

Would you want to split your workspaces into an `underlay`, `overlay`, and `downlay`? You could build the moveit dependencies into the `underlay`, and then the core of moveit stack into the `overlay`, and then any downstream packages into the `downlay`. You could also then test the `downlay` from a separate job in the workflow DAG. I’ll give some example workflows in a little bit.

> If we adapted our docker images to have a similar structure we could have the images built for noetic, melodic, noetic/ros-testing and noetic/clang-debug as separate docker images that would all be based on the same "cacher" image.

If by “cacher” image, you mean the stage from this [figure](https://github.com/ros-planning/navigation2/blob/643bc089bbd7d8f64a68330543f6871361f50def/doc/continuous_integration/dockerfile.md#advanced-optimizations), then that is not entirely correct. I think you meant to say “based on the same Dockerfile”, as the parent image for each tag would build `FROM` different ROS distro images, but could still be built from the same Dockerfile via docker build `ARG`s.

> By doing a build in docker hub and then doing a checkout of the pr branch in moveit_ci and build we would take advantage of cmake not rebuilding parts of moveit that didn't change.

Building the image in Docker Hub helps skip through installing apt/build dependencies for each workspace, as well as bake in a warm ccache for CI jobs. The cache restoration of build artifacts is currently done on a workspace level, as opposed to a per package level. This is for build higenics, as I didn’t feel like trusting whatever build system below colcon to invalidate its own build artifacts if package source files changed. I’ve stated some tickets on colcon to do this so that the workspace could always be restored without fear of not rebuilding stale artifacts:

- https://github.com/colcon/colcon-core/issues/213
- https://github.com/colcon/colcon-package-selection/issues/25

If you trust your cmake build setup to maintain a hygienic build and monitor source and install files for changes, then we could instead alway restore the workspace from a prior cache, and rely on cmake to rebuild package level changes for even faster rebuilds.

> clang-tidy itself is timing out on some of the files in the pilz planner.

I’m not as familiar with this linter, but this doesn’t need to be invoked during compile time, correct? This might be a good task to spin out into its own parallel job in the workflow, so it can run concurrently/independently of other build/test jobs with it’s own timeout.

> The downside to this is that we would lose or have to adapt many of the very nice features that moveit_ci has such as only running clang-tidy on cpp files that were changed in the given or.

How is this done currently? I imagine you could move the linters to their own job as mentioned, but reuse your same script that checks only the source files that changed by comparing the git history of the checked out repo. The CI config would take a few config lines, and the bash/python script need not be CI specific. An example:

- https://github.com/osrf/docker_images/blob/814bac8fa14b4f7ce4fa4de40f587f5e955afcdf/.ci/ci_script.py#L129-L130

---

> I've also reached out to a salesperson at circleci and was told that for large OS projects they can increase the number of credits and/or turn on some of their more advanced features.

Oh neat, TDIL! I was going to mention that *docker build layer caching* isn’t a free tier feature, and so I delegate DockerHub auto build to do the work to build the CI docker images. This also reserves more credits/workers for PR jobs on CircleCI. CircleCI for nav2 just pokes DockerHub’s auto build service to rebuild images via a scheduled “cron” job with curl and a web hook, but that may be worth changing if CircleCI where to enabled docker build layer caching for the ros-planning org.

- https://circleci.com/docs/2.0/docker-layer-caching

> The yaml file is huge and complex. If you have ventured into the nav2 ci configuration you've discovered a huge yaml file is where all of the logic is.

At the same time, it’s quite self contained, and doesn’t necessitate any external plugins that could move underneath you. The first 300 lines could remain unchanged and largely ignored when repurposing it for moveit, given it’s all just operating on ROS and colcon workspaces. The last 50 lines or so are what you’d have to repurpose, given how you’d like to shape, schedule, or trigger your CI workflow.

> The circleci solution to this is a thing they call an orb. Here is the documentation. Orbs are a way to standardize a workflow for a type of project. This is something we should consider doing in [...]

[... cut ...] nce just yet.

---

## Example

You’ll want to auto build docker image tags for the varent of tags you’d like to use to run the CI with, the use a matrix of jobs to build and test moveit for the distro and build modes youd like to run. There isn’t a upstream image ros using the testing repos, so that may need to be hosted as well, like under the osrf org. You may also like to run the linter tests befor building and testing code to encourage maintaining code style while iterating on PRs.

[image]

tylerjw, 2020-12-19:
> I’m not as familiar with this linter, but this doesn’t need to be invoked during compile time, correct? This might be a good task to spin out into its own parallel job in the workflow, so it can run concurrently/independently of other build/test jobs with it’s own timeout.

clang-tidy is a static analysis tool and afaik needs the compile commands json files similar to lcov.  We are currently running after doing a build and test with the clang compiler.  I do agree that it would make sense to make into a separate stage though.

> How is this done currently? I imagine you could move the linters to their own job as mentioned, but reuse your same script that checks only the source files that changed by comparing the git history [...]

It currently depends on some environment variables from Travis to work.  I think a more generic script would also make it much easier for people to run locally.  clang-tidy is run once for each compilation unit, this means that even if you want to run it over the full codebase you need some sort of script that calls it.

For reference see ament_clang_tidy: https://github.com/ament/ament_lint/blob/master/ament_clang_tidy/ament_clang_tidy/main.py
The implementation we currently use in moveit_ci: https://github.com/ros-planning/moveit_ci/blob/master/check_clang_tidy.sh

My plan is to try to adapt the bash script we have in moveit_ci to work standalone.  If however, we do switch to using colcon, we could use ament_clang_tidy.  I'm not sure if it has features for only running on changed packages though.  clang-tidy can take a small eternity (30sec per compilation unit) to run, so if you do it over the whole codebase each time it just won't be feasible with moveit.

> If I’m not mistaken, you could still use colcon with the packages as they are now; cmake change would only be required if you wanted to use features from ament, no?

I haven't tried this.  I didn't realize you could use colcon with cmake files that contain catkin functions.  Even if building work does testing work?  I'm going to have to try this locally and find out.

> Wouh, where did you find that neat stat on the org account?

I feel crazy now.  Yesterday it was showing the cache statistics here: https://app.circleci.com/settings/plan/github/ros-planning/usage  But now it doesn't.  It was showing a ton more details than just credits.  It showed the ros-planning org using 1.1tb of cache in November.

tylerjw, 2020-12-19:
Relating to building with colcon, the mixin argument seems to be missing on the version of colon you can install with apt on 18.04.

tylerjw, 2020-12-20:
On the current state of master, you can build just fine with colcon but some of the tests fail (it just runs ctest).   Note that these tests do not fail with catkin run_tests in a workspace built with catkin.

```
$ colcon test-result
build/moveit_commander/Testing/20201219-2357/Test.xml: 3 tests, 0 errors, 3 failures, 0 skipped
build/moveit_commander/test_results/moveit_commander/rostest-test_python_moveit_commander.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_commander/test_results/moveit_commander/rostest-test_python_moveit_commander_ns.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_commander/test_results/moveit_commander/rostest-test_python_time_parameterization.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_core/Testing/20201219-2351/Test.xml: 25 tests, 0 errors, 3 failures, 0 skipped
build/moveit_core/test_results/moveit_core/MISSING-gtest-test_aabb.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_core/test_results/moveit_core/MISSING-gtest-test_orientation_constraints.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_core/test_results/moveit_core/MISSING-gtest-test_robot_model.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_ros_perception/Testing/20201219-2351/Test.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_ros_perception/test_results/moveit_ros_perception/gtest-mesh_filter_test.xml: 24 tests, 0 errors, 1 failure, 0 skipped
build/moveit_ros_visualization/Testing/20201219-2357/Test.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/moveit_ros_visualization/test_results/moveit_ros_visualization/rostest-test_moveit_joy.xml: 1 test, 0 errors, 1 failure, 0 skipped

Summary: 922 tests, 0 errors, 16 failures, 2 skipped
```

I'm looking into understanding why these tests are failing.  Here are the next steps (and plan for now) I'm going to work through: 

1. Adapt docker image builds 
2. Begin testing on circleci using as close to a straight copy of nav2's config as possible on my fork
3. Adapt linter scripts
4. Add support for the full matrix of tests we do on Travis

The way I expect it to work in the end is this: 

Base docker images: 
melodic-release
noetic-release
noetic-release-ros-shadow-fixed
noetic-release-ros-testing
noetic-debug
noetic-release-clang

Pipelines:
```
On pull request commit and commits to main repo:
noetic-release -> clang-tidy
noetic-releaase -> build upstream -> build target (moveit) -> test target 
                                                             -> build downstream -> test downstream
melodic-release -> build upstream -> build target (moveit) -> run tests
noetic-debug -> build upstream -> build target (moveit) -> run tests -> report code coverage
noetic-release-clang -> build upstream -> build target (moveit) -> run tests
                                                                       -> run clang-tidy test

Nightly (same as above + these extra jobs):
noetic-release-ros-shadow-fixed -> build upstream -> build target (moveit) -> test target 
                                                                                      -> build downstream -> test downstream
noetic-release-ros-testing -> build upstream -> build target (moveit) -> test target 
                                                                             -> build downstream -> test downstream
trigger docker builds
```

I'll be splitting the rosinstall into two separate ones:
upstream.rosinstall: moveit_msgs, moveit_resources, geometric_shapes
downstream.rosinstall: rviz_visual_tools, moveit_visual_tools, moveit_tutorials, panda_moveit_config

If I can get colcon to work that'll require less change from the nav2 config.  I doubt this will not be as easy as just adapting it to use catkin instead though.  Thoughts?

tylerjw, 2020-12-20:
Another problem I just realized is if a user wanted to extend a catkin workspace from one of these colcon built ones they can't:
```
$ catkin config --install --extend ~/code/moveit/ws_upstream_colcon/install --cmake-args -DCMAKE_BUILD_TYPE=Releaseerror: could not configure catkin workspace: Cannot load environment from resultspace "/home/tyler/code/moveit/ws_upstream_colcon/install" because it does not appear to be a catkin-generated resultspace (missing .catkin marker file).
```

tylerjw, 2020-12-22:
On my debug test when it does the step where it moves the cache into the ramdisk I'm getting a bunch of these:

```
mv: error writing '/mnt/ramdisk/.ccache/5/f/54e7b3ddc7d667681e8a5acacf618f-10288799.o': Cannot allocate memory
```

I wonder if your limiting the cache to 200mb is the ccause of your low hit rate on debug.  That is if by setting that it is deleting a useful part of your cache so that it can fit it in ram.

tylerjw, 2020-12-22:
> This is for build higenics, as I didn’t feel like trusting whatever build system below colcon to invalidate its own build artifacts if package source files changed.

cmake (and make under it) doesn't work correctly for this?  Do we not all develop locally with incremental builds?  I'm trying to understand the corner cases where incremental builds won't work.  Do you have any examples of what these might be?

tylerjw, 2020-12-23:
@ruffsl and others an update on what I've tried and where I'm at with trying to use your circleci config for moveit:

Building and running tests with `colcon` vs `catkin` is not trivial.  There are a handful of edge cases where cmake configs in moveit depend on inner workings of catkin and I haven't been able to figure any of them out.  The one I've spent the last few hours on is `_moveit_roscpp_initializer` which is a symbol that is built for use in python.  This is used by moveit_commander and as a result if you build with colon you can't get any of the tests to work for moveit_commander.  I believe this has something to do with how the cmake file is written for building that into a python library and how it uses the devel space (that only exists in catkin).  Secondly, not all dependencies are handled the same and I've been having a weird build error in docker where it complains about a missing library that exists in the source.

That being said, I've also made some progress doing a nieve conversion to `catkin` of your config.  I say nieve because I'm not able to use caching of the upstream workspace because you depend on features of colon to tell if the previous build (in docker) finished and was successful.  I need to look into the source of these colcon features and see if I can figure out a way to get a similar affect in catkin.

Next, logs in catkin are done differently and that has to be adjusted in the config.  I haven't figured that out yet.

Then there is the issue of ccache in docker.  As you can see from the outputs I posted above, when you build with debug you get a huge cache with 68 files "cleaned up" because 5gb isn't even enough.  With a release build it creates 313mb of cache.  One of the downsides to the huge debug cache is that there isn't enough ram in the circleci machine to copy 5gb of cache into a ramdisk.  The size of a unrestricted debug cache might have something to do with your awful hit rate.  If you run ccache stats, how large is the cleanups performed number?

I've learned all about the latest features of colcon and feel a bit dumb that it took me a while to figure out how to get the latest features on 18.04.  Here is how you do it if you didn't already know (someone should call this out somewhere in the docs, it's all there just took me a while to figure it out):
```
sudo apt install python3-colcon-mixin
colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml
colcon mixin update default
```

Another thing, I discovered in the colcon docs there is support for ninja.  I was very excited about this and charged ahead with trying this.  This simple change will probably speed up any colcon builds by a bit: `--cmake-args -G Ninja`.

If anyone else has the desire or time to hack on this (or any other ci solution for moveit) let me know.  I'm sure there are other things I'm missing in this update I could help answer if you run into snags.

tylerjw, 2020-12-23:
> Adapting/changing things could still be "hell".

yes, sadly, yes. :'(

tylerjw, 2021-01-05:
An update on this.  I've been trying to factor out the fuctionality of our clang-tidy test from moveit_ci with these goals:

* Standaone script that can be run locally, and therefore CI provider/system agnostic
* Has options for filtering clang-tidy tests to packages in a repo
* Has options for filtering clang-tidy tests to packages with changes since some point in git
* Collaberate with others in the ROS community to contribute these features so they can be used more widely

I've taken the approach of making a pr against ament_clang_tidy for this.  As a result of the need for tools "at a higher abstraction level" from ament to do the filtering it has been pointed out that these changes probably don't belong in ament_lint.  I don't know what to do about this yet.

The other thing I've been working on is getting moveit to work in circleci with modified versions of the yaml and dockerfiles from nav2.  I've taken both the approach of adopting to colcon as the build tool and adapting the script to catkin in parallel.  These both have drawbacks I have yet to overcome:

Convert to colcon is blocked by our use of the inner workings of catkin.  There are packages in moveit that use cmake variables that are catkin specific and not part of the more generic catkin interface that we use.  Specifically this has to do with our generated python interfaces.  I've not figured out how to adapt our cmake files to work without these variables.

Convert CircleCI config to use catkin_tools.  This has the problem of missing freatures that the circleci and docker config is taking advantage of for caching.  Unless I can figure out how to replicate the missing functionality from colcon or find another aproach to some of the caching steps we will loose some of the benefits of the CircleCI system.

There are pros and cons to both of these apraches.  If we convert to using colcon there are a handful of upsides: The opertunity to share more ci configuration between nav2, moveit, and moveit2, and potentially many other projects.  Locally we can convert to using colcon and that makes the transition to ros2 easier as there are less tools to keep track of in our heads.  It also makes it easier when going back and forth, and even facilitates having mixed ros1/ros2 projects.  The downside is that not everyone will convert from catkin and colcon and we'll have to deal with bug reports if we accidentaially make changes that are not compatible to both.  This increases the difficulty of maintence.

For the option of converting the config to catkin_tools.  This presents the least amount of change to users, and makes the CI system resemble our current workflows the closest, and requires no changes to the cmake of MoveIt.  The only real downside is misssing features in catkin_tools relative to colcon and less overlap with ci config for ros2 projects.

ruffsl, 2021-01-06:
> Convert to colcon is blocked by our use of the inner workings of catkin. There are packages in moveit that use cmake variables that are catkin specific and not part of the more generic catkin [...]

Could you link to some examples here specifically?

> I've taken the approach of making a pr against ament_clang_tidy for this. As a result of the need for tools "at a higher abstraction level" from ament to do the filtering it has been pointed out that these changes probably don't belong in ament_lint. I don't know what to do about this yet.

Could you link to that PR as well?

ruffsl, 2021-01-06:
> Then there is the issue of ccache in docker. As you can see from the outputs I posted above, when you build with debug you get a huge cache with 68 files "cleaned up" because 5gb isn't even enough. [...]

For Nav2, the ccache size is on the order of 100MB, but you are right in that the debug builds never had as high of a hit rate:

* https://app.circleci.com/pipelines/github/ros-planning/navigation2/4459/workflows/82fb7439-2feb-433e-bf18-44b604da8261/jobs/16666/parallel-runs/0/steps/0-122
* https://app.circleci.com/pipelines/github/ros-planning/navigation2/4459/workflows/82fb7439-2feb-433e-bf18-44b604da8261/jobs/16665/parallel-runs/0/steps/0-122

I've always suspected that some entropy is leaking into the debug build mode, like a time stamp or ephemeral debug symbol output for code coverage. I'm guessing there could be a gcc or clang flag to suppress that entropy, but I haven't looked at it for a while.

* https://answers.ros.org/question/331493/poor-ccache-hit-rate-on-debug-build-with-coverage-set-using-colcon

But if your ccache size exceeds a substantial percentage of the host's RAM, then you may want to change the `CCACHE_DIR` back to use a `/tmp` directory rather than the mounted ram disk path.

tylerjw, 2021-01-06:
> Could you link to that PR as well?

Here are my instructions for a local source install of the ament linters for use by moveit: https://github.com/ros-planning/moveit/pull/2470
Here is my PR to ament_lint: https://github.com/ament/ament_lint/pull/280

> Could you link to some examples here specifically?

Here is the cmake file that I've been having trouble with in colcon: https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/py_bindings_tools/CMakeLists.txt

The code builds, the problem is when tests run that depend on these python bindings they can't be found.

tylerjw, 2021-01-07:
More details on the issue with the planning_interface py_bindings_tools.  The issue is the use of the variable `CATKIN_DEVEL_PREFIX` as colcon doesn't have the concept of a devel space.

tylerjw, 2021-01-08:
I needed this for the perception test:

```
export LIBGL_ALWAYS_SOFTWARE=1
```

For the rest of them, I was able to change the one line in the python interface that used the devel space to just point at the install space, and that fixed those issues.

tylerjw, 2021-03-25:
We have now migrated to github actions here: https://github.com/ros-planning/moveit/pull/2562

Closing this for now.

closed 2021-03-25 by tylerjw