moveit/moveit #2465: the following packages/stacks could not have their rosdep keys resolved

oridong, 2020-12-21:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Source  build
* If source, which branch?
when I try to install according to  "https://moveit.ros.org/install/source/"
every step succeeds until 
this step "rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}"
produces
 rosdep install -y --from-paths src --ignore-src --rosdistro noetic
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_ros_visualization: Cannot locate rosdep definition for [pkg-config]
pilz_industrial_motion_planner: Cannot locate rosdep definition for [liborocos-kdl-dev]
moveit_ros_occupancy_map_monitor: Cannot locate rosdep definition for [eigen]
moveit_commander: Cannot locate rosdep definition for [python3]
rviz_visual_tools: Cannot locate rosdep definition for [libqt5x11extras5-dev]
moveit_core: Cannot locate rosdep definition for [liborocos-kdl-dev]
moveit_resources_prbt_support: Cannot locate rosdep definition for [eigen]
geometric_shapes: Cannot locate rosdep definition for [libqhull]
moveit_ros_planning: Cannot locate rosdep definition for [eigen]
moveit_setup_assistant: Cannot locate rosdep definition for [yaml-cpp]
moveit_tutorials: Cannot locate rosdep definition for [eigen]
moveit_ros_perception: Cannot locate rosdep definition for [opengl]
moveit_ros_planning_interface: Cannot locate rosdep definition for [python3]
moveit_ros_manipulation: Cannot locate rosdep definition for [eigen]
moveit_kinematics: Cannot locate rosdep definition for [liborocos-kdl-dev]

rhaschke, 2020-12-21:
Did you follow the "Install ROS" instructions? All these packages should be known after `rosdep init` + `rosdep update`.

oridong, 2020-12-21:
> Did you follow the "Install ROS" instructions? All these packages should be known after `rosdep init` + `rosdep update`.

yes,as it is

> d@ubuntu:~/Robotic/ws_moveit$ rosdep update
> reading in sources list data from /etc/ros/rosdep/sources.list.d
> Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
> Skip end-of-life distro "ardent"
> Skip end-of-life distro "bouncy"
> Skip end-of-life distro "crystal"
> Add distro "dashing"
> Add distro "eloquent"
> Add distro "foxy"
> Skip end-of-life distro "groovy"
> Skip end-of-life distro "hydro"
> Skip end-of-life distro "indigo"
> Skip end-of-life distro "jade"
> Add distro "kinetic"
> Skip end-of-life distro "lunar"
> Add distro "melodic"
> Add distro "noetic"
> Add distro "rolling"
> updated cache in /home/d/.ros/rosdep/sources.cache
>

closed 2020-12-22 by oridong