moveit/moveit #2475: Wrong gripper commands for parallel gripper with mimic joint

bjsowa, 2021-01-06:
### Description
I have a parallel gripper where 2 opposite fingers are mechanically connected, so I define 2 prismatic joints where the second one mimics the first one. I also have a controller which implements the GripperCommandAction, which expects to receive the gap between the fingers (sum of finger joint positions) as its goal.
On the moveit side, I define the group for the gripper with some defined states and the GripperCommand controller with the `parallel` option set to true.
The problem is that when I execute the path, moveit does not sum the position of the mimicked joint when sending the GripperCommand. If I remove the mimic tag from URDF, the GripperCommands are now sent correctly, but I would rather avoid it, since it requires sending duplicated joint state.
Is this a bug or am I misunderstanding something?

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build: 1.0.7

### Steps to reproduce
I can provide a full demo if needed, but for now, here are some snippets:

URDF:
```xml
<joint name="gripper_finger1_joint" type="prismatic">
  <origin xyz="0.019 0 0" rpy="0 0 0"/>
  <axis xyz="0 1 0"/>
  <limit effort="30" velocity="0.01" lower="0.001" upper="0.0158"/>
  <parent link="gripper_finger_base_link"/>
  <child link="gripper_finger1_link"/>
</joint> 

<joint name="gripper_finger2_joint" type="prismatic">
  <origin xyz="0.019 0 0" rpy="${pi} 0 0"/>
  <axis xyz="0 1 0"/>
  <limit effort="30" velocity="0.01" lower="0.001" upper="0.0158"/>
  <parent link="gripper_finger_base_link"/>
  <child link="gripper_finger2_link"/>
  <mimic joint="gripper_finger1_joint"/>
</joint>
```

SRDF:
```xml
<group name="gripper">
  <joint name="gripper_finger1_joint" />
  <joint name="gripper_finger2_joint" />
</group>
<group_state name="gripper_open" group="gripper">
    <joint name="gripper_finger1_joint" value="0.0158" />
    <joint name="gripper_finger2_joint" value="0.0158" />
</group_state>
<group_state name="gripper_mid" group="gripper">
    <joint name="gripper_finger1_joint" value="0.0084" />
    <joint name="gripper_finger2_joint" value="0.0084" />
</group_state>
<group_state name="gripper_closed" group="gripper">
    <joint name="gripper_finger1_joint" value="0.001" />
    <joint name="gripper_finger2_joint" value="0.001" />
</group_state>
```

controllers:
```yaml
  - name: gripper_controller
    action_ns: gripper_action
    type: GripperCommand
    default: true
    parallel: true
    joints:
      - gripper_finger1_joint
      - gripper_finger2_joint
```

Simplified version of the launch file I use for starting moveit
```xml
<launch>
  <include file="$(find my_moveit_config)/launch/move_group.launch">
    <arg name="allow_trajectory_execution" value="true"/>
    <arg name="fake_execution" value="false"/>
    <arg name="info" value="true"/>
    <arg name="pipeline" value="ompl"/>
    <arg name="load_robot_description" value="true"/>
  </include>

  <include file="$(find my_moveit_config)/launch/moveit_rviz.launch">
    <arg name="rviz_config" value="$(find my_moveit_config)/launch/moveit.rviz"/>
  </include>
</launch>
```

### Expected behaviour
I expect the goal positions in the GripperCommands sent by moveit to be equal to the sum of positions of two joint (the active one and the mimic joint).

### Actual behaviour
The goal positions in the GripperCommands are equal to the position of the `gripper_finger1_joint`.

simonschmeisser, 2021-01-13:
I had some issues and discussions wrt gripper opening vs finger joint positions as well before. Without having looked in detail at the  GripperAction I still think it would be hard (and therefore error prone) to calculate the gripper opening in all cases (there are for example grippers with one active joint and five mimic ones). Therefore I assume the action only cares about joint positions and this is all it can do.

You could either modify your hardware driver, add a wrapper node in-between or subclass/copy the GripperAction ....  or look at the code :-D

https://github.com/ros-planning/moveit/blob/023b11def6329b165ed7509e5de9aec4c1e29c6c/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/gripper_controller_handle.h#L202

defines a `bool` that can be set to enable your desired behavior. I have not found quickly how to actually set it, please try searching a bit for where that function is called and report back here. Then we can also add this to the docs (if missing)

bjsowa, 2021-01-13:
> I had some issues and discussions wrt gripper opening vs finger joint positions as well before. Without having looked in detail at the GripperAction I still think it would be hard (and therefore [...]

I understand that calculating the gripper opening from joint positions can be difficult in some cases but my case is the most basic and common for these types of grippers (or at least I through so). 

[image]

I have two parallel fingers where the position on the second one is mimicked from the first one.

> You could either modify your hardware driver, add a wrapper node in-between or subclass/copy the GripperAction .... or look at the code :-D [...]

To set this bool to true I add this line to the controller configuration:
```yaml
parallel: true
```

If the second joint does not contain the `mimic` tag, it works as expected by sending the gripper openings which are the sum of the positions on both joints.

As soon as I add the `mimic` tag to the joint, the position on the second joint is not added to the gripper opening.

simonschmeisser, 2021-01-13:
please run a `rostopic echo` on the gripper action goal topic. I guess it will just contain one joint if you set the second to mimic. And this makes sense from my POV. So the code here https://github.com/ros-planning/moveit/blob/023b11def6329b165ed7509e5de9aec4c1e29c6c/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/gripper_controller_handle.h#L130  might be actually outdated and need some discussion

bjsowa, 2021-01-13:
The gripper command does not contain the list of joint, it only sends the desired gripper opening and max effort. Here's the output of `rostopic echo`:
```
header: 
  seq: 1
  stamp: 
    secs: 1610567451
    nsecs: 764294958
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1610567451
    nsecs: 764297333
  id: "/move_group-2-1610567451.764297333"
goal: 
  command: 
    position: 0.00848705444014631
    max_effort: 0.0
---
```