moveit/moveit #2482: Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info

YijiongLin, 2021-01-10:
### Description

Hi gentlemen,

I tried to run the moveit tutorial posted [here](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html). And I have set up a Conda environment with python 2.7 and then ran it. I could successfully run the first command which showed up the RViz with a Panda arm, however, when I ran the second command it popped out an error:

`Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
`
I have checked out the issues 86 but it seems it is not relevant to my case. 

And I also have run these two commands without Conda (comment out the Conda path exported in the ~/.bashrc file), and they all ran successfully. So I don't have any clue about how it would fail with Conda.

Any help and suggestion would be highly appreciated!

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build?
source
* If binary, which release version?
* If source, which branch?
I used the file  from here: https://github.com/ros-planning/moveit_tutorials.git -b melodic-devel

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
It should execute the plan coded in the py file.

### Actual behaviour
It showed an error: Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info

### Backtrace or Console output

closed 2021-01-10 by YijiongLin

YijiongLin, 2021-01-10:
Something weird happened. I just comment out all the "print" statement in this file, and it would run successfully without Conda but unsuccessfully with Conda (popped out the aforementioned error). However if I keep all the "print" statements, it both run successfully. 

(The reason why I commented out the "print" was that I first ran it with python 3 and it not worked. Then I commented them out but soon I realized I should run it with python 2, therefore I set up python 2 via Conda.) 

Though it solved, could anyone give me an insight into it?

Thank you!

rhaschke, 2021-01-11:
The Conda version of python sets up its own module path, which might comprise modules that are incompatible with the standard system ones. Either stay with the system installation or install all required modules yourself.