moveit/moveit #2485: Segfault from the PlanningScene after allocating for Bullet collision detection

AndyZe, 2021-01-12:
Attempting to allocate for Bullet collision detection like this:

```
planning_scene_monitor_->getPlanningScene()->setActiveCollisionDetector(collision_detection::CollisionDetectorAllocatorBullet::create());
```

or like this:

`
  {
     planning_scene_monitor::LockedPlanningSceneRW ls(planning_scene_monitor_);
      ls->setActiveCollisionDetector(collision_detection::CollisionDetectorAllocatorBullet::create());
  }
`
Causes a seg fault. Per gdb:

```
[Switching to Thread 0x7fff86ffb700 (LWP 25923)]
0x00007ffff20a1b9b in planning_scene::PlanningScene::clearDiffs() ()
   from /home/andyz/ws_al/devel/lib/libmoveit_planning_scene.so.1.1.1

```

AndyZe, 2021-01-13:
Note: Bullet woks on the tutorial (a standalone Panda manipulator) and another dual-arm mobile robot we have. There is something particular to this setup.

AndyZe, 2021-01-22:
Starting to track where exactly this crashes...

It is in `planning_scene_monitor_->providePlanningSceneService();`

AndyZe, 2021-01-22:
I think what's happening is, we set the CollisionDetector to Bullet for the child scene, but it's still FCL for the parent scene.

tjh49, 2021-05-24:
same error, even not work on the tutorial