moveit/moveit #2486: Joint Limits not showing in the joint_limits.yaml

ly19264, 2021-01-13:
### Description

After creating the moveit package using moveit setup assistant, the joints_limits.yaml does not show the joint limits present in the URDF. Does moveit consider the joint limits if yes, then where does it store them, if no, do we manually have to add them to joints_limits.yaml

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04

Steps to reproduce
Creating a moveit package from moveit setup assistant

Expected behaviour
Joint Limits should be taken from the URDF and be displayed in the joint_limits.yaml

### Actual behaviour
The joints_limits.yaml only shows the velocity limit
joint_limits:
joint0:
has_velocity_limits: true
max_velocity: 100
has_acceleration_limits: false
max_acceleration: 0

simonschmeisser, 2021-01-16:
If you have position limits set for a joint in `joint_limits.yaml` then those will override the ones from URDF. If you don't specify them in `joint_limits.yaml` then the ones from URDF will be used.

So it appears to me that you would like to have the values from URDF copied to `joint_limits.yaml` for convenience? Please open a PR for it if that is what you intend. Don't hesitate asking questions for any problems you encounter!

The code is here: https://github.com/ros-planning/moveit/blob/664ae01803abf5e0b4649063102357262de9e05c/moveit_setup_assistant/src/tools/moveit_config_data.cpp#L1118

If you are super enthusiastic, you could of course also create a widget/page to adjust these

ly19264, 2021-01-16:
@simonschmeisser ,

Thanks for the help, a quick query ,

Does the moveit package stores the joint limits from the URDF or does it reads those limits from the URDF during execution ?

simonschmeisser, 2021-01-16:
The package you create by using MoveIt setup assistant contain launch files that load config files (including joint_limits.yaml) as well as the SRDF and URDF to the parameter server (rosparam). The URDF is typically part of a robot package that you can install (from source or deb). MoveIt (class `RobotModel`) first reads the joint limits from the URDF (from parameter /robot_description) and then overrides them with values from joint_limits.yaml (if there are any). Typically the URDF contains the hard limits from the manufacturer and joint_limits.yaml sets your local setup (if you don't need a certain rotation or have some cable that is not long enough etc.)

ly19264, 2021-02-05:
@simonschmeisser , Is there a way to fetch/read the joint limits from the moveit package during runtime. I would like to access those values to check the solution is in limit or not.

AndyZe, 2021-02-05:
Hello Nitish, long time no see!

Yep, you can do that. [Here's](https://github.com/ros-planning/moveit/pull/2471/files) an example.

ly19264, 2021-02-05:
Hello @AndyZe , thanks for your help. 

In the example, I beleive the values being fetched from this command,
const auto& bounds = joint->getVariableBounds(joint->getName());
Is the command correct ?

would it fetch all the bounds or the one specified in joints_yaml file ?

AndyZe, 2021-02-05:
joint_limits.yaml. That takes precedence in MoveIt, if it's non-empty

ly19264, 2021-02-05:
Ok, for the joint limits the file initially has empty values and as per my understanding from Simon's help that the values are read from the URDF during execution. 

So will the command mentioned above be able to fetch those values and do we have a python implementation of the same ?

AndyZe, 2021-02-05:
That sounds correct.

It looks like there is [not](http://docs.ros.org/en/noetic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html) a Python way to retrieve it. Sorry. Make a pull request, if you can.

ly19264, 2021-02-05:
Thanks for the help I will make a pull request. 

Also unrelated to the current thread. Is there a way to update the current position of the robot arm end_effector in moveit using python? I know how to fetch the current position in moveit but the real position and moveit current position are different for my case.

felixvd, 2021-03-01:
I'm labeling this as "good first issue" because both the widget for the joint limits and the Python bindings are very isolated changes that could be useful and educational, even if grokking the codebase is not necessarily easy. You can discuss here if you are confused while reading.

Sunny478, 2021-03-16:
So what is the final desired output? Do we want that if no limits are set in joint_limits.yaml, then the limits from URDF get copied to the joint_limits.yaml file?

sourabhsingh282, 2021-03-16:
I want to work on this issue kindly assign me @ly19264 ly19264

v4hn, 2021-03-16:
> So what is the final desired output? Do we want that if no limits are set in joint_limits.yaml, then the limits from URDF get copied to the joint_limits.yaml file?

I would expect that if the position limits are added to the file (as proposed here), they will be *disabled* (`has_position_limits=false`) or *commented out* by default.
A very nice addition would be a MoveIt Setup Assistant panel where you can customize all the limits via sliders (in which case they would become active of course).

Sunny478, 2021-03-17:
> The code is here:
> 
> https://github.com/ros-planning/moveit/blob/664ae01803abf5e0b4649063102357262de9e05c/moveit_setup_assistant/src/tools/moveit_config_data.cpp#L1118

I am not so sure about this. I commented the whole function, still the output of the joint_limits.yaml file is exactly the same (with velocity_limits and acceleration_limits and all other stuff) . Can someone help me in resolving this behavior.