moveit/moveit #2504: Bullet: attached collision objects are not detected

AndyZe, 2021-01-27:
Ubuntu 18.04, building from master branch.

There is a branch [here](https://github.com/AndyZe/moveit_tutorials/tree/andyz/bullet_two_links_collision) that attaches a 1m cube to the base of the robot. Run the tutorial like normal to reproduce the problem:

`roslaunch moveit_tutorials bullet_collision_checker_tutorial.launch`

JafarAbdi, 2021-01-27:
I tested the branch locally and this's what I got, are you getting a different output .?

```
-----------------------------------------
PlanningScene Known Objects:
  - Collision World Objects:
 	- world_cube
  - Attached Bodies:
	- attached_cube
-----------------------------------------
```

AndyZe, 2021-01-27:
That part is OK. It's just checking that the object was attached. In RViz, if you drag the robot around, there won't be any collision with the box near the base of the robot.

Also, I believe the attached box won't be displayed.

JafarAbdi, 2021-01-28:
That's because the robot state of  `g_planning_scene` isn't the same as the one in `interactive_robot` (which is what we see in RViz) to sync them you can do the following after that you should be able to see the `attached_cube` object

```cpp
g_planning_scene->processAttachedCollisionObjectMsg(attach_object);
*interactive_robot.robotState() = g_planning_scene->getCurrentState();
```

AndyZe, 2021-01-28:
Thanks @JafarAbdi. It appears to work as it should (on master branch, at least).

closed 2021-01-28 by AndyZe