moveit/moveit #2505: Make parameters for python get_joint_names more explicit

PeterMitrano, 2021-01-28:
It's odd to me that `get_joint_names` returns fixed/mimic joints when you do not pass a group name, and simultaeously does not let you get fixed or mimic joints when you do. There's no way to get "all" non-fixed or non-mimic joints without making a group for all joints and passing that group name.

It seems to me this could be simplified by including new parameters, `include_fixed` and `include_mimic`, which default to false, and control the behavior. When no group is specified we consider all joints, and the include_* parameters are obeyed regardless of whether a group name is passed

Alternatively, we could introduce a new function, or make only one new parameter `include_fixed_and_mimic`. I'm happy to PR this if we can agree on an API.

PeterMitrano, 2021-02-28:
On second thought, I think it's probably better to just make two functions, `get_joint_names` and `get_active_joint_names`, where active excludes fixed and mimic. This would be easier to implement. I'll make a PR for this

v4hn, 2021-03-01:
I wasn't even aware of the difference! :disappointed: 

The same is true for the underlying C++ interfaces. 
Should we also create matching `getActiveJointModelNames` methods in `moveit::core::RobotModel`?
If we want it, it probably makes sense to add an `std::vector<std::string> active_joint_model_names_vector_` to the class to avoid a different runtime behavior than all the other functions like that in `RobotModel`.
That increases the memory footprint of the class a bit, but that should not be a problem.

closed 2021-03-02 by PeterMitrano