moveit/moveit #2506: Noetic source build error with FCL

captain-yoshi, 2021-02-02:
### Description

There are undefined refrerences to FCL methods when doing `catkin build` to compile moveit from source. I have added the FCL package (See Steps to reproduce). Cannot compile Moveit Noetic at the moment.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? Source
* If source, which branch? master

### Steps to reproduce
[MoveIt install steps](https://moveit.ros.org/install/source/)
[FCL install steps](https://moveit.ros.org/install/source/dependencies/)
- Make using the master branch
- Added fcl package in the catkin workspace with the modified package.xml
- Renamed package.xml to libfcl-dev and also tried with fcl

### Backtrace or Console output

Here is the [console output](https://gist.github.com/captain-yoshi/271cc22e29f5fc905cfc56704c7b05c3).

AndyZe, 2021-02-02:
Can you please take a look at the deleted notes here:

https://github.com/ros-planning/moveit.ros.org/pull/546/files

Especially the part re. manually installing `ros-noetic-fcl`. Does that help?

captain-yoshi, 2021-02-02:
Weird... I did a `catkin clean` when I read about this [PR](https://github.com/ros-planning/moveit/issues/1491) and now it compiles fine. Is installing FCL from https://moveit.ros.org/install/source/dependencies/ the official way, or is the deleted notes the proper way? Tanks for your help.

AndyZe, 2021-02-02:
I don't think you need the "deleted notes" but wanted to bring it up, just in case. Sounds like your problem is solved, then?

captain-yoshi, 2021-02-02:
Yes problem solved!

closed 2021-02-02 by captain-yoshi