moveit/moveit #2511: Manually updating robot_state

ly19264, 2021-02-09:
Hi, 

I am using a bio_ik_service to compute the inverse kinematics solution and the solution is published to the robot controller topics to move the robot. Is there a way to update the inverse kinematics solution to the robot state so that robot.get_current_state refers to the achieved solution of the inverse kinematics. Also, if possible is there a python implementation of the same ?

I am using ROS Melodic on Ubuntu 18.04

Thanks for the help and guidance in advance.

felixvd, 2021-02-13:
If you use a `move_group_interface`, you can [set the start state](http://docs.ros.org/en/noetic/api/moveit_core/html/classmoveit_1_1core_1_1RobotState.html#afd01134540eb6ccc57e75d1f8c4f48d5) for planning, but the "actual" (current) state of the robot is determined by the `current_state_monitor`, e.g. your robot driver. It makes sense that you can't just overwrite that, since this interface is supposed to be used with real robots (or simulated ones).

If you are running a simulation (e.g. Gazebo), you can set the robot position in your simulated environment, and the `move_group_interface` will update accordingly. If you are running `demo.launch` which launches MoveIt's fake controllers, I'm not sure if there is a way to set the joint positions at runtime (there might be - I just never looked into it).

If you are only interested in visualizing, you can use the [RobotState](http://docs.ros.org/en/noetic/api/moveit_core/html/classmoveit_1_1core_1_1RobotState.html) directly (not through the `move_group_interface`), [set joint positions](http://docs.ros.org/en/noetic/api/moveit_core/html/classmoveit_1_1core_1_1RobotState.html#afd01134540eb6ccc57e75d1f8c4f48d5) (and retrieve them via the getter functions), and publish that to Rviz. I don't think this has Python bindings, but you are welcome to contribute.

felixvd, 2021-02-23:
I guess this is solved.

closed 2021-02-23 by felixvd

ArsenicBismuth, 2024-02-28:
@felixvd Then how do you plan for a consecutive actions before executing them? I want to plan several pose targets before executing them in a row. And the plan(n) will need to set the start point based on the plan(n-1)'s last joint states.

In simulation this def works perfect, the arm go for the first plan and the 2nd plan. But this doesn't work in real robot.

You mentioned in another thread here (and a clearer instruction by another):
- https://answers.ros.org/question/374875/how-can-i-set-last-state-of-a-trajactory-as-the-start-of-new-trajactory-in-ros/
- https://answers.ros.org/question/274209/planning-together-multiple-sequential-motions-in-moveit/

felixvd, 2024-02-29:
> how do you plan for a consecutive actions before executing them? [...] the plan(n) will need to set the start point based on the plan(n-1)'s last joint states.

That's exactly what you do - you specify the RobotState in each planning request. This is without a doubt the correct approach. It's also how MoveIt Task Constructor works internally.

> this doesn't work in real robot.

Then there's a problem with your implementation. You should post it in more detail on the [Robotics Stack Exchange](https://robotics.stackexchange.com/) (the replacement to answers.ros.org) and cross-link it in/with this thread.

ArsenicBismuth, 2024-02-29:
@felixvd That's unfortunate. I was using Open Manipulator and it seems its implementation with moveit is broken/imperfect.

One example was that using `execute(planA)` somehow doesn't actually execute the `planA`, but instead the last `plan()` being called. And then `path, success = cartesianComputePath()` directly executes the path, before I even called the `execute(path)`

Also for the Robotics Stack Exchange, I posted various questions there that I thought would get attention. But none of it are answered, so I kinda gave up using it. But yeah I will post there anyway since this is def not the place.